bigtreetech/BIGTREETECH-TouchScreenFirmware #1732: BTT UPS

sinole1987, 2021-03-14:
I have the btt ups, and i see that there is a ups feature, i wanted to know if i should connect the panic pin to tft screen as well for the signal or main board will send to signal to tft? and if I should connect to screen, which pin should I connect to?

gordo3di, 2021-03-15:
Refer to this https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1458

Lots of testing with the UPS but it seems like it only works properly through Marlin.

oldman4U, 2021-03-15:
sinole1987 in the BIGTREETECH-MINI-UPS-V2.0 repository you ask about the Relay v1.2. Which device are you finally using or is it about both?

sinole1987, 2021-03-15:
I bought both of those moduls, I though I can build my printer with automatic shutdown and power fail protection, which to be honest power fail is not that necessary. But it wasn't to hard to add until TFT screen came into picture. I didn't know that TFT software is so under developed.  Relay works great, since it just need to hear M80 from board. the ups on the other hand. if it doesn't work right. it can even corrupt tft Firmware, which is kinda scary.

sinole1987, 2021-03-17:
> sinole1987 in the BIGTREETECH-MINI-UPS-V2.0 repository you ask about the Relay v1.2. Which device are you finally using or is it about both?

can you tell me the file in which pins mapped for tft program, because someone wrote program for  connecting UPS to tft screen but I cant find to which pin they assigned the UPS.

bigtreetech, 2021-03-21:
Hello, BTT UPS needs to be connected to the motherboard and detected by Marlin in order to obtain the highest priority. Once the power is failed, some high-power loads such as hotbed and heating rod will be shut down immediately, so that the power in UPS can supply the stepper to move a longer distance.

First of all, you need to set the `POWER_LOSS_RECOVERY ` function in Marlin, and then you need also to set some parameters on the touch screen config.ini. The most important thing is to enable start_gcode and add M75 if printing from TFT SD or TFT Udisk

In addition, if it is printed OnBoardSD and the TFT is working in touch mode, after power failure and restart, the TFT can not detect the `POWER_LOSS_RECOVERY` information of Marlin by Serial data, so it needs to manually send` M1000 S1` gcode in the Terminal menu to resume printing

You can get some information in here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/890#issuecomment-667027180

sinole1987, 2021-03-21:
> Hello, BTT UPS needs to be connected to the motherboard and detected by Marlin in order to obtain the highest priority. Once the power is failed, some high-power loads such as hotbed and heating rod will be shut down immediately, so that the power in UPS can supply the stepper to move a longer distance.
> 
> First of all, you need to set the `POWER_LOSS_RECOVERY ` function in Marlin, and then you need also to set some parameters on the touch screen config.ini. The most important thing is to enable start_gcode and add M75 if printing from TFT SD or TFT Udisk
> 
> In addition, if it is printed OnBoardSD and the TFT is working in touch mode, after power failure and restart, the TFT can not detect the `POWER_LOSS_RECOVERY` information of Marlin by Serial data, so it needs to manually send` M1000 S1` gcode in the Terminal menu to resume printing
> 
> You can get some information in here [#890 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/890#issuecomment-667027180)

i had issues with activating star Gcode in the past. should I remove star Gcode from slicer and and only used the tft Gcode? 
also can I change tft start Gcode ?

sinole1987, 2021-03-21:
Im going to try that. thank you

sinole1987, 2021-03-30:
> Hello, BTT UPS needs to be connected to the motherboard and detected by Marlin in order to obtain the highest priority. Once the power is failed, some high-power loads such as hotbed and heating rod [...]

it works, but when print logs in and I start a print. it wont proceed to printing. I have to disconnect and reconnect screen from menu or go to marlin and back to start the print again.

sinole1987, 2021-03-30:
worked great thank you.

oldman4U, 2021-03-30:
👍🏻

oldman4U, 2021-03-30:
How have you solved the need to send M1000 S1?

sinole1987, 2021-03-30:
It just sends it to main board. and it interrupts the host. I had no issue there. but it is very important to set start G code exactly as was recommended by BTT post so the host knows where it is. I tested a USB print and even that one worked.

sinole1987, 2021-03-30:
Although, there is a hardware issue with UPS module that I hope BTT  solves in revision 2. It seems that power delivery and power draw is not very smooth. when you turn of the printer it turns of slowly, screen flashes and stuff. and when you turn it back on capacitor are empty. they draw so much power that printer doesn't boot. you have switch it on and off again. I will post this on respective page later.

oldman4U, 2021-03-30:
As far as I understand all the TFT does is to let Marlin know that a print
has started and the resume after failure which is independent of the UPS
device.

sinole1987, 2021-03-30:
well, I connected the ups to the main board, and it is sending m1000 to marlin. perfectly interrupts any print on tft and raise the nozzle.

oldman4U, 2021-03-30:
Once the power is back the TFT also works? BTT suggested to send M1000
using Terminal.

sinole1987, 2021-03-30:
yes, once the power is back, I just go to menu and select the media I was printing from, then it will ask if I want to continue print that was interrupted. If you confirm it will heat up printer, homes, purges and continues printing. but it will dump a little filament on the print again for some reason just like m600 used to, its much less though. I also activated these two lines in marlin: [https://github.com/sinole1987/Marlin/pull/1/commits/8fb7fff97f8e45e458deac915b3ab8c0973eeb2f](url)

oldman4U, 2021-03-30:
Great, thank you!

Happy printing

ahmetbozaci, 2021-04-17:
After upgrade to  this "BIGTREE_TFT35_V3.0.27.bin" version. Ups is not working anymore with marlin mode or touch screen mode. I don't know I miss something or not.

**Here is my configs:**

- marlin/src/feature/powerloss.h

`//#define DEBUG_POWER_LOSS_RECOVERY`
`#define SAVE_EACH_CMD_MODE`
`#define SAVE_INFO_INTERVAL_MS 0`

- marlin/configuration_adv

`#define POWER_LOSS_RECOVERY`
`#if ENABLED(POWER_LOSS_RECOVERY)`
`#define PLR_ENABLED_DEFAULT false // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)`
`#define BACKUP_POWER_SUPPLY       // Backup power / UPS to move the steppers on power loss`
`//#define POWER_LOSS_RECOVER_ZHOME  // Z homing is needed for proper recovery. 99.9% of the time this should be disabled!`
`#define POWER_LOSS_ZRAISE 10      // (mm) Z axis raise on resume (on power loss with UPS)`
`#define POWER_LOSS_PIN P1_25      // Pin to detect power loss. Set to -1 to disable default pin on boards without module.`
`#define POWER_LOSS_STATE HIGH     // State of pin indicating power loss`
`#define POWER_LOSS_PULL           // Set pullup / pulldown as appropriate`
`#define POWER_LOSS_PURGE_LEN 20   // (mm) Length of filament to purge on resume`
`#define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.`
`// Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,`
`// especially with "vase mode" printing. Set too high and vases cannot be continued.`
`#define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data`
`#endif`

- marlin/pins

`#ifndef POWER_LOSS_PIN`
`#define POWER_LOSS_PIN P1_00 // PWRDET` // **I connected ups cable to this pin. I used 'P1_25' for btt relay**
`#endif`

- tft/config

`pl_recovery_en:1`
`pl_recovery_home:0`
`pl_z_raise:10`
`btt_mini_ups:1`
`start_gcode_enabled:1`
`end_gcode_enabled:1`
`cancel_gcode_enabled:1`
`start_gcode:M75\nG28 XY R20\n`
`end_gcode:M77\nM104 S0\nM140 S0\nM107\nM18\n`
`cancel_gcode:M77\nM104 S0\nM140 S0\nG28 XY R10\nM107\nM18\n`

sinole1987, 2021-04-17:
PLR is not working in marlin 2.0.7.2 . I really don't see myself going back and setting up for marlin 2.0.6. but i know it works. As for marlin bugfix, I couldn't get it to work with TFT. X and Y didn't home. and everytime i pressed home X or Y on TFT nozzel came down full force and smashed into the bed.

ahmetbozaci, 2021-04-17:
I am not sure problem is marlin version. Because ı use powerloss before upgrade tft and ı am using last version of marlin.

sinole1987, 2021-04-17:
I know for a fact many other people had problem using PLR in marlin 2.0.7.2, which is that last version. there are some open tickets on marlin GitHub. [https://github.com/MarlinFirmware/Marlin/issues/19602](url)

oldman4U, 2021-04-28:
Most likely you have seen the https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.git which has a minor bug fix for PLR.

Check if it solves your problem

closed 2021-07-04 by stale[bot]