bigtreetech/BIGTREETECH-TouchScreenFirmware #1733: MIissing Translation

abeksis, 2021-03-15:
Adding Language - Hebrew.

I ask you to add support for the Hebrew language. I can provide the translated file, I would love for you to include it in your version.

[https://github.com/abeksis/Hebrew-Language--Artillery-Genius](url)

oldman4U, 2021-03-15:
Hi.

Why not just pushing a PR?

abeksis, 2021-03-16:
`Hi.

Why not just pushing a PR?`

PR? What does that mean

oldman4U, 2021-03-16:
This is open source software. So users like you and me are providing new features and bug fixes by contributing so called Pull Requests (PR).

In case you can not do so, we can try to find someone who is willing to do it for you.

radek8, 2021-03-16:
If you have created your own translation with Hebrew, and you want to use it, all you have to do is rename the ini file as Italian, for example, and upload it to TFT.
In TFT you will choose Italian but the translation will be in Hebrew :-)

radek8, 2021-03-16:
There are currently 21 language translations created in FW.
Unfortunately, half of them are out of date. because no one cares about them. Nevertheless, programmers must create items in them for the FW to pass the check. Creating another official translation that no one will care about is not effective.

radek8, 2021-03-16:
The translation you offer is not even 40% translated

oldman4U, 2021-03-16:
abeksis

Could you please provide 100% of translation.

Thank you

abeksis, 2021-03-17:
Yes I will do that ... 100% translation

oldman4U, 2021-03-17:
Perfect. To keep it usable you will also have to maintain it from time to time.

abeksis, 2021-03-18:
Hi,
There is a slight problem with translation from English to Hebrew, and I want to consult before I continue.
The translation I translate in the opposite way - a mirror effect.

I do it deliberately because once I translated normal and moved to the printer, see Hebrew upside down. The question can provide support to the Hebrew language in your code it will display correctly and without the possibility of a mirror effect.

In addition, the display in Hebrew presented has large profits between a signal and a sign that it does not sit well.

radek8, 2021-03-18:
Another user has already tried to solve a similar problem with writing fonts from right to left.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1290

abeksis, 2021-03-19:
Ok, what does it mean? For further translate. Translate regularly? Or effect shows

abeksis, 2021-03-19:
I translate in the usual way without a mirror effect. I have already reached 90 percent. I will be happy to have a link to the right to the right.

radek8, 2021-03-19:
If you want to implement in FW
it is necessary to insert the translation into the file language_xx.h
the language_xx.ini file is generated automatically.

abeksis, 2021-03-19:
Ok, I've already translated 90 percent

radek8, 2021-03-19:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT/src/User/API/Language

abeksis, 2021-03-19:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT/src/User/API/Language

https://github.com/abeksis/Hebrew-Language--Artillery-Genius/blob/main/language_he.ini

I'd be happy to do so.

Here's my file.

abeksis, 2021-03-19:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT/src/User/API/Language

Ok, I'll translate the Language_he.h file.

abeksis, 2021-03-20:
The translation is ready.

There are no values that are not suitable for Hebrew and therefore remained about their original.

I will continue to translate each time an update left.

Is the Font problem resolved? Not to be given profits between a signal and a signal? Hebrew and Arabic language?

closed 2021-05-27 by stale[bot]