bigtreetech/BIGTREETECH-TouchScreenFirmware #1736: Probe offet for mesh validation

RocksaltX, 2021-03-16:
Add Z offset for UBL mesh validation (business card mode?)
When validating the mesh data the Z position is always zero. 
Example: G29 performed, mesh data stored. Go to mesh editor, pick a point, start. Z position is always to low with shim under nozzle. Go back to babystep, increase babystep height by paper thickness, re-enter mesh editor, validate.

**Describe the solution you'd like**
Option added for business card mode to set the business card thickness

**Describe alternatives you've considered**
Below is a copy and paste of the config.ini. Could this be applied to the Z position for mesh edit?
## Z Position (Manual Leveling)
# Lower z axis to this position after reaching a leveling point.
#   Unit: [position in mm]
#   Value range: [min: 0, max: 2000]
level_z_pos:0.05

**Additional context**
None

digant73, 2021-03-16:
if you mean something like:
1) one Mesh point is Z = -0.25
2) current mesh editor moves the nozzle to Z = -0.25 (as the mesh value)
3) your desired mesh editor moves the nozzle to Z = -0.25 + bed_level_z_pos
   where bed_level_z_pos is a configurable value in config.ini

then the answer is yes, it can be done.
Something like that is already done for MBL only. The nozzle is always (hard coded) moved to Z = 0.2. While by default Marlin will first move the nozzle to the previous set Z height

RocksaltX, 2021-03-16:
Yes, just like that! I have changed the config.ini to .05mm but it does not change the UBL mesh edit moves. It's more of an annoyance in that I have to change my probe offset before validating the mesh.

RocksaltX, 2021-03-16:
I have been compiling my own BTT 3.5 firmware. Is there a place I can just put the offset as I use the same shim to validate. So instead of Z to mesh offset+0 can I change it, and where, to the mesh+.05mm?

RocksaltX, 2021-03-16:
What would be even better... if the probe offset is set, with shim, the mesh data would already be compensated. For instance, if I set my probe offset such that it is .05mm above zero, the mesh editor automatically uses the probe offset.  This is what I expected to happen. When I go to validate, the nozzle is always too low, but when homing Z the offset is correct.

thineight, 2021-03-26:
Hi, with the latest Marlin 2.0.7.2 and TFT firmware on a Artillery Sidewinder X1 I experience the following issue (not present with the previous 2.0.6.1 release):
- Manual bed levelling MBL 4x4 points (number of points should be irrilevant anyway)
- Performed the usual 4 corner calibration with the skrews (bed and nozzle preheated)
- During mesh calibration the nozzle goes up 15mm at each point, and I have to lower down that much before the fine tune of the z.

It is a bit annoying in my opinion to lower down each point 15 mm, so that initial offset should be nearly zero to me.
I didn't find any obvious setting in the TFT config nor the Marlin configuration.h, unless I missed something.

Any clue?
Thanks a lot.

digant73, 2021-03-26:
@thineight my mistake. The nozzle should be raised by "level_z_pos" (e.g. 0.2mm). I wrongly used "level_z_raise".
I will patch the bug

thineight, 2021-03-26:
not sure I fully understood, but in the TFT config.ini i see level_z_pos: 0.2 , that looks the value you mentioned.

I imagine then that the bug is hardcoded and recompile is needed.
thanks!

digant73, 2021-03-26:
@thineight the code (a bug) is using level_z_raise instead of level_z_pos. I will fix the bug

digant73, 2021-04-01:
@thineight wait for the PR #1785 merge. it fixes the bug on MBL and implements also this FR

thineight, 2021-04-01:
Great, thanks.
one last thing: with the firmware 2.0.7.2, once performed the MBL on the 4x4 points with the piece of paper, I would expect that during the first layer the nozzle keeps those (tight) z elevation, but I see that I have to lower it all the times with the babystep by at least 0.14 mm otherwise the filament does not stick to the bed.
Is it some other settings to change in the config or in the tft, e.g. the paper thickness or some global z offset?

digant73, 2021-04-01:
not yet. The problem with MBL is that the Z height is saved by Marlin each time you move to the next point. So if you use a shim in MBL, the value including the shim is saved by Marlin when moving to next point. E.g. if for the current point I have a z height of 0.200 including a shim of 0.100, 0.200 will be saved by Marlin when moving to next point. But the value we would like to save is 0.200 - 0.100. The solution could be bo lower the nozzle to the desired z height (so 0.200 - 0.100) just before moving to next point but in this case you will have the nozzle pressing on the shim/bed just before moving to next point. I will study a better solution.

closed 2021-04-03 by bigtreetech