bigtreetech/BIGTREETECH-TouchScreenFirmware #1741: [FR] Manual bed leveling Ender 5

Vybramo, 2021-03-18:
Hello

I fix position in script for bed leveling from TFT for ender 5 cause its pointing to wrong position. Now its read correctly MIN MAX Y X axis and point head to right position regarding to icons on LCD and positions. 

[image]

#include "Leveling.h"
#include "includes.h"

const MENUITEMS manualLevelingItems = {
  // title
  LABEL_LEVELING,
  // icon                          label
  {
    {ICON_POINT_4,                 LABEL_POINT_4},
    {ICON_POINT_3,                 LABEL_POINT_3},
    {ICON_LEVEL_EDGE_DISTANCE,     LABEL_DISTANCE},
    {ICON_DISABLE_STEPPERS,        LABEL_XY_UNLOCK},
    {ICON_POINT_1,                 LABEL_POINT_1},
    {ICON_POINT_2,                 LABEL_POINT_2},
    {ICON_POINT_5,                 LABEL_POINT_5},
    {ICON_BACK,                    LABEL_BACK},
  }
};

void moveToLevelingPoint(uint8_t point)
{
  int16_t pointPosition[5][2] = {
    {infoSettings.machine_size_max[X_AXIS] - infoSettings.level_edge, infoSettings.machine_size_max[Y_AXIS] - infoSettings.level_edge},
    {infoSettings.machine_size_min[X_AXIS] + infoSettings.level_edge, infoSettings.machine_size_max[Y_AXIS] - infoSettings.level_edge},
    {infoSettings.machine_size_min[X_AXIS] + infoSettings.level_edge, infoSettings.machine_size_min[Y_AXIS] + infoSettings.level_edge},
    {infoSettings.machine_size_max[X_AXIS] - infoSettings.level_edge, infoSettings.machine_size_min[Y_AXIS] + infoSettings.level_edge},
    {(infoSettings.machine_size_min[X_AXIS] + infoSettings.machine_size_max[X_AXIS]) / 2, (infoSettings.machine_size_min[Y_AXIS] + infoSettings.machine_size_max[Y_AXIS]) / 2},
  };

  if (coordinateIsKnown() == false)
  {
    storeCmd("G28\n");
  }

  storeCmd("G0 Z%.3f F%d\n", infoSettings.level_z_raise, infoSettings.level_feedrate[FEEDRATE_Z]);
  storeCmd("G0 X%d Y%d F%d\n", pointPosition[point][0], pointPosition[point][1], infoSettings.level_feedrate[FEEDRATE_XY]);
  storeCmd("G0 Z%.3f F%d\n", infoSettings.level_z_pos, infoSettings.level_feedrate[FEEDRATE_Z]);
}

void menuManualLeveling(void)
{
  KEY_VALUES key_num = KEY_IDLE;

  menuDrawPage(&manualLevelingItems);

  while (infoMenu.menu[infoMenu.cur] == menuManualLeveling)
  {
    key_num = menuKeyGetValue();
    switch (key_num)
    {
      case KEY_ICON_0:
        moveToLevelingPoint(3);
        break;

      case KEY_ICON_1:
        moveToLevelingPoint(2);
        break;

      case KEY_ICON_2:
        {
          char tempstr[30];
          sprintf(tempstr, "Min:%d | Max:%d", LEVELING_EDGE_DISTANCE_MIN, LEVELING_EDGE_DISTANCE_MAX);

          int val = numPadInt((uint8_t *)tempstr, infoSettings.level_edge, LEVELING_EDGE_DISTANCE_DEFAULT, false);
          infoSettings.level_edge = NOBEYOND(LEVELING_EDGE_DISTANCE_MIN, val, LEVELING_EDGE_DISTANCE_MAX);

          menuDrawPage(&manualLevelingItems);
        }
        break;

      case KEY_ICON_3:
        storeCmd("M84 X Y E\n");
        break;

      case KEY_ICON_4:
        moveToLevelingPoint(0);
        break;

      case KEY_ICON_5:
        moveToLevelingPoint(1);
        break;

      case KEY_ICON_6:
        moveToLevelingPoint(4);
        break;

      case KEY_ICON_7:
        infoMenu.cur--;
        break;

      default:
        break;
    }

    loopProcess();
  }
}

bigtreetech, 2021-03-19:
This seems to be caused by different machine structures. The lower left corner is (X_min 0, Y_min 0) of the common printer. According to your description, The upper right corner is (X_min 0, Y_min 0) of this printer.

digant73, 2021-03-19:
@bigtreetech we could use the value of "infoSettings.invert_axis" to invert the logic as it is done on Move menu.

if (infoSettings.invert_axis == false) -> the current logic is used

else

if (infoSettings.invert_axis == true) -> the logic proposed by Vybramo is used

digant73, 2021-03-19:
@Vybramo did you configure inverted axis in config.ini just also to properly use the Move menu?
That means someting like:

invert_axis:X1 Y1 Z0

Vybramo, 2021-03-19:
> @Vybramo did you configure inverted axis in config.ini just also to properly use the Move menu?
> That means someting like:
> 
> invert_axis:X1 Y1 Z0

Yes I have all axis inverted x1 y1 z1 because of ender 5. I have another idea how to make it. Just select which corner is x0 y0 and select which axis is y and x or by invertion.

digant73, 2021-03-19:
@Vybramo I already made the PR to support your FR. It is based on checking the "invert_axis" settings.
Please test the PR when it will be available

closed 2021-03-21 by bigtreetech

Vybramo, 2021-03-21:
Works good but missing Inverting from menu. Its possible do over config.ini file. Few features disappear from screen settings / features.

Also please add :

## Z Position (Manual Leveling)
# Lower z axis to this position after reaching a leveling point.
#   Unit: [position in mm]
#   Value range: [min: 0, max: 2000]
level_z_pos:0.1

to menu with :

## Leveling Edge Distance (Manual Leveling)
# Inset distance from bed edges. This distance is added to minimum X & Y bed coordinates and
# subtracted from maximum X & Y bed coordinates to calculate manual leveling points.
#   Unit: [distance in mm]
#   Value range: [min: 0, max: 2000]
level_edge_distance:30

Personally I measure paper/any other think which I use for level and set there this value. Its greatly help with first layer cause you don`t need reconfigure your slicer.  Default value its 0.2mm but paper its 0.08-0.1mm thick

digant73, 2021-03-21:
"level_edge_distance" can be already changed from manual leveling menu

open a new FR for new enhancement requests. Just to keep track.

oldman4U, 2021-03-24:
digant73

Should we really think about changing a preset which has been working for other users. 80g paper is about 0.1mm but initial layer hight for most users is 0.2 and if you use a 2 paper thick 80g or another paper with a different hight or a filler gauge with another value, this will always result in different z hight adjustment. I can not see how changing this value to something else would resolve for all other users but only fir this particular user.

Maybe I am wrong, you are the level expert.

PR #1748 (2021-03-21): resolves FR #1741 + increased max mesh grid size + MBL relative initial height
* max mesh grid size is now set to 15x15 (it was 10x10) just to be compliant to the limit suggested on Marlin fw (a 15x15 for UBL). The mesh data are already dinamically allocated entering on the Mesh Editor and released on exit, so no global memory is statically increased. That max limit makes sense to be reached only on big build volume (big printers).
* resolves FR #1741
* MBL: when nozzle is moved on next point, the nozzle is raised (relative coord) by the configurable setting "level_z_raise" (already used for m [... cut]