bigtreetech/BIGTREETECH-TouchScreenFirmware #1746: BTT TFT35 V3.0 is not showing print status with SKR1.4T RRF 3.2.2

basha318i, 2021-03-19:
When printing from DWC, the TFT35 doesn't show the print status. it shows the temp changes, but not the print screen.

bttguy, 2021-03-19:
Hi! Did you made the required sys changes mentioned in #1743 
After I set those up, it worked.

basha318i, 2021-03-19:
@bttguy Hey, thanks for the suggestion. I'll re-flash a fresh FW & test since I've been testing with couple of other options like Thor42's FW.
I will keep you posted.

basha318i, 2021-03-19:
@bttguy Thanks again, this worked fine for me, although the pause & stop behaviour is not working. for example if the printer is still heating up & you try to pause, it would show a message "Busy.. still processing..." until it finishes the heat cycle then accepts any commands. While if you try to pause mid-way in the heating cycle from DWC, it would immediately execute the pause.

bttguy, 2021-03-19:
Hmm, thanks for the info, will try to reproduce. This feature is new, and might need some further polish with the help of some real use feedback. @Msq001 Does this a limitation or an user error?
By far now the main branch will be the way to go with RRF, as you can see in #1400 compatibility work has started and going pretty nice.

jaysuk, 2021-03-19:
@basha318i did you add M0 to your end gcode?
Stop.g doesn't get called automatically the same way start.g does

bttguy, 2021-03-19:
@jaysuk Where should the M0 added? Only to the end gcode, or to stop/pause.g as well?

Msq001, 2021-03-20:
> @bttguy Thanks again, this worked fine for me, although the pause & stop behaviour is not working. for example if the printer is still heating up & you try to pause, it would show a message "Busy.. [...]

In my test(start print from DWC + abort print from TFT, or start print from TFT + abort print from TFT ), Abort printing by the TFT"stop" button will not wait for the temperature to rise, it only waits for RRF to give M27 a "Not SD printing" response, and then TFT exits the "Busy Processing, " menu.

bttguy, 2021-03-22:
@basha318i Did it works for you, have you tested it further?

jaysuk, 2021-03-22:
@bttguy M0 should be added only to the end gcode. M0 causes stop.g to be ran because, unlike start.g, its execution is not automatic
See here for stop.g info https://duet3d.dozuki.com/Wiki/Macros#Section_Start_pause_stop_and_power_fail

basha318i, 2021-03-22:
@bttguy I tested, but still the pause takes a while to be executed.
not sure if I need to modify anything in my TFT config.ini?
I added M0 to the end of my g-code from my slicer settings, but not sure if this is relevant? since the pause is my current issue & the M0 will be executed after the print is finished.

basha318i, 2021-03-22:
@bttguy okay, I just re-flashed again & tested & it's working as it should. I started a print from DWC, paused it from the TFT & it immediately executed the pause & updated the DWC.
Thanks, I think I have a working TFT now.
I also noticed that when I try to print from the TFT, the on-board SD doesn't show the folders I have under /gcode directory correctly. it shows them as files, is that a known issue?

bttguy, 2021-03-22:
@basha318i  Glad it works! There are still features that are missing, like what you mention, files right now only listed without any folder system. For now I tested my screen for the features that should work with the lastest commits, but I will test it for possible missing/not properly working features, and will make a FR, hopefully some dev could help us :)
Until that if you have any positive/negative experience/suggestion/FR regarding RRF comment on #1400 for easier tracking of the compatibility development! If it is working now, you could close this issue, since its working, and discuss the others in #1400 :)

basha318i, 2021-03-22:
@bttguy now while it's printing; it's not updating the values (extruder temp, bed temp, layer height, progress, fan speed..etc) the only thing I can see is the timer, please find the attached photos while it's printing.
[image]
[image]
[image]

basha318i, 2021-03-22:
it's inconsistent; I tried again & it works fine!
There must be a specific sequence I'm doing that causing this..
[image]

bttguy, 2021-03-23:
@basha318i Well, I have no idea... It's working fine for me ,with the same release and settings. In which configuration do you use RRF, with a WiFi adapter on with an SBC?

basha318i, 2021-03-23:
I’m using a WiFi adapter.

closed 2021-04-03 by basha318i