bigtreetech/BIGTREETECH-TouchScreenFirmware #1747: [BUG-Hunt] TFT freeze (even when not printing or printing with other host)

DaStivi, 2021-03-19:
### Description

i had ah couple of freezes on the TFT again, even i print with Octoprint only the last weeks.. so basically 98% of the prints work without freeze, when its freezing its not related to the printtime (i guess?) because i had freezes on long prints and also on short 1:30h prints..

### Steps to reproduce what i, at least, suspect now

1. Start print (maybe its also happening with prints started from TFT too?!)
2. changed babysteps and/or Fan Speed
(3. on the recent freeze, i left the display on fan speed change view)

**Expected behavior**
not freezing... so i just try to track and maybe got some others that noticed this, to help to track down (hunt) this random bug

oldman4U, 2021-03-19:
Whats wrong with #244?

DaStivi, 2021-03-20:
> Whats wrong with #244?

>Check
>Another USB stick
>That you have a TFT firmware from this week  and Marlin 2.0.7 or newer
>The TFT defined as the first serial port in Marlin config and 115200 as the
>baud rate
>The buffer settings in Marlin

other USB and SD card can be rules out, as the hangs also occoured while printing from OctoPrint-Pi
TFT Firmware is the latest nightly one and Marlin 2.0.7.2
115200baud rate is set
#define TX_BUFFER_SIZE 128 (not sure on RX right now i think its 1024) but also i guess this shouldn't really matter as TFT isn't sending gcode
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define MAX_CMD_SIZE 96
#define BUFSIZE 32

oldman4U, 2021-03-20:
Yes. But this repository is for the firmware of the TFT. So the question is if the problem is related to the software or not.

bigtreetech, 2021-03-21:
Hello, The freeze you described mean that: "the screen can't be touched at all, and it also cann't update the status information?" looks like the screen is dead?
When you print with octoprint and TFT is only used to monitor status information, The TFT will also die?
Please help to test whether the TFT will also die after it is only power on and keep the TFT and Printer all idle.
If the idle state will still die, it sounds more like a hardware problem.

bigtreetech, 2021-03-21:
By the way, what's your screen hardware type？ TFT35 V3.0?

oldman4U, 2021-03-22:
And is this a new problem you see on your printer? What changes, firmware?

DaStivi, 2021-03-22:
> Hello, The freeze you described mean that: "the screen can't be touched at all, and it also cann't update the status information?" looks like the screen is dead?
> When you print with octoprint and [...]

not had it happen without anything gooing on... so i am almost sure its some issue under certain circumstances, and as i wrote in steps above, the only thing i can remember i did was changing babysteps and fan speed... 

> By the way, what's your screen hardware type？ TFT35 V3.0?

its ah B1 TFT (v3) Varriant but basically i don't think its really hardware related

> And is this a new problem you see on your printer? What changes, firmware?
yeah sometimes... tft firmware updates.. but even there it happend with different compiles..  

i still try each octoprint print to change settings in tft and to figure out whats causing the tft freezes, but its so random and rare (at least i didn't figure out how to trigger it)

oldman4U, 2021-03-22:
Random and rare sounds great, but hard to find the reason. In case it is really firmware related, I hope another user can find it and therefore solve it.

Happy printing

DaStivi, 2021-03-24:
got another freeze right now... printed back-to-back from octoprint the whole day from 7:45 until now... where suddenly and apperently right on turning of the heaters and fan it looks like the display frozen... printer was now cooled down alleready, just tried to reheat with octoprint again, the display isn't also now just reacting on touch, its also not updating anymore... temps froze on print temp, 235 and 70, fan 0% -> whats strange because there where definitely fan running until the last layers!

its also not reacting to the longpress, switch to marlin change request...

so add another variable now, it looks like this takes at least some time the tft needs to run, to occur!
short vid: https://1drv.ms/v/s!AiIZqPOtqD4XqNdrEzkQJUW6gZoiDg

longer one, where i also try to heat again with octoprint and home/move printer, so marlin still running, after reset display is working again:

i'm not sure if i propapbly could to anything when the display is frozen to get more info out of the status, i guess without some debug pin connect no "memory-dump" or something like this is possible?

Vybramo, 2021-03-31:
I have also issue sometimes when printing in marlin mode. During printing its not possible to do anything.  Screen is freezed but printer printing without problem. I try any configuration in FW but still same issue. 
Mainboard skr 1.4 turbo and TFT its TFT 35 V3.0 E3

DaStivi, 2021-04-04:
> I have also issue sometimes when printing in marlin mode. During printing its not possible to do anything. Screen is freezed but printer printing without problem. I try any configuration in FW but still same issue.
> Mainboard skr 1.4 turbo and TFT its TFT 35 V3.0 E3

hi @Vybramo you printed from board SD? or octoprint (external printhost)?
anything you can remember you've changed while print on TFT??

Vybramo, 2021-04-05:
> hi @Vybramo you printed from board SD? or octoprint (external printhost)?
> anything you can remember you've changed while print on TFT??

When printing from onboard SD, For example in marlin mode I cannot do anything. Its just click and nothing, but printner print without issue.

DaStivi, 2021-04-05:
@Vybramo I meant if you set something specific on tft in marlin mode, like manual change temp, fan speed, feed rate, or babystepped the print

sinole1987, 2021-04-06:
Hi , I get freezes middle of Long printers recently. i dont know what is the problem. I have TFT 43 use it with skr 1.4.

digant73, 2021-04-07:
@DaStivi keep us updated if with the latest BTT fw you still continue to experience random freezes

sinole1987, 2021-04-07:
I updated the firmware, Although I made some changes. turned of powerloss recovery In case it creates a hiccup  in the system, and changed my flash driver in case my 10 years old flash drive couldn't keep up with all of the random IOPs and had a sucessful 18 hours print. now i'm turning on the PLR on my next print and let ya'll know if it works.

sinole1987, 2021-04-12:
> Hello, The freeze you described mean that: "the screen can't be touched at all, and it also cann't update the status information?" looks like the screen is dead?
> When you print with octoprint and [...]

I updated to latest firmware. I get hard freezes in the middles prints. first the printing stops, as i go to touch the screen, after first touch screen goes dead too, only thing remains to do is to reset the printer. you could recover the print if there wasn't a giant blob on the print location.

DaStivi, 2021-04-12:
did you print from SD or USB on the TFT? you could try to remove the Storagedevice... ihad this a couple of times that the device had an issue, either removed the SD or bad USB Pen and the TFT started to work again... i then changed to another USB Pen and didn't had that issue... 

i didn't managed to reproduce the freezing TFT yet with octoprint :( tried each printjob different things to change on TFT but also have to admit i hadn't any real long jobs gooing on..

sinole1987, 2021-04-12:
> did you print from SD or USB on the TFT? you could try to remove the Storagedevice... ihad this a couple of times that the device had an issue, either removed the SD or bad USB Pen and the TFT [...]

I thought it was my old flash drive, so I changed to a brand new metal flash drive so it doesn't heat up either. but it didt freez on me 1 time again. I have a 0.9 A , 0.9 degree stepper motor running my Y axis. this motor is very powerful ,but it has very unique specs. The phase resistance of the motor is 6 ohms which is quite high. I might have been overheating the driver on Y axis. I am not quite sure. I have reduced the Vref to 320miliamps this , motor is running cooler now. I hope this this was the issue. I have now idea how this motor produce 36Ncm on such low current values.

digant73, 2021-04-12:
> did you print from SD or USB on the TFT? you could try to remove the Storagedevice... ihad this a couple of times that the device had an issue, either removed the SD or bad USB Pen and the TFT [...]

yes, please let us know trying also with long prints as when you had the problems

DaStivi, 2021-04-12:
> I thought it was my old flash drive, so I changed to a brand new metal flash drive so it doesn't heat up either. but it didt freez on me 1 time again. I have a 0.9 A , 0.9 degree stepper motor running my Y axis. this motor is very powerful ,but it has very unique specs. The phase resistance of the motor is 6 ohms which is quite high. I might have been overheating the driver on Y axis. I am not quite sure. I have reduced the Vref to 320miliamps this , motor is running cooler now. I hope this this was the issue. I have now idea how this motor produce 36Ncm on such low current values. I put the link to the motor if anyone can explain black magic in this motor to me: [https://www.amazon.com/gp/product/B00W98OYE4/ref=ppx_yo_dt_b_asin_title_o07_s01?ie=UTF8&psc=1](url)

if the print froze, and tft also froze... just connect an USB Cable to the board and see if marlin it self is frozen... but then you've another issue... 

btw. i've also running the 17HM15-0904S 0.9s running on X and Y now... not had an freeze since installed them... the get warm/hot yeah, but as long as i touch it for ah few seconds with the fingers its fine... really ned to get an IR tempgun sometimes... but yeah as long as you can touch them for ah couple of seconds they'r well under the 70 or 80 °C what the can handle with ease...

sinole1987, 2021-04-12:
> if the print froze, and tft also froze... just connect an USB Cable to the board and see if marlin it self is frozen... but then you've another issue...
>
> btw. i've also running the 17HM15-0904S 0.9s [...]

yes , its not the motor , I even changed them and I still got frozen.

sinole1987, 2021-04-12:
are you using the power loss recovery feature? with M75 and M77 , in start and end  G_code ? I got freezes after I setup PLR feature. and I am wondering why not many other people have this freezing issues. If it wasn't an specific thing, a lot more people would post bug report.

DaStivi, 2021-04-14:
> are you using the power loss recovery feature? with M75 and M77 , in start and end G_code ? I got freezes after I setup PLR feature. and I am wondering why not many other people have this freezing issues. If it wasn't an specific thing, a lot more people would post bug report.

interessting thought... and yes, i did had powerloss recovery enabled... at least at some point in time, had some issues with ups and turned on and off ofter.. not sure what was the last status on how i got the freezes... but at least since ah few weeks i didn't had it turned on, as i moved every addon to the skrboard directly... and i'am sure i had also the M75 command in the start code... but still, as my screen also freezed while starting the job from Octoprint it didn't really used the M75 Stard Gcode from TFT...

sinole1987, 2021-04-14:
All of my freezes started when I added M75. I'm sutting down PLR as well. Marlin PLR is bugged on 2,0.7 and im not going back to 2.0.5. and on TFT it hardly works. as for octoprint its very mind twisting. I'm not using it, but it should have nothing to do with TFT. Maybe marlin don't really like external Hosts. this is why prusa charges premium for their printers. marlin 2.0.x still didn't catch up. I say post this on octoprint as well and see if anyone get freezes using it with SKR boards and marlin.

pauluchin, 2021-04-24:
I vave the same problem...
Board : E3 mini V2  /  Firmw: Marlin 2.7.2  /  TFT35-E3 V3.0 Firmw V27
Print from usb or octoprint
When printing files generated by cura or prusaslicer the screen begins to emit sounds after 1 hour or more of printing and after a while you cannot access marlin emu and after that the screen completely locks

DaStivi, 2021-04-24:
@pauluchin do you have also powerloss recovery enabled in TFT Menu?

pauluchin, 2021-04-24:
If it was activated I will deactivate and try again

sinole1987, 2021-04-25:
> If it was activated I will deactivate and try again

that might not help, I have don a lot of test , It is not really the PLR , its is M75 command most likely. Somehow timing of the print messes up and printer stops. That is my hunch . there should be a line of program that compensate for that. It may be a marlin issue though even though marlin doesn't freeze at all. it might cause tft to freeze.

digant73, 2021-04-26:
@sinole1987 no specific handling is performed for M75 on TFT. A good info could be to monitor the communication with Marlin and provide it to me just to make some cross checks on the code

sinole1987, 2021-04-26:
> @sinole1987 no specific handling is performed for M75 on TFT. A good info could be to monitor the communication with Marlin and provide it to me just to make some cross checks on the code

Can I do that with connecting printer to the PC and run Pronterface? Or do I need to connect something to screen?

digant73, 2021-04-26:
Yes If with that you can see the messages. Otherwise try with the terminal menu

sinole1987, 2021-04-26:
> Yes If with that you can see the messages. Otherwise try with the terminal menu

Well terminal menu is no longer accessible when Screen freezes. leaves us with an external option.

digant73, 2021-04-27:
I know but i could have logged something interesting before freezing

sinole1987, 2021-04-27:
well I monitored communication through  TFT terminal and esp3d and prontoface. nothing interesting happens. on TFT screen it just changing. stops on last two Gcodes. and on  ESP3D I got following:

T:214.79 /215.00 B:59.90 /60.00 P:35.00 /0.00 @:71 B@:16
X:145.00 Y:109.86 Z:13.20 E:0.00 Count X:14500 Y:21971 Z:5293
 T:215.00 /215.00 B:59.95 /60.00 P:35.00 /0.00 @:44 B@:0
 T:215.21 /215.00 B:59.95 /60.00 P:35.00 /0.00 @:44 B@:52
 T:215.42 /215.00 B:60.05 /60.00 P:35.08 /0.00 @:43 B@:0
X:133.37 Y:128.89 Z:13.20 E:0.00 Count X:13147 Y:26564 Z:5318
X:136.28 Y:129.85 Z:13.20 E:0.00 Count X:13360 Y:25832 Z:5314
X:125.27 Y:132.24 Z:13.20 E:0.00 Count X:13623 Y:25972 Z:5312
 T:215.21 /215.00 B:60.05 /60.00 P:35.00 /0.00 @:43 B@:87
X:118.95 Y:126.01 Z:13.20 E:0.00 Count X:11895 Y:25203 Z:5326
 T:214.58 /215.00 B:60.14 /60.00 P:35.08 /0.00 @:61 B@:0
 T:214.79 /215.00 B:60.28 /60.00 P:34.90 /0.00 @:19 B@:0
 T:214.37 /215.00 B:60.28 /60.00 P:35.00 /0.00 @:55 B@:38
 T:214.37 /215.00 B:60.37 /60.00 P:34.90 /0.00 @:47 B@:127
 T:214.58 /215.00 B:60.42 /60.00 P:35.00 /0.00 @:45 B@:61
X:111.12 Y:128.68 Z:13.20 E:0.00 Count X:11449 Y:25767 Z:5331
 T:214.58 /215.00 B:60.46 /60.00 P:34.90 /0.00 @:45 B@:55
X:113.06 Y:119.10 Z:13.20 E:0.00 Count X:11597 Y:23803 Z:5324
X:110.46 Y:121.69 Z:13.20 E:0.00 Count X:11402 Y:23740 Z:5325
X:108.63 Y:130.32 Z:13.20 E:0.00 Count X:11067 Y:24180 Z:5330
X:113.00 Y:133.28 Z:13.20 E:0.00 Count X:10884 Y:25932 Z:5337
 T:214.58 /215.00 B:60.46 /60.00 P:34.90 /0.00 @:48 B@:66
X:123.79 Y:130.38 Z:13.20 E:0.00 Count X:12399 Y:25557 Z:5322
X:117.43 Y:128.27 Z:13.20 E:0.00 Count X:11805 Y:25653 Z:5328
X:118.24 Y:123.79 Z:13.20 E:0.00 Count X:11950 Y:25126 Z:5325
 T:214.79 /215.00 B:60.46 /60.00 P:34.90 /0.00 @:49 B@:88
X:116.93 Y:119.42 Z:13.20 E:0.00 Count X:11506 Y:24670 Z:5328
 T:215.21 /215.00 B:60.51 /60.00 P:34.90 /0.00 @:52 B@:0
X:113.13 Y:114.28 Z:13.20 E:0.00 Count X:11313 Y:22856 Z:5323
X:117.07 Y:89.74 Z:13.20 E:0.00 Count X:11683 Y:18325 Z:5313
X:113.56 Y:86.88 Z:13.20 E:0.00 Count X:11300 Y:17347 Z:5316
X:119.44 Y:85.67 Z:13.20 E:0.00 Count X:11946 Y:17111 Z:5308
 T:215.21 /215.00 B:60.42 /60.00 P:34.90 /0.00 @:41 B@:61
X:118.18 Y:77.85 Z:13.20 E:0.00 Count X:11818 Y:15569 Z:5307
 T:215.21 /215.00 B:60.37 /60.00 P:34.90 /0.00 @:31 B@:40
 T:215.42 /215.00 B:60.37 /60.00 P:34.79 /0.00 @:25 B@:0
X:114.15 Y:91.58 Z:13.20 E:0.00 Count X:11415 Y:18316 Z:5315
 T:215.21 /215.00 B:60.32 /60.00 P:34.90 /0.00 @:61 B@:53
X:118.11 Y:89.83 Z:13.20 E:0.00 Count X:11811 Y:17966 Z:5311
 T:215.00 /215.00 B:60.32 /60.00 P:34.90 /0.00 @:55 B@:32
X:116.14 Y:82.62 Z:13.20 E:0.00 Count X:11577 Y:16578 Z:5312
 T:215.00 /215.00 B:60.32 /60.00 P:34.90 /0.00 @:52 B@:26
X:109.32 Y:85.83 Z:13.20 E:0.00 Count X:10932 Y:17166 Z:5319
X:110.64 Y:89.76 Z:13.20 E:0.00 Count X:11064 Y:17953 Z:5318
X:108.48 Y:92.18 Z:13.20 E:0.00 Count X:10848 Y:18437 Z:5321
 T:215.00 /215.00 B:60.37 /60.00 P:34.90 /0.00 @:50 B@:0
X:103.51 Y:92.70 Z:13.20 E:0.00 Count X:10351 Y:18540 Z:5326
X:99.16 Y:96.23 Z:13.20 E:0.00 Count X:9953 Y:19192 Z:5330
 T:215.21 /215.00 B:60.28 /60.00 P:34.90 /0.00 @:51 B@:46
 T:215.83 /215.00 B:60.28 /60.00 P:34.90 /0.00 @:46 B@:60
 T:216.04 /215.00 B:60.28 /60.00 P:34.90 /0.00 @:42 B@:0
 T:216.04 /215.00 B:60.23 /60.00 P:34.90 /0.00 @:37 B@:6
 T:215.83 /215.00 B:60.14 /60.00 P:34.90 /0.00 @:41 B@:46
 T:215.83 /215.00 B:60.14 /60.00 P:34.90 /0.00 @:36 B@:22
 T:215.62 /215.00 B:60.05 /60.00 P:34.90 /0.00 @:45 B@:95
 T:215.42 /215.00 B:60.05 /60.00 P:34.90 /0.00 @:51 B@:104
 T:215.42 /215.00 B:60.09 /60.00 P:34.90 /0.00 @:38 B@:31
 T:215.42 /215.00 B:60.09 /60.00 P:34.90 /0.00 @:38 B@:78
 T:215.00 /215.00 B:60.09 /60.00 P:34.90 /0.00 @:48 B@:31
 T:214.58 /215.00 B:60.09 /60.00 P:35.00 /0.00 @:65 B@:69
 T:214.79 /215.00 B:60.09 /60.00 P:35.00 /0.00 @:40 B@:24
 T:214.79 /215.00 B:60.09 /60.00 P:35.00 /0.00 @:40 B@:18
 T:215.00 /215.00 B:60.09 /60.00 P:35.00 /0.00 @:36 B@:13
...............

digant73, 2021-04-27:
and then did you get the freeze?

sinole1987, 2021-04-27:
it is kinda sad I can reliably get printer to freeze on long prints. time to toss this screen into the trash.

sinole1987, 2021-04-27:
> and then did you get the freeze?

yes it is just reporting temp as you can see

sinole1987, 2021-04-27:
screen was functional this time until I pressed pause and it immediately froze. for the first time I unplugged the screen as printer was on and as soon as I unplugged it performed the pause, but when i plugged it back in there was nothing . no light on screen, until I pressed reset. I guess marlin stops responding to printer at some point.

sinole1987, 2021-04-27:
I removed m75 from my Start Gcode, lets see if we get a freeze.

bigtreetech, 2021-04-28:
> screen was functional this time until I pressed pause and it immediately froze. for the first time I unplugged the screen as printer was on and as soon as I unplugged it performed the pause, but [...]

Hello, according to the description of here, the TFT can still operate before clicking pause, and it freezes after clicking. This phenomenon is caused by the interruption of communication between TFT and marlin. After clicking pause, the screen will wait for Marlin to perform the pause action before TFT allows the customer to carry out the next operation, so the TFT looks frozen. At this time, you can insert/remove the SD card on the TFT to detect whether there is a prompt insert/remove of the SD card in the title bar. If there is a prompt, it means that the firmware on the TFT is still running, just waiting for the response from Marlin (because the communication was interrupted, and the TFT will wait all the time), If there is no prompt, the TFT is really frozen.

The causes of communication interruption are mostly poor cable contact or nearby electrical noise. Please make sure that the cable connecting the motherboard and TFT is reliable, And it is best not to pass the cable near the power supply, because the noise near the power supply is the biggest.

According to the issues reported at present, there are many different phenomena and causes of freezing, So we need to distinguish between screen freezing or communication interruption.

sinole1987, 2021-04-28:
it does look like communication interruption. since screen comes with short cable I have crimped my own cable but I am sure it is very well crimped and it is passing near a lot of cable witch is unavoidable near mainboard, but not PSU. only thing i can do is making shielded cable.

sinole1987, 2021-04-28:
if I cut a quality USB cable and crimp it for TFT would it make a diffrense? those are shielded .

digant73, 2021-04-28:
you could also use a ferrite bead (https://www.amazon.co.uk/Suppressor-High-frequency-Anti-interference-Loudspeaker-diameter/dp/B07P3GDMGS/ref=sr_1_4?dchild=1&keywords=ferrite+bead&qid=1619599651&sr=8-4) on one side of the cable (use mainboard side) and connect the shield on both side of the cable to the GND pin (on mainboard and TFT). With that I suppressed EMI on my printer. Without them it was impossible to suppress EMI. Keep the cable as much as possible attached to the chassis.

sinole1987, 2021-04-28:
> you could also use a ferrite bead (https://www.amazon.co.uk/Suppressor-High-frequency-Anti-interference-Loudspeaker-diameter/dp/B07P3GDMGS/ref=sr_1_4?dchild=1&keywords=ferrite+bead&qid=1619599651&sr=8-4) on one side of the cable (use mainboard side) and connect the shield on both sides of the cable to any free GND pin (on mainboard and TFT). With that I suppressed EMI on my printer. Without them it was impossible to suppress EMI. Keep the cable as much as possible attached to the chassis.

I dont think i could put those on my printer , you goto twist the cable around them two . but damn motors are making all the EMI you need, they have inductance rating, and my tft line has to pass near a Z motor and its cable. you just cant avoid it a printer.

digant73, 2021-04-28:
ok for the cable path, but apply the solution I gave you (ferrite bead and shield to GND pin on both sides). it should be enough

sinole1987, 2021-04-28:
> ok for the cable path, but apply the solution I gave you (ferrite bead and shield to GND pin on both sides). it should be enough

thank you. Grounding shield makes a lot of sense to me, you translate magnetic noise into current and send it to ground. but shield is a aluminum foil how to you crimp it to ground? do you have a picture example?

digant73, 2021-04-28:
Simply twist the shield with a wire and use a tape or heat shrink to fix them. Then crinp a dupont connector to be connected to the GND pin. Do that on both sides of the shielded cable

sinole1987, 2021-04-28:
[image]
do you like this arrangement? i would have to run a reset line outside which doesn't really matter

sinole1987, 2021-04-29:
I ended up with this cable , seems like all of the cables are 26awg so it doesn't have the beefy red cable but its 1 bunch. do you think 26awg is enough to handle power for tft? it does turn on but I don't know how many amps a tft 43 pull.
[image]

digant73, 2021-04-29:
yes the cable is OK. AWG26 is also OK. You can try with that cable if you still have problems try to put a ferrite bead on one side (mainboard side)

sinole1987, 2021-04-29:
its been going strong for 4H:25m now. I also increased baudrate to 250000 I had no problem when I was at that speed.

sinole1987, 2021-04-29:
failed again , ferrite choke on motherboard side is the last chance. BTW I have a buck convertor inside motherboard box could hat be the issue?

digant73, 2021-04-29:
try with the ferrite bead

kisslorand, 2021-04-29:
>BTW I have a buck convertor inside motherboard box could hat be the issue?

Can you shield that one?

sinole1987, 2021-04-29:
> > BTW I have a buck convertor inside motherboard box could hat be the issue?
> 
> Can you shield that one?

well I don't wana put aluminum foil in my motherboard tray. but I guess it is possible. as a last solution I will put some loud fans n the printer and take buck convertor out. If only noctua Had 24v fan .

sinole1987, 2021-04-29:
btw I feel its crazy that only I have this issue. I know it not hardware problem meaning board or screen, because I already replaced them once.

oldman4U, 2021-04-29:
Lower the baud rate to something very low. Lower than 115k

sinole1987, 2021-04-30:
> […](#)

I tried all of that speed thing and buffer thing from the last discussion. I have a strong feeling about this tiny buck convertors being the problem. I have been researching about them. they produce high frequency EMI. I took them out, if it solves the problem, I guess we are rocking Chinese fan. It feels bad to take out 60$ worth of fan and put in $6 fan in the printer. 😢

oldman4U, 2021-04-30:
You can connect 2 noctua fans seriell to one 24V source.

[25 more replies]

closed 2021-08-04 by stale[bot]