bigtreetech/BIGTREETECH-TouchScreenFirmware #1750: TFT35 FW mod to run as a GRBL CNC offline controller?

Sprint8, 2021-03-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

MKS has firmware options to change their TFT35 touchscreen (and other sizes) to run  as an offline controller for GRBL CNC and Laser machines but they don't seem too inclined to release their FW into open source or actively fix the bugs.  

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I have the BTT TFT35, TFT50 and TFT70 screens and would love to convert the TFT35 to run my Benbox 1310 instead of getting one the 1" screen offline controllers which doesn't have touch capability.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
The MKS TFT35 is the only one that I know of that has some capability to do this but as I stated above they're not inclined to release their code into opensource and there isn't much development or motion into fixing issues. My guess is that there might only be 1 person who's doing it so time and priority would probably be the factor but anything more is just further speculation.

**Additional context**
Add any other context or screenshots about the feature request here.
There's a few of us who have the MKS TFT35 who are willing to help them with language typos, code revisions etc but are also looking for a cool, attractive solution to the tiny little button operated offline controllers that's currently found elsewhere. This could be an opportunity for BTT to branch out a bit and possibly secure another segment of an open market that doesn't have a working solution for a touchscreen GRBL controller (that some of us are aware of)

We have nothing against MKS, I'm just noticing that BTT has a good user community and following regarding their firmware solutions and we suspect that some of you who are very skilled programmers could whittle down the 3D printer code  in no time to have a working solution. I personally haven't written code since College so it'd be a huge curve for me to get up to speed just to parse the code (and wrap my head around all the nuances). 

At the very least I think it's something that BTT and this team should look into and consider.
Thanks for taking the time to hear me out.

Msq001, 2021-05-24:
remove bot

wiring80, 2021-12-02:
I agreed. I buyed an mks tft24 too, cause my hope was to control grbl nicely offline. Bad thought, mks is not Community friendly, no source, tooooo many bugs, no support. For me it would be pretty to have at least Realtime axis coordinates. The thing is useless...

I would love to see an direction to grbl from Bigtreetech

xPew, 2022-01-04:
+1 for that, but from an other side. I would be also happy for a CNC friendly Ui redesing (as a separate option like the added vertical UI). I will use it with a RepRapFirmware controlled CNC rather than GRBL, so that adds an other layer of difficulty, but mainly just different commands (and Marlin+RRF runs side by side quite nice, it can be done witg GRBL and RRF i hope)
But first a CNC UI to be able to start working on the firmware and coding.

HaasNL2, 2022-01-21:
I am using it for remote, the TFT cable is easy to extent. I there is also a firmware for the MPCNC but it s outdated
https://github.com/BlomsD/MPCNC-TFT35-V2.0
[image]

uronour, 2022-12-28:
> I am using it for remote, the TFT cable is easy to extent. There is also a firmware for the MPCNC but it s outdated https://github.com/BlomsD/MPCNC-TFT35-V2.0 I am really interesting to update mpcnc version but I find it difficult to modify anything. (every time I modify anything I break everything) [image]

Would you be willing to share your wiring? did you the 2 cables or the spi?
i have mksdlc32 and 2 spare bigtreetech tft35's this would be a god send and bonus to use one of these with either the grbl v1.1h i think it is rattm board or of course my mks dlc32 v2.1
i'm going to rib firmware's for both screens and try to port the mks firmware over if i can, if it's doable i have zero ideas as this would be new and 1st for me. surely the pins would be changed and then firmware edits for $$ commands etc? i have no idea any help would be greatly appreciated

jsagot, 2023-12-18:
I have some spare tft (btt & mks) and CNCs running with grbl-mega-5x, i would really be interested to work on that with you @xPew

maxmenGMK, 2024-01-28:
I have CNC3018 and BTT TFT3.5 V3.0 also and would be nice to connect it together.

MisterMakerNL, 2024-01-28:
Mine already died so don't need it anymore. Now I just use a computer.

closed 2024-04-05 by github-actions[bot]