bigtreetech/BIGTREETECH-TouchScreenFirmware #1751: [BUG] 'Back' button sometimes go to main interface in Heat/Preheat menus

GregSKR, 2021-03-20:
Maybe I'wrong, but looks like the 'Back' button has sometimes a strange behavior...

eg (1):
-Go to: Menu > Movement > Bed Level > ABL
-Touch 'Preheat' button then 'Back'

eg (2):
-Go to: Menu > Extrude
-Touch 'Heat' button then 'Back'

Sometimes the 'Back' button goes to the previous menu (as it should), but sometimes (and always if you enter a temperature) it return to the main interface...

digant73, 2021-03-20:
it seems the SMART_HOME feature enabled by default on Configuration.h.
A long press on Back button to move quickly on main screen.
Try to disabe it and you should solve it

kisslorand, 2021-03-20:
@GregSKR 
When you short-press the Back button what sound do you hear? Everytime the same or sometimes a "click" sound and other times a double sound (like "tu-di")?

oldman4U, 2021-03-20:
Welcome back Greg. Seems you missed some new features 😁

GregSKR, 2021-03-20:
Thanks for your anwers!

@digant73 : thanks for the information, disabling SMART_HOME gives normal 'back' fonction everywhere
@kisslorand : the sound is 2 short bips "do"..."di" when returning to the main screen but different that the smart home sound.
@oldman4U : yes, I didn't make an update for a long time (was waiting for a stable version) and missed a lot of things 😁 

See the video: first press is a long press (smart home), all others are short press (touch_sound disabled in my config).
Sometimes, the back button goes back, but often it goes to the main interface (and sounds different than smart home).
I discovered that the problem is in all menus!

https://www.youtube.com/watch?v=SdptFecX5WA

My screen is a TFT35 v2.0
Will disable SMART_HOME (but it was a nice functionality) if there no other solution...

bttguy, 2021-03-21:
@GregSKR Maybe update the fw or wait a few days for the sable version. I tried to force the issue as seen in the video, but for me it works 100% as intended using a few days old fw. Which version do you have?

kisslorand, 2021-03-21:
@bttguy 
It doesn't matter which version he has, the smart_home hasn't been changed since it's been merged.
Looking at the video seems to me it is some kind of timer issue. The long "do"..."di" sound is a hint that the TFT is very busy with something (or the same timer issue), it cannot execute the two beep commands one after the other fast enough.
I cannot reproduce it. :(

GregSKR, 2021-03-21:
I did a few more tests...
I changed the long press delay to 3 seconds
`#define LONG_TOUCH 3000`
What I can see now is that when a menu is displayed, a short press on 'Back' before 3 seconds gives normal back function, but if I wait  more than 3 seconds and short press 'Back', I got a smart home with 2 bips...

My screen is a TFT35 v2.0, so no Marlin mode, and no encoder knob, maybe something is different here from other screens?

I'm going to disable smart_home, it's buggy for my TFT.
Thanks for your help! 😉

oldman4U, 2021-03-21:
So the value defines how long a return press is a previous page press instead of defining how long the button has to be pressed.

@kisslorand Would it make sense to see if it behaves the same on a BTT TFT28?

kisslorand, 2021-03-21:
@GregSKR
Can you please reactivate touch sound and check if you hear it when it is not supposed to sound?

GregSKR, 2021-03-21:
@kisslorand
touch sound reactivated: yes, I hear it when short press goes to main menu (got smart home sound too).
And doing a long press: touch sound when button is pressed, smart home sound when I release it.

kisslorand, 2021-03-21:
I am working on a fix now. Is it OK for you if I upload a file and compile the FW with that file?
I cannot reproduce the issue so I rely on you. Until a fix is found it is pointless to make a PR for it.

GregSKR, 2021-03-21:
@kisslorand 
Thanks! Yes I'm OK... 😃

kisslorand, 2021-03-21:
Here's the file:

[menu.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6178048/menu.zip)

GregSKR, 2021-03-21:
Just tested... but the problem is still here, I see no difference 😢

kisslorand, 2021-03-21:
I cannot find any software issue. The way I rewrote the procedure, it cannot go back to main menu unless the touch is long-pressed. I suspect a HW issue. Long press on any other icon triggers the Marlin/Touch selection menu?

oldman4U, 2021-03-21:
TFT35 v2 has no encoder and can not switch to Marlin
[image]

GregSKR, 2021-03-21:
Like @oldman4U just said, no Marlin mode here, so long press on other icons do the same than short press...

kisslorand, 2021-03-21:
I have a MKS TFT28 without encoder and it enters into Marlin/Touch selection menu.

GregSKR, 2021-03-21:
This is not related to the encoder, but the Marlin emulator.
In my case, `HAS_EMULATOR` is not def

kisslorand, 2021-03-21:
Understood, I just wanted to see how does your TFT handles long press in another situation.
Would you please enable the emulator just to see what happens?

GregSKR, 2021-03-21:
I added `#define HAS_EMULATOR` in **pin_TFT35_V2_0.h**
All is working as it should now!!!
Long press on the screen trigger the menu "Marli mode / Touch mode", long press on back trigger the smart home and short press on back always go back...

Try to comment `#define HAS_EMULATOR` in your firmware, i'm sure you will reproduce the bug !

kisslorand, 2021-03-21:
Yes! I can reproduce it now. Back to fixing it...

kisslorand, 2021-03-21:
@GregSKR 
Please test this file:
[menu.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6178330/menu.zip)

GregSKR, 2021-03-21:
Yesssss! You did it!
All is fine now 😃 

Thanks a lot, @kisslorand 👍🏼

kisslorand, 2021-03-21:
No, I thank you! You've been more than helpful.
Let's make a nice PR of it...

GregSKR, 2021-03-21:
Nice! Issue closed 😉

PR #1760 (2021-03-23): Back-button-held fix
### Description

Fixes a bug where Back button is not really "smart" if SMART_HOME is enabled and Marlin emulator is disabled.

The Back-button-held procedure has been fixed and reworked also RAM and flash usage has been reduced a tiny bit.

Fixes #1751.
closed 2021-03-21 by GregSKR