bigtreetech/BIGTREETECH-TouchScreenFirmware #1756: Manual bed leveling enchance [FR]

Vybramo, 2021-03-21:
Also please add :

## Z Position (Manual Leveling)
# Lower z axis to this position after reaching a leveling point.
#   Unit: [position in mm]
#   Value range: [min: 0, max: 2000]
level_z_pos:0.1

to menu with :

## Leveling Edge Distance (Manual Leveling)
# Inset distance from bed edges. This distance is added to minimum X & Y bed coordinates and
# subtracted from maximum X & Y bed coordinates to calculate manual leveling points.
#   Unit: [distance in mm]
#   Value range: [min: 0, max: 2000]
level_edge_distance:30

Personally I measure paper/any other think which I use for level and set there this value. Its greatly help with first layer cause you don`t need reconfigure your slicer. Default value its 0.2mm but paper its 0.08-0.1mm thick

oldman4U, 2021-03-22:
Hi.

Settings are provided to be changed so they fit the personal needs.

Invert axis is available in config.ini.

Hope this helps

Vybramo, 2021-03-22:
> Hi.
> 
> Settings are provided to be changed so they fit the personal needs.
> 
> Invert axis is available in config.ini.
> 
> Hope this helps
I understant. Few fw versions inverting axis was avaible in screen menu. This is FR for add these settings on screen menu to not every time if I mesure paper for level my bed I need to flash new config.ini for change Z position during leveling just change value in leveling menu.

oldman4U, 2021-03-22:
Yes. Some functions have been removed to save memory which finally allows new features o be implemented.

How many times do you change the paper/filler gauge you use for levelling, which results in the need to change this value?

oldman4U, 2021-03-24:
Hi.

Could you please let me know how many times you level your bed and how many times you use a different paper/filler gauge for this.

Thank you

Vybramo, 2021-03-24:
> Hi.
> 
> Could you please let me know how many times you level your bed and how many times you use a different paper/filler gauge for this.
> 
> Thank you

I do it every time before long print, so now 3 time per week. So its time to learn how to modify firmware by my self

oldman4U, 2021-03-24:
Ok. But I guess you always use the same paper or filler gauge to define the
z offset, right? So you can easily adjust this setting to your needs and
use it from this time on. Like all other users which do not like a certain
setting have to do it.

Editing the config.ini is finally editing a text file.

oldman4U, 2021-03-27:
Please help the community and close the ticket once you do not need it anymore. Thank you

Vybramo, 2021-03-31:
If its possible just add options in leveling menu where its value for distance also add in that menu option for set also level z position. 
Im not skilled in firmware modification but I can read it and modify few thinks. 

Not all time paper its same. for example can be 0.1-0.08mm thick and filler gauge its little hard to use.

oldman4U, 2021-03-31:
It is in config.ini. So all you have to do is to edit a textfile.

It is not possible to add all parameters to the settings you can change on
the TFT. There is not enough memory.

closed 2021-04-28 by bigtreetech