bigtreetech/BIGTREETECH-TouchScreenFirmware #1761: [FR] Sound plays when preheat temp is met

zcohenld, 2021-03-22:
When preheating the board in any of the menus such as the auto leveling menu there is no way to know when preheat is completed other than exiting to the main screen. Would be nice to have a way to know the target temperature was reached.

Either a temperature readout on the preheat menu or a sound that plays to indicate that the temperature has been reached would be really useful.

kisslorand, 2021-03-22:
1. Menu -> Settings -> Feature -> Persistent Status info
2. Menu -> Settings -> Screen -> Sound -> Heater notification

> When preheating the board in any of the menus such as the auto leveling menu there is no way to know when preheat is completed other than exiting to the main screen. Would be nice to have a way to [...]

Let me edit this for you:

" When preheating the board in any of the menus such as the auto leveling menu there **_are two ways_**  to know when preheat is completed ~~other than exiting to the main screen~~. **_It is_** nice to have **_more than one_** way to know the target temperature was reached.
 
 Either a temperature readout on the preheat menu or a sound that plays to indicate that the temperature has been reached **_are_** really useful."

zcohenld, 2021-03-22:
Wow. I can't believe I missed that in the menu. Thank you for the information!

closed 2021-03-22 by zcohenld