bigtreetech/BIGTREETECH-TouchScreenFirmware #1764: Neopixel header

sinole1987, 2021-03-22:
is there a way to control nonpixel header on TFT screens. it seems that it is in parallel to knob led and it can control only 2 LEDs like the knob. can you increase number of LEDs.

bigtreetech, 2021-03-23:
Yes, it is in parallel to knob led. We'll add the function of modifying LED quantity in config.ini. The current firmware only allows you to modify the number of LED in source code  and then compile and update. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/318c98d7a4360cef0cf1696b51d95fadcf87a198/TFT/src/User/Variants/pin_TFT35_E3_V3_0.h#L13
For example, TFT35 E3 V3.0 needs to modify the number of `NEOPIXEL_PIXELS` in `TFT/src/User/Variants/pin_TFT35_E3_V3_0.h` from 4 to the number you want

oldman4U, 2021-04-16:
Stupid question. Where are the additional leds connected to?

sinole1987, 2021-04-16:
I don't think  its on every Variant of the board. here are some examples:
[image]

oldman4U, 2021-04-16:
Thank you. So it is a dedicated port. Time to activate one of the TFTs which has it 😁

oldman4U, 2021-04-17:
In the PR also the TFT35 E3 v3 is mentioned to be supported. But this screen does not have a dedicated port !? I am 🤪

sinole1987, 2021-04-17:
my TFT43 Screen have so many ports that i have no idea what any of them do. I have UART ports near mounting holes, it was a nightmare designing a mount for this screen.

oldman4U, 2021-04-18:
The perfect reason to buy a TFT70!!!😁

PR #1844 (2021-04-19): Neopixel LED number config
### Description
Now we can set the  number of LED(neopixel)  in the `config.ini` file

### Benefits

### Related Issues
fix #1764
closed 2021-04-19 by bigtreetech