bigtreetech/BIGTREETECH-TouchScreenFirmware #1765: Continuous frequent sending of M220/M221 commands from the main menu.

ochkariky, 2021-03-22:
### Steps to reproduce
The problem is the frequent sending of M220 commands in the main status menu(which appears after loading).
Pronterface was connected when the TFT28 was on immediately after booting. 
In the Pronterface window, I noticed the results of requests for speed and flow (M220-M221).
For example TFT28 together with Smoothieware (I didn't press anything) gives this to Pronterface:
Speed ​​factor at 100.00%
Flow rate at 100.00%
Speed ​​factor at 100.00%
Flow rate at 100.00%
Speed ​​factor at 100.00%
Flow rate at 100.00%
....etc.

 Most likely this happens through speedQuery () about once every two seconds (TOGGLE_TIME or UPDATE_TOOL_TIME)
possibly from the function code: toggleInfo () or toggleTool ().

I am using a board with Smoothieware, it uses a different format: I have [not completed this my pull request](https://github.com/ochkariky/BIGTREETECH-TouchScreenFirmware/commit/38a805735764814bb658b4285c67ba5a37aacc8c). But it seems to me that the behavior should not be different in marlin?

Do you need such a frequent polling of the speed and flow of the M220-221? Or is it a mistake or is it for what it is necessary? Can anyone confirm or deny?

### Hardware Variant
TFT28
Smoothieware

bigtreetech, 2021-03-23:
Because the motherboard firmware such as Marlin will not actively report the change of Flow/Speed, So the TFT needs to constantly send M220/M221 to get the newest value in status / printint menu.(Because these two menu need display Flow/Speed value)

ochkariky, 2021-03-23:
oh, I got it!
thank!

closed 2021-03-23 by ochkariky