bigtreetech/BIGTREETECH-TouchScreenFirmware #1766: [BUG] tft 35 for ender fw blocking issue on pause/restart

EnzoCastellaneta, 2021-03-23:
### Description

Updated to last fw, adding red icons and italian language, the system when paused under a job, do not resume showing continuosly a window in asking "paused for user" and looping continuosly - no way to close it and start over the job.

Posted also on Reddit and several other users have the same problem
### Steps to reproduce

1. last firmware updated showing red icons
2. start a job
3. pause it. Then a window in saying "paused for user" pop ups and even if closed (pressing x) re-pops up asking the same thing. No way to restart the job. Cutting off current is the only way to restart - loosing the job.

**Expected behavior**
restart the job once I click the window and the button restart under that window

**Actual behavior**
I try to not stop and it works... but if ever I do need to pause... I will loose the job

### Hardware Variant
35 for ender
### TFT Firmware Version & Main Board Firmware details
Last one BIGTREE_TFT35_V3.0_E3.26.x and btt skr v2 last firmware too (compiled one with 3dtouch and z sensor removed)

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-03-23:
Read in config.ini which parameters in Marlin Firmware have to be activated and change your Marlin accordingly. Make sure to use nightly build Marlin otherwise a Marlin bug in STM32 will cause a freeze or crash when pressing Purge More. Once it works let all the other users on Reddit know and close this ticket. 

Thank you

EnzoCastellaneta, 2021-03-24:
hi,out of home, back on late friday - I will do my best yo support and give fbkInviato dal mio Galaxy
-------- Messaggio originale --------Da: oldman4U ***@***.***> Data: 24/03/21  17:31  (GMT+01:00) A: bigtreetech/BIGTREETECH-TouchScreenFirmware ***@***.***> Cc: EnzoCastellaneta ***@***.***>, Author ***@***.***> Oggetto: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] [BUG] tft 35 for ender fw blocking issue on pause/restart (#1766) 
Any news?

oldman4U, 2021-03-24:
Perfetto.

oldman4U, 2021-03-27:
And now?

digant73, 2021-03-29:
ciao Enzo, hai avuto modo di verficare? facci sapere per favore

oldman4U, 2021-03-30:
He does not want to eat, even you put the spoon in his mouth...

Maybe someone can tell the lost souls at Reddit to read the manual...

Let the bot do the job here

EnzoCastellaneta, 2021-03-30:
wise people listen others screaming in the lone silence of daily life.THE QUESTION IS: are you listening or screaming?Have a good day.E.Inviato dal mio Galaxy
-------- Messaggio originale --------Da: oldman4U ***@***.***> Data: 30/03/21  20:53  (GMT+01:00) A: bigtreetech/BIGTREETECH-TouchScreenFirmware ***@***.***> Cc: EnzoCastellaneta ***@***.***>, Author ***@***.***> Oggetto: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] [BUG] tft 35 for ender fw blocking issue on pause/restart (#1766) 
He does not want to eat, even you put the spoon in his mouth...
Maybe someone can tell the lost souls at Reddit to read the manual...
Let the bot do the job here

oldman4U, 2021-03-31:
And there are others screaming and crying because they do not read what is written.

Please close this ticket as it is unrelated to this repository and steals the valuable time of us.

Thank you

EnzoCastellaneta, 2021-04-01:
Hi, just updated with last nightly build fw and other files. 
Checked config.ini.
Updated, saved, restarted - started printing - paused ... still same problem

[image]

I add photo of it and details of what updated.
I also used 3h ago firmware update BIGTREE_TFT35_V3.0_E3.26.x.bin
In the following the config.ini

Pls. remember I add italian language file (updated) and red icons theme (same)

#--------------------------------------------------------------------
#
# DEFAULT CONFIG FILE FOR BigTreeTech TFT CONTROLLERS
#
#--------------------------------------------------------------------

#--------------------------------------------------------------------
# Supported TFT Variants
#--------------------------------------------------------------------
#
# BIGTREE_TFT35_V1_0    / BIGTREE_TFT35_V1_1 / BIGTREE_TFT35_V1_2
# BIGTREE_TFT35_V2_0    / BIGTREE_TFT35_V3_0 / BIGTREE_TFT35_E3_V3_0
# BIGTREE_TFT35_B1_V3_0 / BIGTREE_TFT43_V3_0 / BIGTREE_TFT50_V3_0
# BIGTREE_TFT70_V3_0    / BIGTREE_TFT28_V1_0 / BIGTREE_TFT28_V3_0
# BIGTREE_TFT24_V1_1    / MKS_32_V1_4        / MKS_28_V1_0
#
# Firmware source: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

#--------------------------------------------------------------------
# Supported Marlin Firmware Versions
#--------------------------------------------------------------------
#
# Minimum Marlin firmware version: 2.0.5.4
# Distribution date: 2020-05-12
# Firmware source: https://github.com/MarlinFirmware/Marlin/releases
#
# In order the TFT firmware is able to provide all of its functionalities/features,
# ensure that the following options are enabled in Marlin firmware.
#
# General options:
#   EEPROM_SETTINGS (in Configuration.h)
#   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
#   M115_GEOMETRY_REPORT (in Configuration_adv.h)
#   M114_DETAIL (in Configuration_adv.h)
#   REPORT_FAN_CHANGE (in Configuration_adv.h)
#
# Options to support printing from onboard SD:
#   SDSUPPORT (in Configuration.h)
#   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)
#
# Options to support dialog with host:
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#   HOST_PROMPT_SUPPORT (in Configuration_adv.h)
#
# Options to support M600 with host & (Un)Load menu:
#   Options to support dialog with host (as pre requisite)
#   NOZZLE_PARK_FEATURE (in Configuration.h)
#   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
#   PARK_HEAD_ON_PAUSE (in Configuration_adv.h)
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)
#
# Options to support Babystep menu:
#   BABYSTEPPING (in Configuration_adv.h)
#
[... 576 lines cut ...]
#   Options: [enable: 1, disable: 0]
start_gcode_enabled:0

#### Default End Gcode Status
#   Options: [enable: 1, disable: 0]
end_gcode_enabled:0

#### Default Cancel Gcode Status
#   Options: [enable: 1, disable: 0]
cancel_gcode_enabled:0

#### Start Gcode
# This gcode will runs before starting a print if `start_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
start_gcode:G28 XY R20\n

#### End Gcode
# This gcode will runs after a print is completed if `end_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
end_gcode:M104 S0\nM140 S0\nM107\nM18\n

#### Cancel Gcode
# This gcode will runs when a print is canceled if `cancel_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\nM107\nM18\n

digant73, 2021-04-01:
@EnzoCastellaneta if all the features reported at the beginning of the config.ini file are enabled in Marlin it could be a problem in Marlin. I add a problem with the pause when I enabled UBL in Marlin. Basically the pause was not working as expected (the nozzle was moved on the park area but then it was moved on top of the print, printing just for few seconds before pausing). Also the M600 was not working (when the dialog asking to "purge more " or "continue" was displayed the print was alsways automatically resumed without waiting for the choice from the user).
If you are using UBL in Marlin try to switch to Bilinear BL. Less resources will be used. I suspect the problems using UBL are due to the resource limit reached in Marlin (on my printer it was 93% and 90% for mem and flash)

oldman4U, 2021-04-10:
Any news? Would be nice if you could answer digant or me.

Otherwise, please help the community and close the ticket once you do not need it anymore. Thank you

EnzoCastellaneta, 2021-04-10:
tried just yesterday but not able to compile something went wrong with classes.Could you compile one with the fix suggested? E3Pro skr mini 2.0 bltouch used for z axis endpoint + tft 35 e3 last fw uploaded.let me know.EInviato dal mio Galaxy
-------- Messaggio originale --------Da: oldman4U ***@***.***> Data: 10/04/21  08:35  (GMT+01:00) A: bigtreetech/BIGTREETECH-TouchScreenFirmware ***@***.***> Cc: EnzoCastellaneta ***@***.***>, Mention ***@***.***> Oggetto: Re: [bigtreetech/BIGTREETECH-TouchScreenFirmware] [BUG] tft 35 for ender fw blocking issue on pause/restart (#1766) 
Any news? Would be nice if you could answer digant or me.
Otherwise, please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2021-04-11:
Please. This is neither a Marlin nor a VSC support page and also not a user group. This repository is about the firmware of TFT displays from BTT only. Your problem is not related to the TFT firmware, so please help the community and close this ticket. Thank you

oldman4U, 2021-04-18:
@Msq001 Please see my last comment and close this ticket. Thank you

closed 2021-04-18 by EnzoCastellaneta