bigtreetech/BIGTREETECH-TouchScreenFirmware #1770: [BUG] TFT Mode Manual Mesh Bed Leveling does not Home first

sarvenn, 2021-03-25:
I use latest fw on my tft 35 e3 v3. I use manual mesh bed leveling
I updated my marlin fw, so I needed to do the manual mesh bed leveling after eeprom init.
My hotend was in the center of bed.
I started the bed leveling sequence in tft mode.
My expectation was a homing at first then go to P1, P2, ...etc. It did not do homing, it assumed that it is at home position and move to P1 (not the actual P1), just move a little right and back.
It should do the homing first, then should go to the real P1 point.

I cancelled the sequence. I did homing then started the mesh bed leveling sequence.

digant73, 2021-03-25:
G29 P1 should also home before starting the probing. At least that was the behavior in Marlin till now

sarvenn, 2021-03-25:
Yeah, that's why I was surprised and wanted to report this unexpected behaviour. Would there be any relation with the TFT fw side or is it a concern to be cascaded to marlin side?

digant73, 2021-03-25:
it is on Marlin. the TFT sends G29 S1 to start MBL

closed 2021-03-25 by sarvenn