bigtreetech/BIGTREETECH-TouchScreenFirmware #1771: Translation Swedish[FR] (feature request title)

ShRTee, 2021-03-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ShRTee, 2021-03-26:
I have translated the Swedish language file and wonder if there is any requests for this ?
[language_se.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6210472/language_se.zip)

oldman4U, 2021-03-26:
Make a PR in case everything is translated and you are willing to maintain it.

radek8, 2021-04-18:
@ShRTee will you be willing to keep your own language up to date if you don't even respond to the ticket you created?
Probably not.

oldman4U, 2021-05-06:
@Msq001

Please close this ticket. Thank you

closed 2021-05-06 by bigtreetech