bigtreetech/BIGTREETECH-TouchScreenFirmware #1772: TFT 34V2 manual move screen units

RocksaltX, 2021-03-26:
### Description

On the manual move screen, 10mm is selected for distance travel. Move axis 10mm. Change screens and come back to manual move screen. Increment is changed on screen, but 10mm is still selected. Go to press move button .1mm and it moves 10mm.

### Steps to reproduce

See above

**Expected behavior**
Increment to  update when changing screens

**Actual behavior**
previous increment loaded but icon not changed.

### Hardware Variant
TFT35V2
### TFT Firmware Version & Main Board Firmware details
<BIQU_TFT35_APP1_V2.0.26.x>

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-03-26:
Could you please attach a movie showing the issue. 

Thank you

kisslorand, 2021-03-26:
Fixed: PR #1774

RocksaltX, 2021-03-26:
I applied the changes in PR #1774 and is working. Thank you for the link.

oldman4U, 2021-03-27:
Kisslorand faster than light 👍🏻

@RocksaltX
Please help the community and close the ticket once you do not need it anymore. Thank you

PR #1774 (2021-04-01): Move and ExtruderTune menu unit steps bugfix

### Description

It's a little bugfix for the Move and the ExtruderTuning menu.  Entering into one of these menu, the unit steps displayed the default steps, not the actual ones.

Fixes #1772
closed 2021-03-27 by RocksaltX