bigtreetech/BIGTREETECH-TouchScreenFirmware #1778: Updated MKS28 1.0 TFT display... now i just have 8 white bars

Smartich0ke, 2021-03-27:
Hi,

I have an Artillery Sidewinder X1 and I wanted to update the TFT firmware with smart filament detection enabled because i would like to use my new smart filament sensor.
I followed the instructions in the read-me: drag the MKS_TFT28 1.0 binary to the SD card, drag the TFT 28 icon pack, and drag the config.ini file. Reboot.
On reboot it says Updating... with a green loading bar and then it shows another white loading bar with other white bars at the top of the screen, similar to the photo below and then when ever you reboot the machine, all it does is show these 8 white bars. See picture:
[image]
Printer: Artillery Sidewinder X1
Mainboard: MKS Gen L
Screen: MKS TFT28 v1.0

radek8, 2021-03-28:
You must rename the bin to update the MKS display
file on MKSTFT28.bin,
and rename the folder to MKS.

Smartich0ke, 2021-03-28:
Thanks, that worked!

closed 2021-03-28 by Smartich0ke