bigtreetech/BIGTREETECH-TouchScreenFirmware #1779: [Feature request] Add portrait orientation TFT24

technoo10201, 2021-03-28:
Hello guys !

I would like to know if you can implement a parameter to let the screen orientation in portrait ?
This is the design i will use with my printer and i would like to use the screen as it is presented on this design :

https://www.thingiverse.com/thing:4809101

[Magnetic support for TFT24 - Portait orientation](https://i.ibb.co/TcTZqX1/Anet-AM8-TFT24-Big-Tree-Tech.jpg)

Regards
Technoo'

oldman4U, 2021-03-28:
#524 #602 #1160 #1381

Hope it is ok for you to be the fifth asking for the same feature.

technoo10201, 2021-03-28:
> #524 #602 #1160 #1381
> 
> Hope it is ok for you to be the fifth asking for the same feature.

No i should guess it is a well asked request ;)
I close the ticket then..

Thanks for the fast reply.

closed 2021-03-28 by technoo10201

oldman4U, 2021-03-28:
Yes it is and I hope there will be something like this available soon.

Thank you and happy printing