bigtreetech/BIGTREETECH-TouchScreenFirmware #1786: marlin neopixel Color cycle.

sinole1987, 2021-04-01:
I connected neopixels to SKR 1.4 board, but in TFT mode the set color cycle that marlin performs during heating doesn't happen. it would be cool to have that. 

thank you

gordo3di, 2021-04-01:
Weird. I have the same board and mine does the startup color cycle with no issues since the skr v1.4 and marlin are independent of the LCD screen. Marlin 2.0.6.2 I believe. I would guess it's something in marlin not the touch screen.

Can you change the color via the LED menu?

oldman4U, 2021-04-01:
You both print from?

oldman4U, 2021-04-01:
Please see #1620 and also the PR from traffic-light. 

@sinole1987

Maybe you can collect all your neopixel related tickets into one.

sinole1987, 2021-04-01:
> Weird. I have the same board and mine does the startup color cycle with no issues since the skr v1.4 and marlin are independent of the LCD screen. Marlin 2.0.6.2 I believe. I would guess it's something in marlin not the touch screen.
> 
> Can you change the color via the LED menu?

it works in marlin mode, although we know from load unload issue, that doesn't mean anything. it could be very well a marlin issue. but it just works in marlin mode.

sinole1987, 2021-04-01:
> Please see #1620 and also the PR from traffic-light.
> 
> @sinole1987
> 
> Maybe you can collect all your neopixel related tickets into one.

sure, didn't wana mix two tings up.

oldman4U, 2021-04-01:
2 Q open.

Can you change the color using the menu?

You print from mainboard or TFT?

sinole1987, 2021-04-01:
ye i can manually change the color from machine settings in TFT. Btw I think the light setting should be a little more accessible. Automatic color cycling doesn't happen during a TFT print. it works in marlin mode. I think marlin performs an Internal Gcode during heat up cycle that does not happen when TFT is hosting print. looks like its more of a marlin issue. The reason I labeled this issue as feature request, was because I though we can replicate the Gcode in TFT like M600.

oldman4U, 2021-04-02:
As far as I know this is exactly what traffic-light want to do.

gordo3di, 2021-04-05:
Sorry. I thought you wanted the "startup color" cycle that the nepoixels do not the rgb status colors. Yeah as far as I know it's still not implemented. If you want the status colors on pause, cancel and print complete you can add them to your gcode via the slicer (or on the tft screen config.ini) so at least you can have that functionality. 

https://marlinfw.org/docs/gcode/M150.html

sinole1987, 2021-04-05:
> Sorry. I thought you wanted the "startup color" cycle that the nepoixels do not the rgb status colors. Yeah as far as I know it's still not implemented. If you want the status colors on pause, [...]

that is a great Idea, it doesn't have to be programed into the TFT, but I think it will be a very loaded Gcode. I will try to write it , real challenge is to set colors for each pixel to and give them temperature condition.

gordo3di, 2021-04-07:
> that is a great Idea, it doesn't have to be programed into the TFT, but I think it will be a very loaded Gcode. I will try to write it , real challenge is to set colors for each pixel to and give [...]

For anyone interested, I actually had a very nice neopixel progress bar working in marlin a long time ago. It would go from red to green as it heated and it would also fill up during a print so you could see progress from very far away. It was super useful since we have a big print farm and it was nice to easily see the progress per printer.

The problem is the 8 bit arduino couldn't handle it and it led to tons of layer shifting issues (it took me weeks and a lot of lost filament to determine the issue). From that point I abandoned it but if anyone wants to make sense of the code its here. My code changes start around halfway through the thread. I'm not sure how hard this would be to implement in the touchscreen or on the SKR.

https://github.com/MarlinFirmware/Marlin/issues/8915

oldman4U, 2021-04-13:
From BTTs to do list.

In fact, it is not a really bug, It's a known limitation. Marlin will activate the LED change only after receiving the "M190/M109" gcode, but this gcode will cause the communication interruption between the TFT and Marlin (because Marlin will not respond to other gcode before reaching the target temperature), so TFT will convert this gcode to "M140/M104" to avoid communication interruption, but Marlin will not be able to activate the LED change normally

Rincewind08, 2021-05-01:
Emergency Parser?

Tannoo, 2021-06-07:
#1974

Marcos2BR, 2021-12-17:
thanks digant73 , the file in "Copy to SD Card - BTT/ BIGTREE_TFT35_V3.0.27.x.bin"  (december 16/21 ) work fine the led event (neopixel Color cycle) in my hypercube with  skr1.4 +TFT35-V3.0 (esp01S coupled on tft ) 

https://github.com/digant73/BIGTREETECH-TouchScreenFirmware/tree/bugfix/Copy%20to%20SD%20Card%20-%20BTT

closed 2021-12-24 by bigtreetech