bigtreetech/BIGTREETECH-TouchScreenFirmware #1787: [BUG] (After #1748 a notification bar flashes and beeps even when menuStatus is active thus it shows redundantly)

kmarty, 2021-04-01:
### Description

Since #1748 an annoying beeps and flashes of M117 are redundantly displayed on main screen with menuStatus enabled (very annoying in combination with DisplayLayerProgress plugin in Octoprint).

### Steps to reproduce

1. Just start to print

**Expected behavior**
When menuStatus is active, no further notification/popup is necessary. It is already there.

When menuStatus is active, just show it in status window. No notification bar, no beep.

**Actual behavior**
Displays in duplicate the same message (and beeps - which can be disabled, but when I disable it, what else did I mute?)

### Hardware Variant
TFT35 V3.0
### TFT Firmware Version & Main Board Firmware details
TFT Firmware: from commit 201cd68e (Thu Apr 1 03:20:04 2021 +0000)
Marlin: bugfix-2.0.x, commit [600ef1e47c](https://github.com/MarlinFirmware/Marlin/commit/600ef1e47cf905bec1e76b35f3912863ade4fff0) (2021-04-01)

### Additional Information
[image]
[image]
[image]

Something like:
```
diff --git a/TFT/src/User/API/parseACK.c b/TFT/src/User/API/parseACK.c
index c20f7ad4..6d0520fe 100644
--- a/TFT/src/User/API/parseACK.c
+++ b/TFT/src/User/API/parseACK.c
@@ -227,7 +227,8 @@ void hostActionCommands(void)
     uint16_t index = ack_index;
 
     if (!ack_seen("Ready."))  // avoid to display unneeded/frequent useless notifications (e.g. "My printer Ready.")
-      addToast(DIALOG_TYPE_INFO, dmaL2Cache + index);
+      if (infoMenu.menu[infoMenu.cur] != menuStatus) // Don't show it when menuStatus
+        addToast(DIALOG_TYPE_INFO, dmaL2Cache + index);
   }
   else if (ack_seen(":paused") || ack_seen(":pause"))
   {
```
would be really nice (no idea whether it is correct way or not. It works for me).

thisiskeithb, 2021-04-02:
You should submit a PR for review 🙂

digant73, 2021-04-02:
it will be fixed by PR #1785

closed 2021-04-03 by bigtreetech

Guilouz, 2021-04-03:
@digant73 I have an issue with your fix #1787, when I set temp manually or stop heating, I have many toast notifications "Printer ready" with sound.

Maybe because I use french translation in Marlin (word is "prête.") and ACK don't find "Ready." ?

My terminal return :

M140 S0
//action:notification Sidewinder X1 prête.
ok P31 B31

EDIT: I just try to replace line in ParceACK

if (!ack_seen("Ready.")) // avoid to display unneeded/frequent useless notifications (e.g. "My printer Ready.")

by

if (!ack_seen("prête.")) // avoid to display unneeded/frequent useless notifications (e.g. "My printer Ready.")

and it's working as expected, so there is an issue when using language in Marlin.

digant73, 2021-04-03:
@Guilouz yes, that was the problem. Replace "Ready." with the keyword you receive from Marlin ("prête." in your case)
All the keywords catched in parseAck use only the english version. However I think only for notifications, Marlin can provide a transalation. Also, that annoying notification every time you change a temp, it seems to me more a bug in Marlin.

Guilouz, 2021-04-03:
> @Guilouz yes, that was the problem. Replace "Ready." with the keyword you receive from Marlin ("prête." in your case)
> All the keywords catched in parseAck use only the english version. However I [...]

@digant73 Yes it works like that but it is binding if you have to edit with each release... Is there not another solution? Before a certain PR there was no this problem of redundant notifications.

kmarty, 2021-04-03:
It all start with #1748