bigtreetech/BIGTREETECH-TouchScreenFirmware #1788: How to connect a SKR1.4 / TFT35 / Raspberry / End of filament?

Magiquall, 2021-04-03:
Hello,

It's not really a bug, but a lack of information.

Do you have any further explanation to operate a SKR 1.4 Turbo and TFT35 BTT (with Marlin menu) with an octoprint (Raspberry) and a filament sensor.

Apparently the end of the filament sensor needs to be connected to the TFT screen. But if I connect the Raspberry to the SKR 1.4's USB, it will never see this sensor (well I guess).

I saw the following settings in the screen ini file:
#### Serial always on
# Keep UART (serial communication) active in Marlin mode.
# Allows a transparent OctoPrint UART connection to the TFT UART / serial extension port, regardless of the mode the TFT is in.
# Options: [activate: 1, deactivate: 0]
serial_always_on: 0

It sounds like a potential solution, but how do you put it in place?
Of course by setting it to 1:
serial_always_on: 1

The UART inputs of the screen are not USB, The Raspberry has UART outputs ?

Thank you

bigtreetech, 2021-04-04:
In fact, you can plug filament sensor into the motherboard. 
You just need to enable the `EMERGENCY_PARSER ` , `HOST_ACTION_COMMANDS ` and `HOST_PROMPT_SUPPORT ` in Marlin to get a good support for TFT(`M75` in start gcode, `M77` in end gcode should add in config.ini of TFT).

This should be the best solution for compatibility

Magiquall, 2021-04-05:
Hello, No, unfortunately, I am not advanced, because I am returning my motherboard to after-sales service, it seems there is a fault on the ventilation output which is at 100% as soon as I supply the card. In any case, I have seen or made these modifications, but I still have doubts. Where do I connect the raspberry? Still in USB?

bigtreetech, 2021-04-06:
> Where do I connect the raspberry? Still in USB?

Yes, Raspberry pi directly connected to the motherboard. Raspberry pi communicates directly with the motherboard. TFT also communicates directly with the motherboard. You can get the most stable communication at present.

Magiquall, 2021-04-06:
Thank you very much, I really appreciate your responsiveness, unlike your competitor, Makerbase, where I'm still waiting for answers. Thank you, thank you and well done. Continue like that. Just for information, is the operation the same with the SKR v2? Because I hesitate to take an SKR v2 instead of an SKR v1. 4.

oldman4U, 2021-04-06:
You mean SKR PRO v1.2 or mini v2 or ...?

Magiquall, 2021-04-06:
> Vous voulez dire SKR PRO v1.2 ou mini v2 ou ...?

No I'm talking about SKR 2. 
https://github.com/bigtreetech/SKR-2

oldman4U, 2021-04-06:
This board is too new for me;-)

I BELIEVE the GTR has a Pi port but you have to check if this is true and if so what it does. Octoprint is not my world.

bigtreetech, 2021-04-08:
> is the operation the same with the SKR v2? Because I hesitate to take an SKR v2 instead of an SKR v1. 4.
Yes, the steps of using octoprint are the same on  these two boards.

closed 2021-04-10 by Magiquall