bigtreetech/BIGTREETECH-TouchScreenFirmware #1791: [FR] RRF improvements

bttguy, 2021-04-03:
The latest implemented changes made the BTT TFT really useful for RRF users.
Here I request some additional features in a hope that @bigtreetech or other developers would help to improve the useability of the screen.

Original FR: #1400 (now closed, as the core support is already there)
Discussion for RRF: #1763

Requested features:
•	Menu system for macros + "interactive" macros
•	Thumbnail and menu system support for onboard gcodes
•	Make the load/unload menu useable
Any RRF users is welcomed to suggest FR's here!

Thanks in advance to anyone who helps! :)

PaulG4H, 2021-04-08:
Macro was working with RRF until this commit I asume:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1769

If I compare the macro and rrfmacro files there is a huge difference...

bttguy, 2021-04-08:
Yeah, sadly something we've wrong lately with the macros, I opened an issue for this bug in #1814

bttguy, 2021-04-19:
Thanks for #1842 :)

bigtreetech, 2021-06-02:
Hello. 

`* Menu system for macros + "interactive" macros`
What does "interactive" macro mean? How does it work? Does the current macro menu not meet the requirements?

`* Thumbnail and menu system support for onboard gcodes`
This feature needs to be discussed, because the current thumbnail is to add bitmap data to gcode,  then the TFT parses it. The difficulty lies in
1. The gcode of onboard sd is parsed by the motherboard, so the TFT can't get the data in gcode. Need the motherboard firmware to send the bitmap data to the TFT.
2. TFT communicates with motherboard by UART, the amount of bitmap data is relatively large, and the transmission through UART is very slow. The amount of data will be much smaller if the thumbnail is compressed into JPEG format, But It's very difficult to parse JPEG based on TFT MCU's performance.

`* Make the load/unload menu useable`
`Movement->Extrude` is available after #1909. “(Un)Load” menu is called "M701" gcode to load filament, I'm not sure if RRF support similar gcode. Anyway `Movement->Extrude` is universal and available extrusion menu

`* Make the PID tune menu useable`
The same is already available after #1856

jaysuk, 2021-06-02:
All gcodes available in RRF are described here
https://duet3d.dozuki.com/Wiki/Gcode#Section_M701_Load_filament
As you can see, M701 is available

bigtreetech, 2021-06-02:
Can someone provide a `filaments` folder containing `load.g` and `unload.g` for me to test?

jaysuk, 2021-06-02:
You basically make your own up. See here
https://duet3d.dozuki.com/Wiki/Filaments

bttguy, 2021-06-02:
Thank you @bigtreetech  the improvements! After work I can test and answer your other questions!

bttguy, 2021-06-02:
@bigtreetech 
So "interactive" macros means complex macros, that require user interaction/confirmation ect. For example a leveling macro: go to point 1, display a massege, wait for encoder press, than go to point 2 ... I only can launch it from the screen but I have to use DWC to click ok and read the msg-s. I am not that technical to understand the different communication protocols between the screen/board/DWC, I am not sure if it is even possible to ask for this function, but I listed it as an idea that I would use.

bttguy, 2021-06-02:
@bigtreetech can you detail me how you indented to use the fixed unload/load menu for further testing? I downloaded the version with the PR included, but the Un/Load menu dont work at all. I can heaet up the nozzle, but nothing happens neither when I click on load neither when I click on unload. And also where can these parameters be specified? Load/unload are completely different in a bowden printer than on a direct drive printer.
Also how the PID tuning works? When I start from TFT nothing shows up on the DWC, is it supposed to be this way, it is only handled by the TFT and after some time it will pop up the values? But maybe something wrong with this also (at least for me), if I press start PID, nothing happenes, the fan kicks on for a sec, thats all. The heaters dont heat, DWC report no change, just I get the TFT screen for wait.
If there is I do wrong with my testing let me know and I will try again. I use RRF 3.2.2 latest stable with the new PR for RRF for TFT firmware

bigtreetech, 2021-06-03:
> @bigtreetech can you detail me how you indented to use the fixed unload/load menu for further testing? I downloaded the version with the PR included, but the Un/Load menu dont work at all. I can [...]

Hello,  
`(Un)Load` menu is not available for RRF now,  `Movement->Extrude` is normal for extrude.
Will the nozzle warm up after PID is started?

bttguy, 2021-06-03:
It did not heat up.

bigtreetech, 2021-06-04:
> It did not heat up.

Can you test whether `M303 T0 S200` can heat the nozzle normally?

bttguy, 2021-06-04:
> > It did not heat up.
> 
> Can you test whether `M303 T0 S200` can heat the nozzle normally?

I will at home. From DWC console this is the gcode that I use for PID, hopefully it can be executed from the TFT PID menu.

bttguy, 2021-06-04:
@bigtreetech Sending M303 T0 S205 worked perfectly from the TFT terminal. I even could follow the responses, than check the final value than save it without issue. These were also refreshed on DWC.
[image]

bigtreetech, 2021-06-05:
> @bigtreetech Sending M303 T0 S205 worked perfectly from the TFT terminal. I even could follow the responses, than check the final value than save it without issue. These were also refreshed on DWC.

Do you compile the source code in #1962? PR needs to compile its own .bin to contain changes.

bttguy, 2021-06-05:
@bigtreetech With a self compiled firmware the PID menu is working, it just don't report the progress, just say to do not touch the screen until t finished, but that's the way it's working I guess. But following the progress on DWC, phase 4 seems to be still broken "tuning with fan off". The part cooling fan is keeps spinning at 100% under the whole PID. Executing M303 TO S200 from the terminal, everything as expected, PID start with fan off, and only turn on at phase 5.
Also after it done it ask if I want to save it to eeprom. RepRap does not use eeprom, but it will require an m500 command to save. If this is what this screen does, it can work.

bttguy, 2021-08-17:
Another great improvement in #2122 by @pfn. Thank you!

pfn, 2021-08-23:
> • Thumbnail and menu system support for onboard gcodes

I believe menu support is already there with folders, thumbnails won't ever become possible.

> • Make the load/unload menu useable (only move-extrude menu working atm)

Unload menu works, load will not ever work as it requires a new argument to M701 which the TFT cannot accommodate. (It *is* possible, but much more trouble than it's worth, involving a filament type picker, etc.)

bttguy, 2021-09-08:
Thanks to all that contributed to this FR. Closing as most of the FR's are solved! :)

closed 2021-09-08 by bttguy