bigtreetech/BIGTREETECH-TouchScreenFirmware #1792: [BUG] MKS TFT28 doesn't boot into firmware after bootloader, displays glitches

drewler, 2021-04-03:
### Description

After flashing the firmware to a MKS TFT28 V3.0 board, it displays some glitches but doesn't boot, as shown in the following video.

https://www.youtube.com/watch?v=XtiFZHMdNLs

I'd like to help and prepare a PR to fix it. It looks like it's not starting the main program after the booloader. Only the main binary is renamed to `.CUR` after the update, so seems like it's not reaching the point where fonts, icons or configuration are applied.

Any hint about what could be wrong?

### Steps to reproduce

1. Add `MKSTFT28.bin`, `config.ini` and `TFT28` folder (renamed to `MKS`) to the root of the SD card
2. Turn on. Update happens
3. Display gets stuck as shown on video

After that, I used an STLink V2 to erase the device and flashed `Bootloader MKS_TFT28.bin` to `0x00000000`. After that, I've flashed `MKSTFT28.bin` using the SD card, but it still doesn't work. Flashing the stock firmware through the SD card still works.

**Expected behavior**
IT should boot into BIGTREETECHS's TFT firmware.

**Actual behavior**

[image]

Also, only the main update is executed (fonts or config are not applied)

[image]

### Hardware Variant
MKS TFT28 V3.0 (3.0.6 official firmware)
### TFT Firmware Version & Main Board Firmware details
3.0.6 - official firmware
master branch - BIGTREETECH's firmware

### Additional Information
* https://www.youtube.com/watch?v=XtiFZHMdNLs

digant73, 2021-04-04:
Try with a different sd card not bigger than 32GB. That fw perfectly works on my MKS TFT

digant73, 2021-04-04:
The fw is compatible with MKS 1.4

oldman4U, 2021-04-05:
Please help the community and close the ticket once you do not need it anymore. Thank you

drewler, 2021-04-05:
@digant73 @oldman4U any hints about where in the code I could look? I wasn't looking for a solution but trying to figure out what is not working so I could help.

Like: does this glitch look like it's running some data as code? Could it be some wrong offset? Or maybe it's beacuse the LCD driver is the wrong one?

Please, let me know if you have any hints. If you think it's not worth it, then I'll quickly close the issue.

digant73, 2021-04-05:
the main settings for the currently supported MKS TFT v1.4 are in file pin_MKS_TFT32_V1_4.h. The used LCD driver is HX8558. I don't know what LCD driver is used by MKS TFT v3.0. Probably platformio.ini and other features in the fw need to be changed in order to support that TFT

drewler, 2021-04-05:
Awesome! Thanks @digant73

[image]

I just had to compile `MKS_TFT32_V1_3` and it works now! In my case, I also had to change a line in [TFT/src/User/Variants/pin_MKS_TFT32_V1_3.h](https://github.com/Uz45/BIGTREETECH-TouchScreenFirmware/blob/5634f771210a7593b89dba4601c6933c1f2c7c0c/TFT/src/User/Variants/pin_MKS_TFT32_V1_3.h#L24) to prevent mirroring, as the comment suggested.

Thanks again. Closing this issue now.

closed 2021-04-05 by drewler

oldman4U, 2021-04-05:
Maybe you want to contribute with a PR?