bigtreetech/BIGTREETECH-TouchScreenFirmware #1793: [BUG] No model preview on print screen with lastest build

GregSKR, 2021-04-04:
I'm using model preview in my STL files, and printing from the SD card of my TFT.

With lastest build v0.0.27, model preview are ok when exploring the SD card, but not in the print screen (was working before).
The babystep button always as the babystep icon, no more model preview...

It's just me or anyone else?

radek8, 2021-04-04:
Yes, this is a mistake.
in point
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/9e5a9a5f32b58ec74d92487429a314ba28dff3c2#diff-c625e283a40da9ab01e83c66cab47ca45199095a1b51d627e21
the preview is OK.
After merging PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1785
print previews are no longer displayed.
@digant73 can you check your PR?

Thank you

bigtreetech, 2021-04-05:
Hello. #1795 should fix this problem, Please check, Thank you very much for your feedback and test.

closed 2021-04-05 by bigtreetech

GregSKR, 2021-04-05:
So fast! 👍🏻 
Yes, just checked, problem fixed in #1795, thanks!

PR #1795 (2021-04-05): printing menu cannot preview model bug
### Description

fix #1793
### Benefits

### Related Issues