bigtreetech/BIGTREETECH-TouchScreenFirmware #1794: [BUG] SmartFilamentSensor does not work when Linear Advance is enabled

phantom-333, 2021-04-04:
### Description

When I enable Linear Advance in the Marlin configuration, SmartFilamentSensor stops working. 
I need both functions, but they don't work together.

### Steps to reproduce

1. Connect the SmartFilamentSensor to the FIL_DET port of the BT TFT35 v3.0 screen.
2. Configure Marlin and the TFT 35 v3.0 configuration, make sure that SFS is working correctly in touch mode.
3. Uncomment #define IN_ADVANCE in Configuration_adv.
4. SFS stops working in touch mode.

**Expected behavior**
SmartFilamentSensor works when Linear Advance is activated

**Actual behavior**
SmartFilamentSensor not works when LA is activated

### Hardware Variant
BTT TFT35 v3.0
BTT GTR1.0 Main Board
BTT SmartFilamentSensor

### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.26.x
Marlin-2.0.7.2

### Additional Information
* Include a ZIP file containing my "Configuration.h" and "Configuration_adv.h"
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6255390/Configuration.zip)

oldman4U, 2021-04-04:
Yes.

This is a known issue and at the moment you have to decide which one to use.

radek8, 2021-04-04:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1612#issuecomment-782926663
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1611

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1752

DaStivi, 2021-04-05:
Connect it to the mainboard, enable it in firmware, recompile and use sfs again with LA...

phantom-333, 2021-04-05:
> Connect it to the mainboard, enable it in firmware, recompile and use sfs again with LA...

But in this configuration, SFS does not work in the touch mode of the TFT35.

phantom-333, 2021-04-05:
> Yes.
> 
> This is a known issue and at the moment you have to decide which one to use.

Can the code change #1752 suggested by @digant73 help with this problem?

oldman4U, 2021-04-05:
Try it. It was a single command which was changed.

phantom-333, 2021-04-05:
> Try it. It was a single command which was changed.

I tried to make changes to the pre-release version of the software Vx.x.27, compile and install it on the TFT35. 
In Marlin M114_DETAIL, M114_REALTIME and M114_LEGACY are activated. 
But these actions did not solve the problem, SFS does not work in touch mode with LA.

DaStivi, 2021-04-05:
> > Connect it to the mainboard, enable it in firmware, recompile and use sfs again with LA...
> 
> But in this configuration, SFS does not work in the touch mode of the TFT35.

you sure? because i even got the filament change messages and prompts on tft when the sfs has been triggered from octoprint print job

phantom-333, 2021-04-05:
> you sure? because i even got the filament change messages and prompts on tft when the sfs has been triggered from octoprint print job

Yes, here's a video of how it works:

https://user-images.githubusercontent.com/81932009/113591135-32d5ab00-963c-11eb-92bb-c92753dd3f86.mp4

https://user-images.githubusercontent.com/81932009/113591157-3bc67c80-963c-11eb-9327-cdbaa34716bc.mp4

bigtreetech, 2021-04-06:
> I tried to make changes to the pre-release version of the software Vx.x.27, compile and install it on the TFT35.
> In Marlin M114_DETAIL, M114_REALTIME and M114_LEGACY are activated.
> But these actions [...]

This is because the `LA` function in Marlin will destroy the `M114 E`'s response. The E-axis coordinate of Marlin's feedback is always 0. You can install SFS on the motherboard, and then modify the `config.ini` of the TFT. Enable `start_gcode`, `end_gcode`, and `cancel_gcode`. And add `M75` to `start_gcode`, add `M77` to `end_gcode` and `cancel_gcode`
Enable `EMERGENCY_PARSER`, `HOST_ACTION_COMMANDS` and `HOST_PROMPT_SUPPORT` in Marlin. Then SFS should work.

I'm going to look at the Marlin code now to try to fix the destruction of `LA` on `M114 E`

oldman4U, 2021-04-06:
Now BTT fixed Marlin issues......;-)

phantom-333, 2021-04-06:
> This is because the `LA` function in Marlin will destroy the `M114 E`'s response. The E-axis coordinate of Marlin's feedback is always 0. You can install SFS on the motherboard, and then modify the [...]

I checked the suggested workaround for the problem, it works!
But I also found a new serious bug:
1. When the SPS is triggered, the printing process is paused and the nozzle is parked;
2. If you start printing again and do not put the filament in the sensor and hotend, the SRS sensor is triggered again and the selection "Action command" will appear on the TFT35:
- purge more
- disable the sensor
3. When I selected "purge more", the TFT35 or mainboard GTR1.0 software crashed and rebooted. Of course, the printing process will be interrupted and the part will be damaged.

Crash video:
https://drive.google.com/file/d/10Rja_2JaxuKdpdL8EtzS91-O3nHZxTYO/view?usp=sharing

Normal work:
https://drive.google.com/file/d/1VIlpTlLv5THkPBKAyy7S-D_3ZDDFVicK/view?usp=sharing

oldman4U, 2021-04-06:
Install a Marlin version after February 28th to fix the crash.

Or install a mainboard with LPC CPU😁

phantom-333, 2021-04-06:
> Install a Marlin version after February 28th to fix the crash.
> 
> Or install a mainboard with LPC CPU😁

The latest version of Marlin 2.0.7.2 is already installed.

phantom-333, 2021-04-06:
Or is it desirable to install this version?
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

oldman4U, 2021-04-06:
Yes bugfix. Sorry.

👍🏻

phantom-333, 2021-04-06:
Ok thanks!
Now I'm going to reinstall Marlin.

oldman4U, 2021-04-06:
You can check the PRs of Marlin and it was February 27th or 28th where a STM32 related fix has been merged. This is the one which fixed the crash with purge more.

phantom-333, 2021-04-07:
I reinstalled Marlin, now the software does not crash.
The issue can be closed; in the solution proposed by bigtreetech, the SFS sensor works perfectly with LA.
Thanks!

oldman4U, 2021-04-07:
Sounds great.

But only you can close this ticket. 

Happy printing

closed 2021-04-07 by phantom-333

phantom-333, 2021-04-07:
In the archive there is a working config for Marlin and TFT35.
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6273199/config.zip)

bigtreetech, 2021-04-10:
https://github.com/MarlinFirmware/Marlin/pull/21579
The bug of `LIN_ADVANCE` has been fixed in Marlin, So after that, We should be able to work with any combination of `SFS<->MotherBoard` or `SFS<->TFT`

oldman4U, 2021-04-10:
This is great!

We should change the minimum Marlin mode in our documentation.

phantom-333, 2021-04-11:
> [MarlinFirmware/Marlin#21579](https://github.com/MarlinFirmware/Marlin/pull/21579)
> The bug of `LIN_ADVANCE` has been fixed in Marlin, So after that, We should be able to work with any combination of `SFS<->MotherBoard` or `SFS<->TFT`

Good news, thanks!