bigtreetech/BIGTREETECH-TouchScreenFirmware #1798: Illegal FLASH APP! after firmware upgrade

peddomgar, 2021-04-05:
I upgraded my BTT TFT35 regular screen and now if power off or reset screen with SD inserted then shows this error and only can flash again the screen to restore system.

Every time you reset or power off with a SD card inserted will though this error on next start and will not work.

I've tried to remove .CUR files and use another SD card and always is the same.

oldman4U, 2021-04-05:
Format the SD card after the firmware update to remove the .CUR files and see if this solves your problem. Are you on macOS?

oldman4U, 2021-04-07:
Hello peddomgar.

Seems this is your first ticket, so I wanted to let you know that it is common to answer a ticket, especially in case you are the one who started it.

Please help the community and close this ticket once you do not need it anymore.

Thank you

peddomgar, 2021-04-09:
Hi! Sorry, I was not available to try before, you are right and worked for me!! I use a Mac and needed to format again the SD card before insert it again to the screen, now is working fine. Thanks so much!!

oldman4U, 2021-04-09:
Makes sense. This is macOS related, where files are not really deleted. I have the same issue.

Please close the ticket, thank you.

closed 2021-04-09 by peddomgar

oldman4U, 2021-04-09:
Happy printing;-)