bigtreetech/BIGTREETECH-TouchScreenFirmware #1802: HELP! SKR 1.4 Turbo Not Connecting to TFT35 V3.0...Have Tried Everything

adcallow, 2021-04-06:
Recently decided to upgrade my printer and bought new TMC 2209 drivers, an SKR 1.4 turbo board, and the TFT 35 touchscreen (and a BL Touch). Thinking that it would be no problem getting them all up and running together because the main parts are all from BTT...cut to ~40 hours of research and testing later and the TFT is STILL saying "no printer connected". I have searched all the boards that I know of for similar issues and everything says mostly the same thing, which I've already done. So far the steps I've completed (but have not been successful)

- Uploaded Marlin 2.6 to SKR board, Serial port #1 is -1, and serial port #2 is 0, Baud rate is 115200
- Commented CR10 and uncommented REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER...I even went so far as to comment out the sub-arguments of the already commented out if statements just to be SURE that the only thing selected was the REPRAP.
- Prior to uploading the TFT firmware I checked that the baud rate for my v3.0 was at 115200 (which it was)
- Built out everything in platform.io with no issues then uploaded the firmware to both the board and the TFT via memory card instead of USB..and NOTHING

I then went in and changed the serial ports to 0 and 1, 0 and -1, -1 and 1, 0 and 2, etc. trying every combo through port 3, and uploaded the firmware to the board everytime. Nothing. I then did the same thing with the baud rate at 250000 (for both the board and the TFT)...I refreshed the upload the the TFT when I switched the baud to 250000. Nothing. I then switched both back to 115200, and tried to use the LCD screen to switch baud rates. Nothing. I then switched the port in marlin mode to always open, nothing (in Marlin mode). Throughout the process I have checked multiple times that the connectors are tight and aligned correctly (which they are). I also checked to make sure that the BLTouch connectors are in the right spots (because I saw a post from whenever that if these are wrong it can screw up the TFT/SKR connection), and they are. I then though that if I connect the SKR to the TFT via USB it should work...NOPE! Still no printer.

I LITERALLY don't know what to do anymore. I feel like I"ve done everything that I can possibly think of and I'm starting to get really frustrated that I spent all this time and money on brand new BTT gear and it's caused more problems than good.

radek8, 2021-04-06:
Before testing anything further, set the correct Marlin parameters on the SKR:
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200
#define EEPROM_SETTINGS
// #define DISABLE_M503
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
// #define CR10_STOCKDISPLAY
#define M114_DETAIL
#define M115_GEOMETRY_REPORT

set to TFT:
set speed: 115200

Check the conductivity of the 5-pin serial cable. If it is OK, connect the RS232 and TFT connectors to it.
Check the correct orientation of the serial cable.

Connect the two 10-pin cables to connectors EXP1 and EXP2

Turn on the printer and switch to simulated Marlin mode.
Is Marlin mode working?

oldman4U, 2021-04-06:
And send pictures.

radek8, 2021-04-06:
> I then though that if I connect the SKR to the TFT via USB it should work...NOPE! Still no printer.
> 

SKR and TFT are not interconnected via USB cable !!!

adcallow, 2021-04-06:
Thanks radek8, I'll give it a go and let you know. Yes, sorry, I was in a bit of a rush writing bc I had to get to a meeting, and there were so many things I had tried.

Will let you know and send pics.

Thx

oldman4U, 2021-04-06:
I am sure you read a lot already but please read the Readme at the front page of this repository.

gordo3di, 2021-04-06:
> Before testing anything further, set the correct Marlin parameters on the SKR:
> #define SERIAL_PORT 0
> #define SERIAL_PORT_2 -1
> #define BAUDRATE 115200
> #define EEPROM_SETTINGS
> // #define DISABLE_M503 [...]

I think my settings are the same but 250000 baud... Not sure why but it works on all 6 of our test machines. Heads up... SKR v2 is on its way to me... v1.4 is discontinued.

radek8, 2021-04-06:
Yes, the speed of 250000 also works without problems for me.
But if @adcallow has a problem, we need to start with a lower speed and if the display connects, we can increase the speed.

adcallow, 2021-04-06:
Thanks for looking out @radek8 unfortunately it is still not connecting to the printer. I have put responses and images to your post below

Before testing anything further, set the correct Marlin parameters on the SKR:
#define SERIAL_PORT 0.......CHECK
#define SERIAL_PORT_2 -1........CHECK
#define BAUDRATE 115200.........CHECK
#define EEPROM_SETTINGS.........CHECK
// #define DISABLE_M503........CHECK
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.........CHECK
// #define CR10_STOCKDISPLAY...........CHECK
#define M114_DETAIL............CHECK
#define M115_GEOMETRY_REPORT...........CHECK

set to TFT:
set speed: 115200...........CHECK

Check the conductivity of the 5-pin serial cable. If it is OK, connect the RS232 and TFT connectors to it.....CONDUCTIVE - CHECK
Check the correct orientation of the serial cable......CORRECT ORIENTATION - CHECK

Connect the two 10-pin cables to connectors EXP1 and EXP2........CHECK

Turn on the printer and switch to simulated Marlin mode......CHECK
Is Marlin mode working?..........Nope
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

adcallow, 2021-04-06:
> I think my settings are the same but 250000 baud... Not sure why but it works on all 6 of our test machines. Heads up... SKR v2 is on its way to me... v1.4 turbo is discontinued.

Heads up...TURBO!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

radek8, 2021-04-07:
Send marlin configuration files. 
Send config.ini TFT file. 
Take a picture of the TFT that the sides of the cables

radek8, 2021-04-07:
First, we will solve the problem with displaying Marlin mode so that we know that Marlin is fine.
Is the BIN file renamed to CUR after the SKR flash?

adcallow, 2021-04-07:
@radek8 thanks for all the help! I've copied the config files below and will take a pic of the TFT in a sec. 

Yes, it does.

SKR:

```javascript
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020006

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * https://reprap.org/wiki/Calibration
 * https://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * https://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * https://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

[... 2336 lines cut ...]
/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES
```

adcallow, 2021-04-07:
TFT:

```javascript
#--------------------------------------------------------------------
#
# DEFAULT CONFIG FILE FOR BigTreeTech TFT CONTROLLERS
#
#--------------------------------------------------------------------

#--------------------------------------------------------------------
# Supported TFT Variants
#--------------------------------------------------------------------
#
# BIGTREE_TFT35_V1_0    / BIGTREE_TFT35_V1_1 / BIGTREE_TFT35_V1_2
# BIGTREE_TFT35_V2_0    / BIGTREE_TFT35_V3_0 / BIGTREE_TFT35_E3_V3_0
# BIGTREE_TFT35_B1_V3_0 / BIGTREE_TFT43_V3_0 / BIGTREE_TFT50_V3_0
# BIGTREE_TFT70_V3_0    / BIGTREE_TFT28_V1_0 / BIGTREE_TFT28_V3_0
# BIGTREE_TFT24_V1_1    / MKS_32_V1_4        / MKS_28_V1_0
#
# Firmware source: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

#--------------------------------------------------------------------
# Supported Marlin Firmware Versions
#--------------------------------------------------------------------
#
# Minimum Marlin firmware version: 2.0.5.4
# Distribution date: 2020-05-12
# Firmware source: https://github.com/MarlinFirmware/Marlin/releases
#
# In order the TFT firmware is able to provide all of its functionalities/features,
# ensure that the following options are enabled in Marlin firmware.
#
# General options:
#   EEPROM_SETTINGS (in Configuration.h)
#   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
#   M115_GEOMETRY_REPORT (in Configuration_adv.h)
#   M114_DETAIL (in Configuration_adv.h)
#   REPORT_FAN_CHANGE (in Configuration_adv.h)
#
# Options to support printing from onboard SD:
#   SDSUPPORT (in Configuration.h)
#   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)
#
# Options to support dialog with host:
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#   HOST_PROMPT_SUPPORT (in Configuration_adv.h)
#
# Options to support M600 with host & (Un)Load menu:
#   Options to support dialog with host (as pre requisite)
#   NOZZLE_PARK_FEATURE (in Configuration.h)
#   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
#   PARK_HEAD_ON_PAUSE (in Configuration_adv.h)
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)
#
# Options to support Babystep menu:
#   BABYSTEPPING (in Configuration_adv.h)
#
# Options to support repeatibility test to test probe accuracy (M48):
#   Z_MIN_PROBE_REPEATABILITY_TEST (in Configuration.h)
#
# Options to support aligment of multiple Z steppers using a bed probe (G34):
#   Z_STEPPER_AUTO_ALIGN (in Configuration_adv.h)

#--------------------------------------------------------------------
# General Settings
#--------------------------------------------------------------------
[... 569 lines cut ...]
start_gcode_enabled:0

#### Default End Gcode Status
#   Options: [enable: 1, disable: 0]
end_gcode_enabled:0

#### Default Cancel Gcode Status
#   Options: [enable: 1, disable: 0]
cancel_gcode_enabled:0

#### Start Gcode
# This gcode will runs before starting a print if `start_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
start_gcode:G28 XY R20\n

#### End Gcode
# This gcode will runs after a print is completed if `end_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
end_gcode:M104 S0\nM140 S0\nM107\nM18\n

#### Cancel Gcode
# This gcode will runs when a print is canceled if `cancel_gcode_enabled` is enabled.
#   Value range: [min: 3, max: 75 characters]
cancel_gcode:M104 S0\nM140 S0\nG28 XY R10\nM107\nM18\n
```

radek8, 2021-04-07:
Please delete the configuration you entered.
the whole discussion becomes confusing.
insert the configuration files into the ZIP file and insert this ZIP file here.
Thank you

oldman4U, 2021-04-07:
Hi adcallow.

radek8 is the best user you can get here when it comes to solving a problem. Unfortunately your issue is 99% in hardware because all you can do in software related to the serial connection is to define the correct serial port and a matching baud rate. I guess your mainboard or TFT or for me, most likely the black RS232 cable is broken. Do you have a chance to get (make) a replacement?

gordo3di, 2021-04-07:
> Hi adcallow.
>
> radek8 is the best user you can get here when it comes to solving a problem. Unfortunately your issue is 99% in hardware because all you can do in software related to the serial [...]

I tend to agree. If you have a spare full graphic display laying around you can try plugging that in to see if marlin shows up otherwise I think the board might be doa. We had 2 boards that just never worked using the same code as all the rest.

adcallow, 2021-04-07:
@oldman4U @gordo3di @radek8 Thanks to all the help guys, I really appreciate it. This is my first time really asking for help on Git (or any other boards for that matter) and I wasn't sure what I'd end up getting...but honestly you guys were awesome. And I feel like I ultimately arrived at a solution...granted it may not have been the optimal one, but it's allowing me to continue to move forward, as opposed to banging my head against the wall.

I'll check the RS232 on my old graphic display, and then I can make one to test as well. If none of that works then I'll send back either the board (and get a V2) or the TFT and get a new one.

Thanks again!!!

oldman4U, 2021-04-08:
You are welcome.

Your old display probably has no rs232 port but the two 10pin ports. Those are used for Marlin and are irrelevant for the touchscreen mode.

radek8, 2021-04-08:
The SKR or TFT may be defective, but there may also be a connection problem. Marlin may fall into the reset loop.
Send a photo of the connection.
Please delete that infinite configuration.
My phone has trouble loading such long discussions :-)

adcallow, 2021-04-08:
@radek8 I've attached two photos of the two connections I tried. Neither of them worked. It also didn't work when I disconnected the reset pin. Note, I am 99.99% they are connected correctly as I checked about a hundred times. They are also secure (I checked this too).

Got the new TFT board today. Ran from the front stoop like it was Xmas morning. Connected it to the printer 'as is' just to see if it was a firmware thing that I was doing...No luck. So I flashed the new firmware onto the new TFT aaaaaaaaaaaaaaaaand nothing. "No Printer Attached" same thing in Marlin. 

I've ordered a new board...While I wait for that to get here, and because I was looking forward to these working so much, I'm going to completely detach my SKR and TFT from my printer and start all over again with both the wiring and the firmware. Hoping that it's something I did. 

I would like to say that while my experience with you gents has been great, my experience with BTT hardware has left a LOT to be desired. One of the 2209 drivers I ordered was broken (made this horrid noise and screwed up whichever axis it was used on) so I returned that to get a new one, then I order the TFT and SKR boards and one of those seems to be defective...At least, based on what we've been doing here, that is what I am led to believe...that's two out of three products.

[image]
[image]

radek8, 2021-04-08:
As I wrote before.
First we have to put the Raspberry regime into operation.
Only then can we proceed to touch mode.
Disconnect serial cable (black 5 pin cable)
Disconnect the USB cable.
Connect with a 10-pin cable EXP1 and EXP2 connector.
Try Marlin mod.
Give a message with the result.

adcallow, 2021-04-08:
Did that, and I get the "Marlin Mode" at the top, then under that I get a larger "Marlin" in an oval, and listed under that I see 2.0.6.1 and http://marlinfw.org. But that is all. It doesn't do anything after that if I click the wheel or spin it.

radek8, 2021-04-09:
Looks like marlin's not running.
Send me your BIN file with Marlin.
I'll test it at my place.

oldman4U, 2021-04-09:
MAYBE it could also be worth to buy a new PS. Broken driver, broken SKR and maybe broken TFT, this could point to an issue with the PS. Do yourself a favour and buy a Meanwell 350/24 and do not use an old PS from a computer.

adcallow, 2021-04-09:
@radek8 Thanks again for the help. I've got the 2.0 coming tomorrow in the mail. Fingers crossed I won't have to deal with ALL this stuff anymore...he says ironically because there is no way it's going to go smoothly.

Thanks again

adcallow, 2021-04-09:
@oldman4U Hm, I never thought about it like that. I always assumed that the power supply was something really basic and easy so there couldn't ever be a problem with it.

@radek8 Ummm I'm embarrassed to say this, but since I'm new to Git, I don't know how to send you something. I've followed you, but I don't think I can send things through that. I'm a goober. Sorry.

gordo3di, 2021-04-09:
The blue USB cable you have is not to connect the LCD to the SKR. The USB on the LCD screen is for a USB memory stick and the USB on the SKR is to connect it to a computer. Not sure if that' causing any weird issues.

radek8, 2021-04-09:
Take the BIN file with the firmware for SKR.
Paste it into a ZIP file.
Insert the zip file into the comment.
The way you sent the photos.

radek8, 2021-04-09:
@adcallow 
I don't know if SKR V2.0 is the solution for you.
For SKR V2.0 I did not see a set of pins defined in Marlin.

adcallow, 2021-04-12:
@radek8 hahaha I think I'm finding that out now (i.e. that the 2.0 isn't the solution hahaha)

gordo3di, 2021-04-12:
@adcallow @radek8 The SKR v2 replaces the v1.4 /v1.4 turbo. It uses an STM32 processor instead of the LPC1678 and LPC1679. Marlin doesn't have configs yet since it's so new. Also looking at the manual, the pins are all different from the v1.4 board so it's gonna require some work. @bigtreetech do you have the config files for the SKR v2 posted yet?

radek8, 2021-04-12:
It is not merged yet. discussion in progress :-)

https://github.com/MarlinFirmware/Marlin/pull/21503

adcallow, 2021-04-12:
@radek8 @gordo3di So I did find this: https://github.com/kad/Marlin/wiki#installing-firmware and updated it a tiny bit for myself and it built with no problems...I'm exploring it further at the moment because (surprise surprise) there is no printer attached haha.

radek8, 2021-04-12:
You must select the correct serial port number in Marlin.
it's a new board so I won't tell you the number, I don't know it

adcallow, 2021-04-12:
HOLY CRAP! I just connected the TFT and it's working!!!!!!!!!!!!!!!!!!!!!!!!! Well i mean that the printer is connected!!!!!!!!!!!!!!!

oldman4U, 2021-04-12:
It can be unbelievable easy. 👍🏻

adcallow, 2021-04-12:
aaaaaaaaaaaaaaaaaand now the screen is completely black...yay. UGH!!!!!!

adcallow, 2021-04-13:
thanks again @radek8 @gordo3di @oldman4U appreciate the support. I will say I have learned a TON over the last week. I think I'm going to reset my old board and return the BTT tech...maybe do some more research on the best options for me, potentially look into an Ender or even a resin printer...Appreciate all the help.

closed 2021-04-13 by adcallow

radek8, 2021-04-30:
SKR 2 really bad choice.

https://docs.google.com/document/d/1IeKgfE2WIDjqH1fx5Yg7n1FOHVwhDFmDlZ-7QMlOEV0/edit

gordo3di, 2021-04-30:
> SKR 2 really bad choice.
> 
> https://docs.google.com/document/d/1IeKgfE2WIDjqH1fx5Yg7n1FOHVwhDFmDlZ-7QMlOEV0/edit

Oh wow. Thanks for the link. Luckily I have (10) Rev B boards which is why I think I've haven't had any of the other problems people have mentioned.

adcallow, 2021-05-03:
@radek8 @gordo3di  HA. Thanks for the link, I will look at what I have running now. Fingers crossed it's the Rev B.

I bought an Ender 3 just to bypass all of the other crap that had been going on with my Anycubic. I installed the 2.0 and everything has been working fantastically...it is near silent and the prints are measurably better than on my Anycubic...not knocking them, just saying that everything was so much easier with the Ender 3. I'm now printing off of the Ender 3 for my "fun stuff" and then trying to uber customize my Anycubic (changing to direct drive, adding new 2209s, raising the Z dimensions, moving all the internal components to an external workstation, etc.) just for fun and to see what I can make of it...knowing that I'll probably end up just pushing it back to stock eventually when nothing works hahaha, but I'm having fun with it because I have another reliable printer going and I have zero expectations. 

Do you guys have any thoughts or customizations that you think are cool or that I should be looking into?