bigtreetech/BIGTREETECH-TouchScreenFirmware #1811: [FR] Double (or more) Z axis stepper TMC2208/09 configuration

RotorVulura, 2021-04-08:
**Is your feature request related to a problem? Please describe.**
Menu item under stealthchop and current (Amps) regulation and steppings, shows just one Z stepper.
Thinking that it was hooked to the "first" Z stepper, but it is not (asssuming you will have two or more identical motors).

**Describe the solution you'd like**
 I would like to easly see and set status/configuration from TFT. I am lazy :-)

**Describe alternatives you've considered**
I can set up all my Z stepper properly by command console, of course.

Many thans for reading this,
cheers

oldman4U, 2021-04-08:
Hi.

Maybe covered by #1671?

Please check

Thank you

RotorVulura, 2021-04-08:
Thank you very much, yes it is. I think that dual Z configuration will be enough. Who has multiple stepper all over the whole machine, can type console command easily:)
Getting back to noob makers (like me), dual Z stepper are very common now. BTT display and controller board fit perfectly as upgrade of those machine. Adding more axis just cause confusion in young makers but useless for advanced hackers. This is only in my opinion, of course. I will close this request looking for the other one. Thanks a lot, you are the man !
Cheers

oldman4U, 2021-04-08:
Only you have the permission to close the ticket. You have to do it in a web browser. The github app does not support it I believe.

Happy printing

RotorVulura, 2021-04-08:
Duplicated request. Please, follow the original one here : #1671 (just in case)
@oldman4U thanks for your time and patience

closed 2021-04-08 by RotorVulura

oldman4U, 2021-04-08:
U are welcome