bigtreetech/BIGTREETECH-TouchScreenFirmware #1812: Levelcorner.bmp missing

Cheo7811, 2021-04-08:
### Description

levelcorner.bmp is missing for all displays except TTF35 in The Round Miracle theme

oldman4U, 2021-04-08:
Hi Cheo7811.

Please use the Unified Menu Theme at the moment.

@Acenotass
@AntoszHUN
@bepstein111

Could you please check your Themes. Guess there is more than one template wrong, because there should be 141 where there are 139 bmp's in your Themes.

Thank you!

AntoszHUN, 2021-04-08:
> 
> 
> Hi Cheo7811.
> 
> Please use the Unified Menu Theme at the moment.
> 
> @Acenotass
> @AntoszHUN
> @bepstein111
> 
> Could you please check your Themes. Guess there is more than one template wrong, because there should be 141 where there are 139 bmp's in your Themes.
> 
> Thank you!

Thank you for reporting!

I uploaded everything.

By: AntoszHUN

oldman4U, 2021-04-08:
Hungary - 100 points;-) Thank you

oldman4U, 2021-04-08:
Please use level_corner.bmp as filename and not level corner.bmp

Thank you

closed 2021-04-09 by bigtreetech

radek8, 2021-04-09:
Icons are added.

oldman4U, 2021-04-09:
@bepstein111

Hi Ben. Will you be able to provide the two missing icons this weekend?

Please let us know

Thank you

oldman4U, 2021-04-13:
Seems bepsetin111 is mot available anymore. So would someone else take over his theme or should we remove it? I am writing the new ReadMe right now, so I can remove it tonight with the PR.

Just let me know in case someone takes it over.

Thank you

bepstein111, 2021-04-13:
Sorry guys, just been super busy lately. Should be able to get to this tonight hopefully.

oldman4U, 2021-04-13:
Cool. Thank you!

oldman4U, 2021-04-15:
Thank you bepstein111;-)

PR #1816 (2021-04-09): New icons for Round Miracle theme
@Acenotass if you create better icons you can change them.
I created these to make your great theme usable.

resolves #1812