bigtreetech/BIGTREETECH-TouchScreenFirmware #1814: [BUG] RRF macro menu broken

bttguy, 2021-04-08:
### Steps to reproduce

1. Update to one of the fw versions after the stable release
2. Go to the macro menu
3. Try to execute a macro or go back from the macro menu

**Expected behavior**
It was working as expected before the stable, macros worked, so the back button.

**Actual behavior**
Since the stable update, the macro menu for RRF users is broken. It lists the macros, but the back button does not work, only a reset takes back to the status screen. Also the macros dont work

oldman4U, 2021-04-08:
Can you try to find out which PR finally broke it.

bttguy, 2021-04-08:
Looks like #1769

oldman4U, 2021-04-08:
Have you tested the release before and after?

bttguy, 2021-04-09:
An another user reported in #1791 that it might be it. But I tested it right before and after that commit, but it worked for me. 
I will try to test it further, but there are a lot of commits to check.

PaulG4H, 2021-04-09:
@bttguy 
in this commit from 23th of March the macros menu still works:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/318c98d7a4360cef0cf1696b51d95fadcf87a198

and the next build on April 1th it's not working anymore:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/201cd68e399c8788c23edefd61749d80ce0e6b72

oldman4U, 2021-04-09:
And when you use 1d601b6091847645ea51e0ccf038b4a4258d83c7 ?

oldman4U, 2021-04-09:
@guruathwal Seems your PR broke this. Could you please check.

c4439452cf3e5857fac66fe7812787eed4094667

Thank you

guruathwal, 2021-04-10:
ok I will look into it.

guruathwal, 2021-04-10:
@bttguy  Can you provide the response from RRF firmware for `M20 S2 P"/macros"`?

PaulG4H, 2021-04-10:
@guruathwal 

10.4.2021, 09:41:44 | M20 S2 P"/macros"{    "dir": "/macros",    "first": 0,    "files": [        "deployprobe",        "resetprobe",        "retractprobe",        "selftestprobe",        "True Bed Leveling",        "calibrate z offset with bltouch step 01",        "calibrate z offset with bltouch step 02"    ],    "next": 0,    "err": 0}
-- | --

bigtreetech, 2021-04-10:
Hello, https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1825 should fix. Please test,  Thank you very much for your time!

bigtreetech, 2021-04-10:
> @bttguy Can you provide the response from RRF firmware for `M20 S2 P"/macros"`?

Hello, Please review #1825

bttguy, 2021-04-10:
I will check it in the evening, than close the ticket accordingly :) Thank you the quick fix!

guruathwal, 2021-04-10:
@bttguy @PaulG4H can you check if #1825 works?

PaulG4H, 2021-04-10:
@guruathwal @Msq001 
Macro menu work again!

Thank's for fixing this!

oldman4U, 2021-04-10:
Please help the community and close the ticket once you do not need it anymore. Thank you

bttguy, 2021-04-10:
Thanks for the fix, closing ;)

PR #1825 (2021-04-10): fix & improve RRF Macros
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

### Benefits

### Related Issues
fix #1814
closed 2021-04-10 by bttguy