bigtreetech/BIGTREETECH-TouchScreenFirmware #1817: [BUG] (LevelCorner.bmp is missing in Hybrid Red Menu. Firmware wont update past this missing bmp)

ShaneZampire, 2021-04-09:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-04-09:
Please. We are aware and there is already another ticket related to the same issue. The buttons are missing for ALL themes not only for the Hybrid Red, except for the Unified Menu Theme. Please use this Theme for now until we fixed the issue and close this ticket.

Thank you

radek8, 2021-04-09:
Icons are added.

radek8, 2021-04-12:
Please close the ticket

digant73, 2021-04-12:
@bigtreetech you could close this ticket. All the icons for the Red theme are present

oldman4U, 2021-04-12:
Use @Msq001 instead of bigtreetech

closed 2021-04-13 by bigtreetech