bigtreetech/BIGTREETECH-TouchScreenFirmware #1822: [Q] (Why the Nozzle and bed temperatures not updating in touch mode of TFT35 )

marouane86, 2021-04-09:
Recently I configured TFT35 with skr 1.4 turbo, everything works fine exept the tempertures of the nozzle and bed are not refreshed to follow the targets in touch mode. Any idea please ?

oldman4U, 2021-04-10:
Using TFT firmware v27? Before and during print?

radek8, 2021-04-10:
uncomment
AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)

manurafal, 2021-04-10:
I have the same problem.
I have uncommented the option AUTO_REPORT_TEMPERATURES (in Configuration_adv.h) but I still have the same problem.

I have SKR E3 TURBO and TFT35 v3 E3

marouane86, 2021-04-10:
Yes I use V27, and I uncomment AUTO_REPORT_TEMPERATURES (in Configuration_adv.h) but still the same issue

oldman4U, 2021-04-10:
Before or during print?

Printing from?

marouane86, 2021-04-10:
Thank you for your replies,
I fixed the issue by comment AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)

oldman4U, 2021-04-11:
Please help the community and close the ticket once you do not need it anymore. Thank you and happy printing

closed 2021-04-12 by bigtreetech