bigtreetech/BIGTREETECH-TouchScreenFirmware #1823: I need help with the filament sensor on skr 1.4 turbo and tft 3.5 v3.0[BUG] (short description)

Sebaspupi, 2021-04-10:
I am trying to put a filament sensor to my printer, it is the same sensor as the prusa mk3, when the screen is in marlin mode it works perfectly for me, it performs the filament change and it prints without problems, the problem comes when I want to use the screen In tft mode the filament sensor does not detect me, I have the sensor connected to the skr and I have already verified that all the settings are activated.

Someone, to give me a hand?

oldman4U, 2021-04-10:
Have u tried to add M75 to your start gCode?

radek8, 2021-04-10:
uncomment
HOST_ACTION_COMMANDS (in Configuration_adv.h)
HOST_PROMPT_SUPPORT (in Configuration_adv.h)

Sebaspupi, 2021-04-10:
@radek8 They are already uncommented

Sebaspupi, 2021-04-10:
@oldman4U Do I have to add a G m75?

oldman4U, 2021-04-10:
This is what I have been told.

Personally I tested (smart) filament sensor connected to TFT printing from TFT and on board sd using touchscreen successfully.

Sebaspupi, 2021-04-13:
everything is the same. only works in marlin mode and not in tft mode

oldman4U, 2021-04-13:
So also with M75 in start gCode?

npa62, 2021-04-15:
@oldman4U : Would you mind sharing what exactly you have done to make the SFS work in TFT mode?
I have followed every guide imaginable, but still can't get it to work in TFT mode. 
I even got as far as suspecting the SFS to be DOA, so I decided to buy a new. But still no luck.

oldman4U, 2021-04-15:
@npa62 

Because there have been so many reported about filament runout not working, I really expected a worst case scenario. But using the latest bug fix Marlin and the latest (at least v27) TFT firmware, it simply worked.

I connected the filament or smart filament sensor to the Fil-Det connector and activated On or Smart in the TFT firmware. And it simply works like expected. In case there is a filament in the sensor it prints. After removing the filament it triggers immediately (or after a few seconds using the smart sensor) the filament runout error, pauses the print and after inserting the filament and pressing resume it resumes the print.

What is the problem exactly on your side?

npa62, 2021-04-16:
@oldman4U 
Does it matter if you enable SFS from the menu or the config file?
Does the on board sd, mean the sd on the TFT or the SKR?
My main problem - I think - is a far too trigger happy SFS. I.e. even when filament runs through the sensor, it triggers with an out of filament message.
Once that is sorted, I may also have to look into the way my SFS is mounted. Right now it is mounted on the gantry with filament running to a Bondtech extruder without a bowden tube in between. I guess that is also going to cause me some problems.

oldman4U, 2021-04-16:
npa62
It does not matter where you enable the SFS, you just have to define smart and not on.

Yes, on board means SKR and there is the TFT SD and the TFT USB.

I do not use the sensor daily but bought it to be able to test it. For this I just used a 20cm long filament, which I moved slowly in and out of the sensor. Finally all that it needs is the wheel inside to move. How fast this happens and in which direction does not matter. To trigger the filament runout, all you have to do is to stop this movement and after a few seconds you should see the filament runout message on screen.

Try it and let me know and make sure to use the latest version of Marlin bugfix and TFT firmware (v27)

Sebaspupi, 2021-04-17:
Hello,

Something funny happened to me, I no longer touched the printer and today I started printing, deactivate the filament sensor of the tft (I did not touch the marlin and I have the sensor connected to the skr), I ran out of filament and printed it It stopped, I managed to detect the filament ahahaha.
It is absurd that it works without wanting to, I am going to try and to tell you how it goes, it should be noted that I have m75 at the beginning of the gcode as recommended by a colleague

Sebaspupi, 2021-04-17:
sorry if some things are not understood, my native language is Spanish and I use the translator.

oldman4U, 2021-04-18:
@Sebaspupi

Do I understand it right that you had the filament sensor activated on the TFT and connected to the SKR?

Sebaspupi, 2021-04-18:
no, the tft option is disabled and the sensor is connected to the skr  @oldman4U

oldman4U, 2021-04-18:
But before you had the filament sensor activated on the TFT?

Sebaspupi, 2021-04-18:
yes, and activated it did not work now that I have it deactivated it works

oldman4U, 2021-04-18:
Makes sense. You have to connect the sensor to the TFT in case you have it activated on the TFT.

In case you have it connected to mainboard you must deactivated the sensor on the TFT and activate it in Marlin.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-04-18 by Sebaspupi

oldman4U, 2021-04-18:
Happy printing

npa62, 2021-04-20:
@oldman4U 
Sorry for not replying before now, but I have been away from my printer.
Were you able to test the functionality of the SFS without actually printing something?

oldman4U, 2021-04-20:
Yes and no. I used a gCode file which includes a real print but with low temperature for nozzle and bed. I finally removed the filament and did a "dry" print so I could test the SFS without wasting energy and filament. Hope this answers your question.

You say you get false filament runout messages. Is this right when the print starts or in the middle?

npa62, 2021-04-20:
I will try that then.
It can be both at the start of the print and in the middle of prints.
As I tried to explain, the SFS is mounted on the left side of the top gantry, and when the extruder/hotend travels, especially from right to left, there is a loom of filamament hanging between SFS and extruder/hotend and not a lot of movement of the filament itself.
My plan is to put a (long) bowden tube in between, but right now my focus is to get the SFS working at all.

oldman4U, 2021-04-21:
It is possible that the reliability of the sensor changes with its position. For this you have settings in the firmware which you can change.