bigtreetech/BIGTREETECH-TouchScreenFirmware #1824: [BUG] UBL Grid Size is 10

Airconaaron, 2021-04-10:
### Description

Currently under my own personal marlin config I set my max grid size to 15 as follows 
```
  #define MESH_INSET 10              // Set Mesh bounds as an inset region of the bed
  #define GRID_MAX_POINTS_X 15      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```

However the TFT 35 UBL probe appears to only look at a 10x10 grid. I see that the latest `configuration.h` under `src/` folder has a 15x15 max grid, perhaps the latest release just doesn't contain these fixes? 

### Steps to reproduce

1. Set 15x15 grid in marlin for board
2.Enable and use UBL on TFT 35 LCD 
3.  Save mesh to s1, board and screen crash.
4. View mesh in terminal via `G29 L1; G29 T`, see that output grid is only 10x10 

**Expected behavior**
I expect that saving my grid will not crash my screen.
I expect that the output from a `G29 T` will be a 15x15 Grid

**Actual behavior**
`G29 T` outputs a 10x10 grid, board crashes during UBL. 

### Hardware Variant
TFT 35
### TFT Firmware Version & Main Board Firmware details
x.x.27, Marlin 2.0.7.2 SKR mini v2

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

digant73, 2021-04-10:
Firstly, I think that you could not have reset the EEPROM. If G29 T provides a 10x10 grid size it can mean that the previous Marlin fw was configured with a 10x10 UBL grid size and that a 10x10 grid was saved on EEPROM.
Second, I think that you could have reached the resource limits on Marlin fw. with a 10x10 UBL grid, my Marlin reached 94% of RAM and 91%  FLASH. The fw had a lot of bugs (crash starting a print from onboard SD, pause and M600 not working, stuttering of the nozzle etc...).
Try to reset EEPROM first. If it doesn't work, simply avoid to configure a 15x15 UBL grid size on Marlin. The max grid size on TFT fw was extended to 15x15 in the recent past

oldman4U, 2021-04-13:
What is the status of the reported issue?

oldman4U, 2021-04-20:
@Msq001

This user does not reply to his own ticket and digant could not verify the reported issue in his own code.

Please check and close it.

Thank you

closed 2021-04-20 by bigtreetech