bigtreetech/BIGTREETECH-TouchScreenFirmware #1826: TFT35 V1.2 after bootloader update screen shows "illegal user App1!"

svsergo, 2021-04-11:
After replacing the processor, is required a bootloader for TFT35 V1.2. Filled in the TFT35 V2.0 loader gives the error "illegal user App1!". Please help with the bootloader version for TFT35 V1.2.

radek8, 2021-04-11:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/197
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/139
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/24

oldman4U, 2021-04-11:
And https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Bootloaders/TFT35_V2%20Bootloader%20fix

and not to forget the **Read me before posting a New Issue**, which mentions the same and which is right above the New Issue button which was pressed to make this ticket.

bigtreetech, 2021-04-12:
Hello, Do you mean that your TFT35 V1.2 has replaced the MCU and flashed the bootloader of TFT35 V2.0, resulting in the failure to update the firmware of TFT35 V1.2 normally. So you need the bootloader of TFT35 V1.2?
If so, please contact your seller for a solution. No bootloader is provided now.

svsergo, 2021-04-12:
All right MCU replaced, requires bootloader for TFT35 V1.2. or recommendations on how to fully flash the MCU with a working program through the programmer.

radek8, 2021-04-12:
Someone who has the same display and would be willing to copy the botloader from their display could help you.

radek8, 2021-04-12:
The bootloader has already been exposed, unfortunately the link no longer works.

https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/issues/10

svsergo, 2021-04-12:
The work was restored using a full dump from the working BIQU_TFT35_V1.2.
The full dump of the BIQU_TFT35_V1.2  version can be obtained here: https://yadi.sk/d/B0S4vr6RUF6XLQ

closed 2021-04-12 by svsergo