bigtreetech/BIGTREETECH-TouchScreenFirmware #1827: Cannot install firmware on MKS TFT28 V2

Buggy123, 2021-04-11:
# Summary
I am trying to install this firmware on a MKS TFT28 V2 touchscreen. The TFT does not finish installing, it freezes at 100% updated for a indefinite duration (~15 minutes tested).

# Details
I followed the installation instructions carefully using the latest version of all files on the master branch. The SD card looks like this before installation:
[image]

After inserting the SD card and powering on the touchscreen, it beeps once, and then begins updating. The update progress bar reaches 100% and then halts without any further progression:
[image]

# Additional Notes

The config.ini is slightly modified; I'm currently using the config_rrf.ini file (renamed), as it seemed to more closely match the settings necessary for my printer, although I don't know what a couple of the differing settings do. I also changed the number of hotends to 2, for the same reason. I have tested it with the normal config.ini file, and it gives the same results.

The white line seen in the above picture is a physical defect; it is always present and has nothing to do with the firmware.

I have let it sit for at least 15 minutes during testing to ensure it does not progress further after enough time.

For simplicity and troubleshooting, I have completely removed the touchscreen from the printer. It also has no peripherals of any sort attached; the only connection is a USB cable connected to my computer for power.

The touchscreen can successfully be updated with the native MKS firmware and functions normally while using it.

The printer is a BIBO 2, which uses a MKS Gen L V1.0 board and also comes with a MKS Wifi TFT V1.0 peripheral for the touchscreen.

digant73, 2021-04-11:
MKS TFT28 V3.0 and v4.0 are supported. V2.0 is not supported

oldman4U, 2021-04-11:
According to the config.ini, the following TFT are currently supported:
#--------------------------------------------------------------------
# Supported TFT Variants
#--------------------------------------------------------------------
#
# BIGTREE_TFT35_V1_0    / BIGTREE_TFT35_V1_1 / BIGTREE_TFT35_V1_2
# BIGTREE_TFT35_V2_0    / BIGTREE_TFT35_V3_0 / BIGTREE_TFT35_E3_V3_0
# BIGTREE_TFT35_B1_V3_0 / BIGTREE_TFT43_V3_0 / BIGTREE_TFT50_V3_0
# BIGTREE_TFT70_V3_0    / BIGTREE_TFT28_V1_0 / BIGTREE_TFT28_V3_0
# BIGTREE_TFT24_V1_1    / MKS_TFT32_V1_3     / MKS_TFT32_V1_4
# MKS_TFT28_V3_0        / MKS_TFT28_V4_0

Please help the community and close this ticket.

Thank you

radek8, 2021-04-11:
Hello
if it is
MKS_TFT32_V1_3 compatible with MSK_TFT28_V3_0
and
MKS_TFT32_V1_4 compatible with MSK_TFT28_V4_0,
will not the MKS_TFT32_V1_2 display also be compatible with MKS_TFT28_V2_0?

Try using this file to update.
https://github.com/Uz45/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/MKS_32_V1_2.26.x.bin

Also use icons from this store. They will correspond better with the bin file
https://github.com/Uz45/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

Buggy123, 2021-04-12:
That's better - it progresses to the custom update screen/progress bar, but it's not quite there yet. All of the text looks like this:

[image]

I'm guessing this is the "update failed" screen. I'm not sure what this issue is, exactly... something with the fonts, maybe? I don't suppose you recognize it?

Buggy123, 2021-04-12:
Figured it out!

[image]

Turns out the installation for that older version is slightly different - you aren't supposed to rename the MKS folder. 
At first, the touch screen wasn't responsive - turns out I needed to trigger the calibration process by putting a blank "reset.txt" on the sd card. It seems fully functional now, though I haven't hooked it up to the printer yet.

Either way, thank you!

closed 2021-04-12 by Buggy123

oldman4U, 2021-04-13:
@radek8

Will you add the firmware for those MKS displays to the precompiled firmware which is already available? If so, I would add them to the ReadMe.

Please let me know

Thank you

radek8, 2021-04-13:
@Buggy123 
can you please try the attached bin file with current icons?
[MKS_TFT28_V2.0.27.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6306829/MKS_TFT28_V2.0.27.zip)