bigtreetech/BIGTREETECH-TouchScreenFirmware #1830: TO DO List [FR]

bigtreetech, 2021-04-12:
## What we should do next

### Bugfix
 - [ ] Klipper 12864 mode #290 
**progress:** The API Application of the upper layer is basically implemented in #705 . In my test, the problem occurs in the SPI slave part of the bottom layer. The SPI data received from klipper will shift by one bit. 
 - [ ] RGB Printing heating color not change #1786
**progress:** In fact, it is not a really bug, It's a known limitation. Marlin will activate the LED change only after receiving the "M190/M109" gcode, but this gcode will cause the communication interruption between the TFT and Marlin (because Marlin will not respond to other gcode before reaching the target temperature), so TFT will convert this gcode to "M140/M104" to avoid communication interruption, but Marlin will not be able to activate the LED change normally.

### Improve
 - [ ] Restructure `platformio.ini` File, using the relying on expansion to avoid redundancy
 - [ ] Restructure `icon/list`/etc... menu API (and related touch API) provides a more general API interface
 - [ ] Neopixel header LED number #1764
 - [ ] Machine RGB setting custom color value #1687

### Development direction
 - [ ] Vertical Screen
 - [ ] CNC Menu #1750

oldman4U, 2021-04-13:
I would add "Developers wanted" to the headline.

digant73, 2021-04-13:
I can implement the machine RGB settings custom color.
My idea is to to provide in the current RGB menu:
1) a standard color select button. A list menu reporting standard colors will be used to select the color
2) a custom color select button. A new menu to select the RGB components will be available to set color. The menu will be usable as general RGB color pickup. A proper callback to handle the selected color will be registered. E.g. a callback to set the printer LED color etc...
3) a button to switch between standard/custom selected color
4) a ON/OFF button to switch on/of the printer LED

digant73, 2021-04-16:
could it OK as menu to select a custom LED RGB color?
I will add also the W, intensity and index for NEO Pixel

[image]

oldman4U, 2021-04-16:
Can you click on the value to get a keypad?
Can you see the resulting colour somewhere (approximately;-))?

digant73, 2021-04-16:
yes for both questions

sinole1987, 2021-04-24:
can you add mid print nozzle freezing to the to do list. Its a very big issue and never got fixed. last night I had another freez. It mostly happen on long prints.

digant73, 2021-04-29:
wasn't "Restructure icon/list/etc... menu API (and related touch API) provides a more general API interface" recently implemented by Gurmeet with his last merged PR?

guruathwal, 2021-05-22:
> wasn't "Restructure icon/list/etc... menu API (and related touch API) provides a more general API interface" recently implemented by Gurmeet with his last merged PR?

not completely, I am still working on it.

bttguy, 2021-05-29:
@bigtreetech Can I ask you to include some RRF related improvements listed in #1791? I can help you with testing :)

bigtreetech, 2021-06-01:
> @bigtreetech Can I ask you to include some RRF related improvements listed in #1791? I can help you with testing :)

Hello, It is being implemented in #1962

guruathwal, 2021-07-31:
Dear Stale bot, please don't mark this one as stale. 😂

Minims, 2021-09-09:
Hello, is it possible to add a moonraker api compatibilty layer to use the TOuchscreen mode on klipper, like paneldue or the ender3v2 screen ? Thx!

sarvenn, 2021-11-08:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

@oldman4U . Topic is about to be closed.

oldman4U, 2021-11-08:
Hi sarvenn.

I am not the owner but such tickets are automatically reopened.

sarvenn, 2021-11-09:
> Hi sarvenn.
> 
> I am not the owner but such tickets are automatically reopened.

Oh, sorry. :) I missed that. You are the first one came up in my mind for this platform 😃

@bigtreetech for your information.

yannick-alch, 2021-12-14:
> can you add mid print nozzle freezing to the to do list. Its a very big issue and never got fixed. last night I had another freez. It mostly happen on long prints.

I have the same problem on long print. The nozzle freeze into the piece, and motor are still lock and temperature are kept. Maybe the screen didn't received the "ok" response from the motherboard ?
Is there a way to resent the "ok" message after an amount of time ?

digant73, 2021-12-27:
@bigtreetech Event LED feature could be now marked as done. Also PR #1209 could be removed. All the features on that PR have been implemented.

sinole1987, 2021-12-27:
> > can you add mid print nozzle freezing to the to do list. Its a very big issue and never got fixed. last night I had another freez. It mostly happen on long prints.
> 
> @sinole1987 I have the same problem on long print. The nozzle freeze into the piece, and motor are still lock and temperature are kept. Maybe the screen didn't received the "ok" response from the motherboard ? Is there a way to resent the "ok" message after an amount of time ?

you need to make a sleeved cable out of a usb cable and use the braded jacket for ground pin and 24 awg red cable for power. after that you need to wrap it in ferrite beads on both ends. It will get much better.

digant73, 2022-02-28:
please stay with us

sinole1987, 2022-05-25:
Not the todo list :)

thisiskeithb, 2022-05-26:
@bigtreetech You need to add the `TODO` label to this issue so stalebot doesn't close this.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/1e5955e76d764e830ca707508052275d31f2e95e/.github/stale.yml#L13-L18

Tannoo, 2022-06-23:
You mean "TO DO"? Or the title needs to be changed to "TODO".

oldman4U, 2022-09-21:
ToDo ;-)

xPew, 2022-09-22:
Just some more time dear bot!

oldman4U, 2022-09-22:
MAGIC!!!😁

guruathwal, 2022-09-22:
Mr Stale is the only person regularly maintaining this repo 🤣

oldman4U, 2022-11-24:
And me. 😁

closed 2023-02-19 by stale[bot]