bigtreetech/BIGTREETECH-TouchScreenFirmware #1831: [Q] What if I have no heated bed, where is the code to take out the icons and options of the heated bed

nobaj, 2021-04-12:
I have no heated bed on my custom printer, is there a way to disappear the heated bed icons/menus on the lcd?

radek8, 2021-04-13:
Icons and functions on the heated bed in the current FW can not be hidden.
They just won't have any function for you.
PS: does anyone else use a printer without a heated bed today? :-)

radek8, 2021-04-13:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1053

oldman4U, 2021-04-13:
On my printer I never used the part cooling fan and there is no LED. But I have a heated nozzle and bed.

😁

oldman4U, 2021-04-13:
To answer the question the user finally asked.

This software is open source so you can download it from this page.

Question. Are you a developer who could improve the code the way that it is easier to take away unneeded elements, even it looks like switzerland cheese at the end?

Or even better - to make a user interface designer which allows each user to build his own interface very easy?

Please let me know

Thank you

oldman4U, 2021-04-15:
Hello nobaj.

Would be great if you could answer your own ticket or close it in case you do not need it anymore.

Thank you

nobaj, 2021-04-16:
Hi, thanks, I will make some options on the configuration to deactivate the icon and put another like "no heatbed" so the ui does not get all messed up, when I finish i can make a pull request if you guys want to incorporate that feature

closed 2021-04-16 by nobaj