bigtreetech/BIGTREETECH-TouchScreenFirmware #1833: [Q] How should the notification screen work?

hapklaar, 2021-04-13:
I use M117 notifications from the Octoprint-DetailedProgress plugin to push progress info to my TFT35 screen. This works fine in recent releases. I was also expecting to find a log of these messages in the 'Notifications' screen after clicking this area on the status screen. However this screen is always empty for me.

Is this a configurable option I can enable somewhere, or a bug or maybe by design?

bigtreetech, 2021-04-14:
The `Notification` menu only disply `"Error:"`  message and `"echo:"` (only when `ACK notification style` is set to `POPUP` in `Feature` menu) message from Marlin

hapklaar, 2021-04-14:
Can I change this somewhere to also include the M117 (and possibly other) messages?

oldman4U, 2021-04-14:
Maybe you missed this and many other tickets the time you searched for M117 before you started this ticket: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1050

hapklaar, 2021-04-14:
@oldman4U This is not relevant. M117 messages already show correctly in the notification area, this is not my complaint. They are just not shown in the separate notification menu.

oldman4U, 2021-04-14:
My fault. Than btt gave you the answer already.

hapklaar, 2021-04-14:
My question remains if I can change this in code to allow M117 notifications to also be kept there. This filtering can hopefully be configured somewhere.

oldman4U, 2021-04-14:
It is open source so you can change it the way you want it. But there is no on/off option for this.

Are you a software developer?

hapklaar, 2021-04-14:
No I'm not, that's why I ask the devs of this software. Your additions don't really help here....

bigtreetech, 2021-04-15:
@hapklaar Hello, Can you send me a picture of what M117 is currently displayed on the screen. So that I can quickly find the location to be modified.

hapklaar, 2021-04-15:
Hello @bigtreetech,

For example I type the following in OctoPrint terminal: "M117 This is an M117 message"
Terminal output is:
[image]

I see this on the TFT35:
[image]

Then I press the notification area to open the notification screen, where I would like to keep a (short) log of recent M117 notifications:
[image]

bigtreetech, 2021-04-16:
@hapklaar Hello. You can temporarily change line 232 in `parseACK.c` from
 `addToast(DIALOG_TYPE_INFO, dmaL2Cache + index);` 
to
 `addNotification(DIALOG_TYPE_INFO, (char*)echomagic, (char*)dmaL2Cache + index, true);`
 so that you can achieve the effect you want.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/6968e53ad3e99eda523dcac57e86bb0d301395e1/TFT/src/User/API/parseACK.c#L232

[image]

I will sort out the code in this part, because the general response protocol and host action protocol are divided into two parts, and I will try to integrate them

hapklaar, 2021-04-16:
Thanks. I made the change and recompiled. Now M117 notifications do appear in the notifications screen, but only when I'm on that screen when the M117 notification comes in.

When I'm on the status screen, the message is not added to the notification screen. Can we make it so M117 messages are always added to the notification screen, regardless of what screen is active at that moment?

hapklaar, 2021-04-16:
I think I found a way to do it: I added this line just below line 225, before the status screen check: 

`addNotification(DIALOG_TYPE_INFO, (char*)echomagic, (char*)dmaL2Cache + ack_index, false);`

This seems to add all messages starting with ':notification' to both the status area and the notification screen, and doesn't give a popup for each message.

As I'm not a dev or programmer, I hope this is the right way to do it. ;)

radek8, 2021-04-17:
> I think I found a way to do it: I added this line just below line 225, before the status screen check:
> 
> `addNotification(DIALOG_TYPE_INFO, (char*)echomagic, (char*)dmaL2Cache + ack_index, false);`
> 

Great. it works for me.

closed 2021-04-20 by bigtreetech

radek8, 2021-04-21:
If PR #1864  is merged,
you can enable the display of M117 notifications on the notification screen in the cofig.ini file
notification_m117: 1

PR #1864 (2021-04-28): Notifikation M117
Display of M117 messages on the notification screen.
With the option to turn this view on or off in the CONFIG.INI file
Fix M569 error

resolves #1833 

[image]