bigtreetech/BIGTREETECH-TouchScreenFirmware #1834: [FR] Show current fan speed when using ADAPTIVE_FAN_SLOWING

hapklaar, 2021-04-13:
I enabled ADAPTIVE_FAN_SLOWING in Marlin firmware. When using emulated mode fan speed is updated to show when ADAPTIVE_FAN_SLOWING kicks in with a new value and asterisk. In touchscreen mode this is not visible.

REPORT_FAN_CHANGE is also enabled.

oldman4U, 2021-04-14:
This is a feature of Marlin.
Marlin is not very good in communicating with other hosts.
In case this would be implemented, I would expect a huge lag between the fan changes the speed and this information to be shown on the TFT, which would make the information useless.
In case this would be implemented, I would expect print issues because of the additional data which would have to be sent to the TFT.

I could be wrong because I am not a developer....

hapklaar, 2021-04-16:
I just found this issue has already been discussed in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/883

It looks like it has since been broken again, as I'm not seeing fan change updates in Touch mode, but I do see them in Marlin mode.

closed 2021-06-22 by stale[bot]