bigtreetech/BIGTREETECH-TouchScreenFirmware #1835: ESP32 Wifi

Maazshakir, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
No actually its an additional feature that i would like to have. BTT TFTs has EXP1 and EXP2 port for  connecting ESP32 wifi module and to be able to send 

**Describe the solution you'd like**
I think it would be great if we could add a wifi interface to BTT screen it will eliminate the need of octoprint .

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

sinole1987, 2021-04-14:
I think you can connect a  ESP32 to the same pinout for EPS01. I saw a tutorial on it before, but it was to the pinout on the SKR 1.4 board. screens have the same pin layout for ESP on it. I don't have a good experience with ESP3D though.

oldman4U, 2021-04-14:
I would never make my print depending on a stable WIFI connection. Octoprint relies on additional hardware for a good reason.

Please help the community and close the ticket once you do not need it anymore. Thank you

Maazshakir, 2021-04-14:
I know that esp 32 chip isn't as capable as octoprint but duet hardware is
relaying on it and it works flawless on duet hardware and their software
which means it's more then enough for 3D printing if used wisely

sinole1987, 2021-04-14:
> I would never make my print depending on a stable WIFI connection. Octoprint relies on additional hardware for a good reason.
> 
> Please help the community and close the ticket once you do not need it anymore. Thank you

I don't think ESP3D runs your prints, it just sends files to your SD card on your board. I think print runs on the main board regardless of eps being connected to board or tft screen. ESP3d also does some monitoring stuff and somehow you can connect camera to it which I didn't really explore. I might be wrong though , but when I disconnected it he print I started with eps kept running.

oldman4U, 2021-04-14:
This is how esp3d works. But it was my intention that we talk about printing over WIFI here and an Octoprint replacement which would be great but requires a bit more resources than available in such a project.

sinole1987, 2021-04-14:
> I know that esp 32 chip isn't as capable as octoprint but duet hardware is relaying on it and it works flawless on duet hardware and their software which means it's more then enough for 3D printing if used wisely
> […](#)

ESP 3D isn't as nice as RepRap firmware. If you have a SKR board I suggest you Run RepRap firmware on your SKR board, there is a way, you cant use it with TFT screens, There are screens developed by mellow for RepRap. watch these two videos you'll learn a bit more.
[https://www.youtube.com/watch?v=s68U6BwxHpI](url)
[https://www.youtube.com/watch?v=NynXQwTTFx0](url)

kisslorand, 2021-04-14:
The way ESP3D connects to the TFT has a very slow speed. I do not remember the exact numbers but it close do 9.6 Kbaud. Do not compare it to other types of connections that can go up to 100 Kbaud.

Maazshakir, 2021-04-14:
> the exact numbers but it is close do 9.6 Kbaud. Do not compare it to other

I don't want it to have esp3D firmware if possible i would like to have it
a firmware clone of duet firmware which adopts 500k/s

sinole1987, 2021-04-14:
> The way ESP3D connects to the BTT TFT has a very slow speed. I do not remember the exact numbers but it is close do 9.6 Kbaud. Do not compare it to other types of connections that can go up to 100 Kbaud.

is that the ESP01 limitation or connection limitation. I think you can increase the speed if you connect a esp32 to the same pins.

Maazshakir, 2021-04-14:
That's why I mentioned esp32 and not esp01 I think it's so stupid to use
esp01 for any thing since it's ultimate slow

sinole1987, 2021-04-14:
> I don't want it to have esp3D firmware if possible i would like to have it a firmware clone of duet firmware which adopts 500k/s
> […](#)

This is not a TFT screen related subject then, please watch the two videos I linked.

kisslorand, 2021-04-14:
> is that the ESP01 limitation or connection limitation. I think you can increase the speed if you connect a esp32 to the same pins.

I have both, limitation is from the connection/communication type chosen.

hapklaar, 2021-04-16:
Is there a way to have faster communication between ESP header and SD card? Having a web interface for the TFT screen where you could upload a gcode file to SD fast, and start the print remotely without needing OctoPrint would be a great addition.

sinole1987, 2021-04-16:
> Is there a way to have faster communication between ESP header and SD card? Having a web interface for the TFT screen where you could upload a gcode file to SD fast, and start the print remotely without needing OctoPrint would be a great addition.

It doesn't have to be TFT Screen, BTT introduced a new mini motherboard with onboard WIFI, Marlin also need to do some Updates on the platform. I Think new stuff will come out soon that will improve Wifi interface and Replace the need for octoprint and raspberry pie.

sinole1987, 2021-04-17:
There are a bunch of UART pins on back of the Screen I was wondering of you can connect the esp32 using those, although It will be rats nest behind the screen I feel like there s a possibility. you need 6 IOP pins a 3.3v pin and a ground.

luc-github, 2021-04-26:
Just to clarify : 
RRF use ESP12 board and SPI to communicate, RRF has specific SPI drivers that allow ESP board to access printer SD, when other Firmware Marlin / Smoothieware / repetier do not have/support it, that is why ESP3D use serial to communicate, that is why RRF upload is fast when the serial GCODE upload protocol is a disaster.

ESP01 do not share spi pins but even using ESP32 you will not get fast upload on Marlin like current RFF because the SPI drivers is not present.

About TFT file upload the problem is the same actually, there is no mecanism to share the SD card reader between TFT and ESP board

May be Marlin will add it for the BTT Marlin/RRF compatible board but I have a doubt as Marlin is still only supporting only DOS file names (8.3) for upload and print commands

sinole1987, 2021-04-26:
> Just to clarify :
> RRF use ESP12 board and SPI to communicate, RRF has specific SPI drivers that allow ESP board to access printer SD, when other Firmware Marlin / Smoothieware / repetier do not have/support it, that is why ESP3D use serial to communicate, that is why RRF upload is fast when the serial GCODE upload protocol is a disaster.
> 
> ESP01 do not share spi pins but even using ESP32 you will not get fast upload on Marlin like current RFF because the RRF SPI drivers is not present.
> 
> About TFT file upload the problem is the same actually, there is no mecanism to share the SD card reader between TFT and ESP board
> 
> May be Marlin will add it for the BTT Marlin/RRF compatible board but I have a doubt as Marlin is still only supporting only DOS file names (8.3) for upload and print commands
> 
> Additionaly ESP8266 code is super optimized, so the result is that currently ESP32 is actually slower than ESP8266 for web communications, this should change in future but currently it is the case.

Is that why they connect ESP32 to EXP1 and EXP2 of the SKR board to simulate a duet board? because some of those pins are used for SD card reader?

luc-github, 2021-04-26:
>Is that why they connect ESP32 to EXP1 and EXP2 of the SKR board to simulate a duet board? because some of those pins are used for SD card reader?

May be I am not up to date but last time I checked Duet use ESP8266 not ESP32, and still is the reference in https://github.com/gloomyandy/RepRapFirmware/wiki
Code can be ported to ESP32 for sure in that case RRF and ESP32 would use SPI to communicate as it is RRF SPI diriver

Yes SKR can run the ported RRF https://github.com/gloomyandy/RepRapFirmware, because EXP2 and EXP1 have the SPI pins for another SD card reader, they can be used by ESP board and the SD card will be the onboard one.

sinole1987, 2021-04-26:
> May be I am not up to date but last time I checked Duet use ESP8266 not ESP32, and still is the reference in https://github.com/gloomyandy/RepRapFirmware/wiki
> Code can be ported to ESP32 for sure in [...]

So could the ESP01 issues be addressed on marlin? or there is a physical limitation on how those pins 28 and 29 communicate with chip as well?

luc-github, 2021-04-27:
ESP01 do not expose the SPI pins but GPIO 0 and GPIO 2, so the only possible solution is to use the MKS protocol or equivalent : https://github.com/luc-github/ESP3D/issues/575 - but if working as expected, speed will be around 110KB/s (same as MKS code in best case), not as  fast as the RRF one, but still far better from the slow like hell  GCODE/transfer protocol, be noted on some board only one pin is actually connected, it was confirmed with BTT team, that is why I did not work on it yet - I need to really think all possibilities.

If you use another ESP board then the SPI pin should be accessible but the SPI driver is not present in Marlin - so better use RRF FW as showed in video above. 

I do not handle communication / bug tracking with Marlin, I do not have bandwidth for that, and looking as long name support for SD files situation - I doubt anything is done to implement SPI RRF modification, unless someone take the ownership and do it, but community is big, so may be ?.

But if board support RRF why use Marlin ? RRF is more reliable and offer better performance no ? and community is also very active

sinole1987, 2021-05-18:
@bigtreetech since we are on TFT branch, I was wondering if we can change TFT serial protocol and when we connect ESP module to screen file transfer become faster. That way we don't need to use the second marlin serial port either.

closed 2021-07-24 by stale[bot]