bigtreetech/BIGTREETECH-TouchScreenFirmware #1836: Smart Filament Sensor and the Linear Advance

AntoszHUN, 2021-04-13:
Hello! Is there already a solution for Smart Filament Sensor to work in TFT mode using Linear Advance?

SFS is dead with Linear Advance enabled. Does not work. Several of my prints have hurt already.

Mady By: AntoszHUN

oldman4U, 2021-04-13:
Using the latest Marlin bugfix version which includes the LA fix and the latest TFT firmware should work.

Try it and let us know.

AntoszHUN, 2021-04-13:
> Using the latest Marlin bugfix version which includes the LA fix and the latest TFT firmware should work.
> 
> Try it and let us know.

What innovations are there? What has been fixed in TFT firmware?

oldman4U, 2021-04-13:
I hope it is OK for you that I send you the link to the related ticket which has all the information regarding LA, S(FS) and MARLIN: 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1794

AntoszHUN, 2021-04-14:
> I hope it is OK for you that I send you the link to the related ticket which has all the information regarding LA, S(FS) and MARLIN:
> 
> #1794

Good then I'll try. It would be nice if it worked the way it is advertised.

sinole1987, 2021-04-14:
> Hello! Is there already a solution for Smart Filament Sensor to work in TFT mode using Linear Advance?
> 
> SFS is dead with Linear Advance enabled. Does not work. Several of my prints have hurt already.
> 
> Mady By: AntoszHUN

for Linear Advanced you can change k value in the settings>machine>Parameter. if you want to change k via Gcode M900 will not work by itself for some reason. you need to add m117 as well. for example for my PETG Gcode I have following:
; Filament gcode
M900 K0.09
M117 K0.09

sinole1987, 2021-04-14:
> Using the latest Marlin bugfix version which includes the LA fix and the latest TFT firmware should work.
> 
> Try it and let us know.

I didn't know people have LA issue with marlin for some reason TFT picked UP M117 and used it for me.

oldman4U, 2021-04-14:
You missed the point of this ticket. It is primary about the SMART filament sensor.

AntoszHUN, 2021-04-14:
> You missed the point of this ticket. It is primary about the SMART filament sensor.

If SFS and linear advance work together, I close the error ticket. It turns out tonight. I'll put up the new Marlin. Need to change something in the display firmware?

oldman4U, 2021-04-14:
Makes sense.

Happy testing

AntoszHUN, 2021-04-14:
Thanks! Works well! Perfect. The only problem is that after loading, the material I squeeze is pushed onto the object. Where he continues. It works by closing the ticket.

closed 2021-04-14 by AntoszHUN

oldman4U, 2021-04-14:
Great.

Where does it purge material on the object? Printing from ?

AntoszHUN, 2021-04-14:
> Great.
> 
> Where does it purge material on the object? Printing from ?

Yes. Exactly on the material. Printing is from an SD card.

oldman4U, 2021-04-14:
And the SD card is in the Mainboard or the TFT. And the sensor is connected to the mainboard or the TFT?

AntoszHUN, 2021-04-14:
> And the SD card is in the Mainboard or the TFT. And the sensor is connected to the mainboard or the TFT?

The SFS is connected to the TFT. SD card is used in TFT.