bigtreetech/BIGTREETECH-TouchScreenFirmware #1837: [BUG] TFT35 - Busy processing, please wait.

Magiquall, 2021-04-13:
Hello,
I have the same problem as another post closed without really solution, with my BTT TFT35, for the moment connected to a MKS SGEN L V2 while waiting for my SKR-2, I have the message "busy processing, please wait "

I reduced the speed from 250,000 to 115,200 it was less frequent, but still present.
I swapped out the COM ports and had no more worries, until I printed. A one hour thirty minute print and the printer is frozen 3 layers from the end.

When it is frozen, it is not easy to put it back into service, you have to turn it off and then wait a little before turning it back on, often twice in a row.

Every time it was frozen there was "Processing busy, please wait" at the top and in red, but nothing at all, the screen gets stuck even after several minutes and as during printing it continues to heat up. have a hole in my impression.

I specify that I put the last update, the Marlin patch and the last firmware available on the github for the screen.

Is it a loss of connection with the motherboard, or the screen which is overloaded?

oldman4U, 2021-04-14:
Hi.

There has been no solution mentioned in this repository, because literally all users do not have the reported issue, but a few, and we are all using the same firmware. This makes it very unlikely that the firmware of the TFT causes this issue and this repository is about the firmware of the TFT's only. In case there is a way to reproduce the issue and there is a way to fix this in the TFT firmware, I am sure it will be done.

Magiquall, 2021-04-14:
Ok, I hope this is from the MKS SGEN L V2 board, because I stop the makerbase. I also had a problem with their screen, maybe this is the same source of the problem.
So I will wait for the SKR2, hoping that it fixes the problem and that it does not come from the BTT screen.

Is the memory of the BTT TFT35 display sufficient where you need to be careful?
For example, do not enable all of the features in the config.ini file.

Or are there functions that should not be activated together? Functions which could interfere with the operation.

In any case, thank you for your responsiveness.

And excuse me for my English.

oldman4U, 2021-04-14:
SKR-2 is for sure a great board, but Marlin support is not finished as we speak, as far as I know.

Resources are very limited on the TFTs but so far you can activate all functions in config.ini. If this makes sense - I do not know;-)

Most of us are not native English - so no worry.

Magiquall, 2021-04-14:
I got the github firmwaire for the SKR-2, I managed to compile it. Otherwise, I would not have ordered this card.
But as it becomes difficult to find an SKR 1.4 turbo on its own, I opted for the SKR2.

I looked a bit at the code and indeed there are things added in this BTT version compared to the official marlin, but the processor already existed on another card that was already released. Adaptation should therefore not take too long. Finally, I hope.
What interested me was not having to cut the diag pin of the tmc 2209 pilots. I don't like it.😭😭

Guilouz, 2021-04-14:
@Magiquall Where do you buy SKR2 ?I can't find it anywhere.

oldman4U, 2021-04-14:
Great board. Mixture of SKR v1.x and Pro/GTR plus some nice add ons both do not have.

oldman4U, 2021-04-14:
@Magiquall

Please help the community and close the ticket once you do not need it anymore. Thank you

Magiquall, 2021-04-14:
Excuse me oldman4U, but do I have to close the ticket now? If I have the same problem with the SKR-2 I will have to recreate another identical post. The card is on its way, but I actually don't have a precise date when it arrived. However, I still had hope to be able to work with the MKS while waiting for what is not possible by the BTT screen.
Tomorrow I will do a test with Octoprint.

oldman4U, 2021-04-14:
Once you do not need it anymore.

Happy printing

Magiquall, 2021-04-15:
Hello,

I tested a long 5h30 printout with Octoprint plugged into the motherboard's USB (MKS).
The screen was connected and turned on too, but I didn't touch it at all during the entire print.
The screen is crashed and it is set to "Busy processing ..." during the 1st layer. 1st layer long enough, because I was printing a box. But he no longer received information from the printer, I no longer had the X, Y and Z co-ordinate news.

Octoprint continued to the end. Once finished I touched the screen and it was not frozen, I was able to navigate the menus (I have not tried them all). But I was in "System / connection" Then I clicked on disconnected. The message "Busy processing" has disappeared, unfortunately there is no connection button, but I tricked. I change the speed of the COM port then I put it back to 115200 baud, and it is connected. I was able to move the axes.
So it is really the connection of the COM port which is blocked, not the processor.
If the screen then freezes, it means that we are trying to communicate with the card and that it is surely waiting for a response.
 
The next print I will try Octoprint with the Marlin menu on the screen. But as it is an emulation I do not know if it is not yet communicating by the COM port?

bigtreetech, 2021-04-16:
@Magiquall Hello, Can you adjust the TFT menu to terminal to monitor the communication between TFT and motherboard in real time? So that we can see the communication information between TFT and motherboard after the screen is frozen.

Magiquall, 2021-04-16:
> @Magiquall Bonjour, Pouvez-vous ajuster le menu TFT au terminal pour surveiller la communication entre TFT et la carte mère en temps réel? Pour que nous puissions voir les informations de communication entre TFT et la carte mère une fois que l'écran est gelé.

Hello,

I don't mind, but when I go to the TFT menu / terminal I have to enter a Gcode command, with the virtual keyboard, I have to put what? Thank you

moebis, 2021-04-17:
@bigtreetech having the same issue, I noticed when I try to go to the menu to disconnect and then touch anywhere to reconnect, a bunch of gcode starts spilling on the screen with an "ok" button. (this is while I'm printing, which is when I need the screen to monitor temp and coordinates). After I close the last invalid window on the TFT35 it goes back to "Busy processing..." I also noticed you released 2 new firmwares on the same day (about 1-2 days ago) one in the morning and one later in the day, both had the same version number (BIGTREE_TFT35_V3.0_E3.27.bin), so I might try the newer one that was uploaded about 12-14 hour later.

Magiquall, 2021-04-20:
Hello,

I have the same problem with the SKR-2.
Even without doing anything, leaving the printer on, after a certain time (random) the display will still show "Processing busy. Please wait".
Sometimes when switching on the connection does not work. Forced to restart the printer and sometimes this has to be done multiple times. It is not manageable.

For the SKR-2, I used the firmware available on the bigtreetech github.

In the firmware I left the communication:
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200

I'm preparing a video, but as the switch to Busy processing mode is random, I don't know when I could shoot the 2nd sequence to show you.

Regarding the display in the terminal I would like a little more details, as I told you when I go to the terminal part, it asks me to enter a command, and if I do send without a command, I have the terminal which opens but nothing appears.
Please a little help.

oldman4U, 2021-04-20:
What else do you have connected to the Mainboard and TFT?

Which power supply do you have?

Magiquall, 2021-04-20:
Hello,

I'll show you a video, more exactly 2 video.
On ignition:
https://drive.google.com/file/d/11xBUbM7MYcW1djlhxD7xPpG1rrHl110b/view?usp=sharing
When I saw the fault:
https://drive.google.com/file/d/11z9huO4asRivbRq-oPNOrAlFMl2_5bIu/view?usp=sharing

I plugged in all that concerns the printer, but nothing more, I did not connect the Raspberry.
Regarding my power supply is an industrial power supply (I work in the field) 24V 240W (the bed being 220V), the regulation is perfect, really very little residual.

Magiquall, 2021-04-22:
Hello, no one to help me anymore , sniffff...(sob out) ;)

I made several attempts to configure Serial Ports.
By default in the firmware of the SKR2 available on this github, there are:
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1
and 115200 baudrate.

I tested several combinations with more or less conclusive results, some do not allow communication with the screen and others seem to improve, less crashes but still afraid of doing a long print because I still had crashes (Busy proces...) .

What would be the configuration for this part:
With the BTT TFT35, it connect on TFT port,  and EXP1 and EXP2 screen also connected to the screen (by the way, no problem when I'm in marlin mode), then the Raspberry also connected in USB or UART on the screen or elsewhere !! !.

For the moment I have tried with the raspberry, it works well, but during the print the screen crashes with "Busy processing", but hey it crashes even if nothing is done on the printer anyway .
But when I tried to make the screen work in "Touch" mode the Raspberry was still disconnected from the USB.

oldman4U, 2021-04-22:
I wrote it before and only got thumbs down for telling the truth but I repeat it here:

If you look for the solution in software, I believe you are looking in the wrong area. Issues like this and similar have been solved so far by:

1. shielding the cables
2. changing the power supply to something more reliable and better suited for the need of a 3D printer
3. one could solve a similar problem by moving the printer to another power line (household)
4. and I remember one user who could solve the problem by removing the electrical filament dryer from the same power outlet

So far I haven't seen a single user solving the problem in the firmware of the TFT. More likely it is a problem in Marlin, where several tickets are related to problems with the serial portS, in case one serial port is in troubles.

It would be important to be able to force the problem to happen, for example by disconnecting things from the mainboard or TFT which are connected there, or by turning electrical devices in the same household on and off. 

I have seen software changing settings itself, in case the software has been running on a computer driven by the power supply of a certain vendor - and other crazy things. If there is a problem in a chain, most of the time you can only see it at the end of the chain, and in this case the end is the TFT.

BTT probably sold hundreds if not thousands of each kind of TFT they offer, but "only" a handful of users has problems and all others do not have it, even they use the same hard- and software. So what is more likely, that the TFT is the problem or that something else causes this?

Btw. your printer looks great!

oldman

Magiquall, 2021-04-22:
Hello oldman4U,

Sorry it's quite long and with my English ... sorry :)

Being in the field of industrial electricity, battery charger, energy supplier in direct and alternating current, I am concerned with problems linked to the electromagnetic field. At work we have a room dedicated precisely to the measurement of these electromagnetic fields to certify CE conformity.

This is also why I separated the 230v alternating current and 24v direct current. But I have some faults in my setup on the subject on the left side so I will try to remedy it quickly so that the separation of the 230V is even further from the 24V. Because you're right, electromagnetic disturbances are sneaky and can cause random malfunctions. There is one place that may be complicated, it is the power cable of the BED with the cable of the temperature sensor. They are together under the BED so it is not easy to separate them.

Regarding the power supply that I use, it is an industrial power supply intended for telecom systems (very precise in this area) and CE validated precisely to avoid electromagnetic radiation, but it is true that it is not of our design and that it has not been tested with us. I will therefore also carry out tests with the power supply of the Ender 3, I do not need more power since my bed is 230V. But honestly, I have less confidence in the Ender 3 power supply than in my industrial power supply.
I am going to ask my boss if he would have a time to check the conformity of this industrial power supply. 
In addition, it is quite far from the screen, but you never actually know.

Then I also heard about COM errors in marlin on the bugfix version. I saw it in a post on an Octoprint forum. Apparently the stable version didn't have this problem, so I used the stable version when I was with the MKS SGEN_L v2, but the result was the same.

Regarding shielded cables, I completely agree with you, using a simple flat cable is really not ideal and using shielded cables would not be a luxury, but do you have any references?
I admit that concerning RS232 communication it is not really disturbed over such a short distance, unlike Ethernet which even with 10cm is declared defective with our testers if we use basic flat cable. In our systems we have transformers of several tenths of a kilowatt, therefore a potentially very strong radiation, and the "simple" communication cables are not disturbed more than its, but we are in RS232 so with a potential between -12V and + 12V, if there is TTL between our motherboards and screen, which is between 0v and 5V, the possibility of transmission errors is indeed much greater.

I will follow your advice for the moment;) and therefore separate the 230V and 24V even more. This would also indicate, if this is the case, that the cards are quite sensitive to this phenomenon. The CE certification is in both directions. We must not emit too much radiation but also be able to collect it without disturbing the system ;)

digant73, 2021-04-26:
@Magiquall Viewing your solution, the first thing I can say is it's a EMI problem. Did you test if the problem is present also not closing the drawer? also did you try to simply power on the system only via USB (so with no main power on) and verify if the problem is present after some hours?

oldman4U, 2021-04-26:
Driver and motor will work only with usb power!?

digant73, 2021-04-26:
no obviously he cannot print, But the TFT on main page periodically communicate with mainboard. I think in that situation, the connection will never be lost. Or he can also use the power supply with no print and probably he will have the problem after some time. Just preheat bed and nozzle and nothing more. Verify if soner or later the problem will be present

Magiquall, 2021-04-26:
Good evening, 
Indeed it is a try to make since it also crashed without printing.  
For the moment I cannot do a test because I have dismantled everything.  I redid my drawer and it is painted.  In order to have the 230v button on the left and the 24v and screen part on the right.  
My colleague, in the laboratory, gave me a part (sorry I can't find the name in English) to filter out of the 24v power supply.  
I also modify the wiring to avoid as much as possible that the 230v wires are in contact with the 24v wires and measures.  There is only the plateau or I have no choice.  The power wires are with the temperature probe.

MLiv79, 2021-05-01:
I have this issue too. BTT TFT 70, Latest 27 firmare version, Marlin 2.0.8 on a SKR 1.4T with everything written in config.ini activated . I got message "Busy processing, please wait" even during print, now for example is printing with progress bar and elapsed time increasing regularly but not layer. Layer freezed at 2.4mm.

Magiquall, 2021-05-01:
Personally, today I finished painting my new drawer and started to put everything back together.

I know I won't reassure you, but I feel less alone.

I'll keep you posted on this new setup, but I still need some time.

As a reminder:
I separate all 230v and 24v voltages to avoid interference.
I also move the driver fan to avoid possible interference as well.
If this is still the case, I test with the power supply of my ender3.

MLiv79, 2021-05-02:
I think i solved my issue compiling the firmware by myself. I have tontest a little bit more before to sing victory but in the last 2 prints no issues.

digant73, 2021-05-02:
@MLiv79 after your verifications, please post here your solution so it will help other users facing the some problems

MLiv79, 2021-05-02:
> @MLiv79 after your verifications, please post here your solution so it will help other users facing the some problems

I don't know the differences from the ones compiled by myself and the ones already included bin file... that's the problem to understand what was the issue at least for me

Magiquall, 2021-05-02:
Excuse me, but what firmware are you talking about? The one from marlin or the one from the screen, I admit compiling the one from the screen to see if it worked, but I haven't tried it.
If it's marlin I'd like to see your configuration files.
Thank you

MLiv79, 2021-05-02:
> Excuse me, but what firmware are you talking about? The one from marlin or the one from the screen, I admit compiling the one from the screen to see if it worked, but I haven't tried it.
> If it's marlin I'd like to see your configuration files.
> Thank you
I'm talking about bigtreetech TFT firmware. On marlin firmware side you've to enable all the required features listed in the .ini file to make the TFT works as expected

Magiquall, 2021-05-02:
Awesome,
It gives me a little more hope.

I have finished connecting the 220v, the 24v part remains to be done. But my cable raceway (chain) is too tight now that I am passing 3 more cables. I have to make it bigger again. Still time ...

Magiquall, 2021-05-04:
> I'm talking about bigtreetech TFT firmware. On marlin firmware side you've to enable all the required features listed in the .ini file to make the TFT works as expected

Good evening,

Does it still work? No new appearance of the error message and blocking?

Magiquall, 2021-05-08:
Hello,

Unfortunately, it is still the same. Same error message that appears at random.

I therefore separate the 220v part from the rest as much as possible.
I have removed the ventilation from the motherboard for now.
I tested without printing anything, just fed.
I also tested by creating my firmware from the source code (I was hoping for this solution).
I also tested with the Ender3 power supply.
I also added a ferrite to filter out the noise.
I have also tested with the SKR-2 and with the MKS SGEN_L v2.
I also tested with the drawer open and closed.

In short, I don't know what to do anymore.
Is it possible that it is just the screen that is faulty? Because I'm fed up, I don't see what more I can do.

However, the screen works perfectly in Marlin mode, but cannot print in Touch mode. Since this completely blocks printing when the message arrives, I don't want to waste time, material, and money on electricity for nothing.

I will come back to using only the Raspberry. I didn't wake up again, but it worked without a hitch before my edits, despite the screen crashing.

Magiquall, 2021-05-15:
Hello,

I see that my problem is not very popular.

Well this morning I noticed something that allows me to freeze the screen with the message "Busy processing ..." systematically in about 30 seconds to 2 minutes.

It is the setting of "Z-Offset" from the "Touch" screen I found the procedure which indicates:

1) Open the Menu
2) Go to Movement - ABL - Z Offset
3) Select Off at the bottom left to turn it on
4) Lower the z-axis by 0.1mm until you get to paper height from the build plate
5) Select next two times until you see Save
6) Select Save, then click back until you get back to the main screen

I do not have time to complete the procedure. It starts up then freezes before ending. I restarted the procedure 10 times and each time it froze. The delay is not exactly the same every time, but it is very fast, especially when I try to adjust the height of Z (step 4), if I press the button quickly it crashes very quickly.

I specify that if I do this same procedure in marlin mode, it works perfectly.
The problem being reproducible, I also tested with the power supply of the Ender 3 and it is exactly the same.

oldman4U, 2021-05-20:
Have you tried Marlin 2.0.8.1 which brings support for a third serial port and maybe something else changed which solves your problem!?

MLiv79, 2021-05-20:
> Have you tried Marlin 2.0.8.1 which brings support for a third serial port and maybe something else changed which solves your problem!?

I'm on marlin 2.0.8.1 but if i change something on the fly like speed% or fan speed, then after a while "busy processing comes and I can't do anything else until the print ends and i restart the printer. Anyway, how i can make my choice for tft serial to use?

oldman4U, 2021-05-20:
I do not understand your question. Sorry

MLiv79, 2021-05-20:
> I do not understand your question. Sorry

I mean, I've 3 serial defined but which one is used by the tft?

oldman4U, 2021-05-20:
This is defined by the mainboard and did not change.

Magiquall, 2021-05-20:
Good evening,
Truth be told, I have done so many tests that I no longer know which one is currently in the printer. I'm not sure I put the latest bugfix but I put it on the latest stable version.

Normally, if I have time tomorrow I will take the motherboard apart to change my card support and fan fixing that I will put back on top of the card.
But suddenly I am going to do a test that I have not done yet, on the table, supplied only with the USB card and the screen to see if it has. The 5v power supply does the same, since I don't need to print for it to crash.

Then, but it won't be for now, I ordered a 2nd tft35 screen from BTT, but for my Ender 3. I would test it on the SKR-2 before. (besides I was waiting for the rev B of the SKR-2 that they must send me)

Here is the news. Otherwise I did several hours of printing with the raspberry and the screen in marlin mode and it works well. Anyway, when I buy a card or a screen I would like them to work correctly.

MLiv79, 2021-05-20:
I noticed that for me happens when: 
1 for some reason i've a notification of an error with a partial string of a gcode "unknown command"
2 when i adjust some setting while printing (like speed, fan speed, flow ecc...)
3 Only in touchscreen mode
4 If i print from skr sd card reader (tft sd card/usb reader is unreliable - got prints stopped many times using these two)

Magiquall, 2021-05-21:
Hello, 
So I'm not the only one to have this problem, it reassures me, BTT should finally face something for this problem.
There doesn't seem to be much of the same problem, but it does exist.
It would be nice if it could be fixed, because when it happens to someone, the screen becomes unusable in touch mode.

oldman4U, 2021-05-21:
As far as I understand, this message comes up as soon as Marlin does not accept any commands via the serial port - because it is too busy.

Question. 1000 average guys buy the same mid class car. 5 use it off road and 2 of them (and one not going off road) have a problem with the bumpers after a year or so. As the owner of one of the cars WITHOUT a problem, would you expect that the manufacturer of the car changes something and what would you tell the others which have the problem you do not have?

Magiquall, 2021-05-22:
So MLiv79, me and the other, in fact you tell me that BTT does not care. And how are we off-road?

If it was you who were in this situation, you would probably not react the way you just did.

I'm really not happy here. Now not speaking English I can really not understand the machine translation, some case I apologize, but even if my car had a problem among 1000 people, I am entitled to wait for a repair. of course if we respect the conditions of use. How to put the card and the screen in a Hypercube instead of an Ender3 change the problem, in any case it is a hack or the product should not be sold to individuals.

MLiv79, 2021-05-22:
> So MLiv79, me and the other, in fact you tell me that BTT does not care. And how are we off-road?
>
> If it was you who were in this situation, you would probably not react the way you just did.
>
> I'm [...]

Are you compiling Marlin by yourself? Check to have all the required options wrote in config.ini active

oldman4U, 2021-05-22:
@Magiquall

In case the problem is not related to the TFT - do you think BTT should take care of the problem?

oldman4U, 2021-05-22:
BTT released a firmware update which is already merged into the master fw. This update allows to disable unneeded serial ports which could eventually help in case your problem is related to EMI.

Good luck

MLiv79, 2021-05-22:
> BTT released a firmware update which is already merged into the master fw. This update allows to disable unneeded serial ports which could eventually help in case your problem is related to EMI.
> 
> Good luck

What means EMI?

oldman4U, 2021-05-23:
Electro Magnetic Interference

Magiquall, 2021-05-23:
Hello,
I'll have the printer tested in our test chamber at work, we'll fix it.
Since as soon as there is this kind of problem it is immediately put on an EMI fault. Which I am not criticizing, because it is quite possible. But at least I would be fixed, I would also compensate the power supply of the Ender 3, not being a major brand power supply either, it may also be involved.
Hoping that my superior has time to pass it to me. It also has the possibility of generating disturbances, in order to see if the motherboard and the screen are in coherent operating tollerences before malfunctioning.
The customs service recently sent us bicycles

By the way, I made the whole thing work with a laboratory power supply and I still had the error which appeared during a print.

And I always have the possibility of making it crash quickly by trying to adjust the Z offset, there may be one parameter too much or not enough which is related to this adjustment and causes it to crash from time to time in other activities .

Magiquall, 2021-05-23:
It is the screen which indicates an error and which crashes the impressions, it is not the motherboard which it is also a BTT, but it is true that it is Marlin which is is on the motherboard. But I also tested with an MKS S_GEN L V2 and with it I have exactly the same problem.

Now I take the problem in the other direction, how to prove that it is not the screen which is in question?

And sincerely I would like to prove it in order to eliminate it from the problem that would allow me to move forward.
And frankly again, I wish it was just a marlin setting that created this problem for me, really. I would like to find this setting if it was.

I'm not here to annoy BTT, I'm here because I have a problem with one of their hardware crashing and showing me an error. It may be an external cause, but it is he who is crashing.

I will ask my manager to test my printer for possible EMI. I am not doing this to blame BTT, because I know that a power supply could be the cause. And although I tested 3 different power supplies, and that I completely redesigned my wiring in order to separate the voltages. I doubt it's the power supplies, but I'm lucky to have something to test them at work. This would really allow me to eliminate EMIs from the problem. Hoping that there is availability for this test.

I noticed that there is indeed now a COM 3 on Marlin, do you think I should activate it and if so,  I put what as a parameter?

Currently:
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1
It is indicated like that in the firmware of the SKR-2 on BTT git hub.

And I have the COM 3 commented out at the moment:
// # define SERIAL_PORT_3 1

Thank you.

I'm repeating myself, but I'm not here to get angry with BTT. Nor of you for that matter who try to help. But my problem unfortunately does not progress despite all the tests that I have already carried out.

oldman4U, 2021-05-23:
The question is how you can explain all the hundreds or thousands of users using the same TFT having no issue. I will try to reproduce the issue with the Z axis. Hopefully I can reproduce it!

👍🏻

oldman4U, 2021-05-23:
Ahh. This is ABL... I only have MBL. Sorry but i can not help in this case.

MLiv79, 2021-05-24:
> It is the screen which indicates an error and which crashes the impressions, it is not the motherboard which it is also a BTT, but it is true that it is Marlin which is is on the motherboard. But I [...]

Serial 3 doesn't solve the issue, i use it and with or without is the same.

What slicer do you use? Because i'm beginning to think that the slicer has something to do with this issue

oldman4U, 2021-05-24:
Cura. Latest version.

MLiv79, 2021-05-24:
> Cura. Latest version.

Cura has no issue as far as i know. I'd like to know also which slicer use 
@Magiquall

kisslorand, 2021-05-24:
EMI is not coming only from PSU. I had EMI problems coming from the stepper motor wires.

Magiquall, 2021-05-24:
Hello,

I use cura too, as well as prusa, but the screen also crashes without doing anything so even if I just put power on it after a while I get the same message.

Regarding EMI, indeed there is not only the power supply, especially what is with integrated inductor, power supply, transformer, motor, ventilation (it is a motor), but not only electronic components too, simple ones son etc.

In general electronic cards now have a ground plane which is supposed to reduce all its, as well as reduce the sensitivity to external disturbances.
But very often communication can be easily disrupted, COM port, I2C, RS485, and many other more or less sensitive to this problem.

MLiv79, 2021-05-24:
> Hello,
>
> I use cura too, as well as prusa, but the screen also crashes without doing anything so even if I just put power on it after a while I get the same message.
>
> Regarding EMI, indeed there is [...]

Yes it happens also to me that goes in busy processing please wait even with the printer started and idling (no bed hot no hotend hot, just idling). I enabled "ack notification"  in settings and noticed that before to get the red weird message there are some alert of "unknown command" to be confirmed. Then after confirmation the red message is on the screen and the printer need to be restarted. This, as i wrote, happens with the printer just idling not after or during a print.

Magiquall, 2021-05-25:
Hello,

How do you activate the function to see "unknown command"?

It would be nice to target this error, as I do believe an unknown order problem.

I'm going to edit at work to capture the frames that go through TFT communication.
Do you know if it's an RS232 or TTL standard?

[67 more replies]

closed 2021-06-09 by Magiquall