bigtreetech/BIGTREETECH-TouchScreenFirmware #1838: [Q] (Your Question?)

sinole1987, 2021-04-14:
I was working on configuration advanced on marlin. I came across  //#define USB_FLASH_DRIVE_SUPPORT , I was wondering if it is possible to access the flash drive on the tft screen from marlin mode? I cant access the flash drive on marlin but we can access the SD card.

bigtreetech, 2021-04-14:
No, the USB flash drive on the TFT doesn't have any hardware connection with the Marlin motherboard,  so Marlin can't read the USB  flash drive on the TFT,  Only TFT can read gcode file in touch screen mode from USB flash drive

oldman4U, 2021-04-14:
User @mark429 wrote this in another repository.

So once again.
Marli only supports a single SD card!

when setting Marlin to
#define SDCARD_CONNECTION ONBOARD
There will be an SD card on board with Marlin available in Marlin.
An SD card connected via EXP2 on the display (even by touch TFT) in Marlin will not be available
Both cards + UDB disk will be available in touch mode.

When setting Marlin to
#define SDCARD_CONNECTION LCD
The SD card on the Marlin board will be unavailable for both Marlin and touch mode.
The SD card on the display connected via EXP2 will be available in Marlin and in touch mode.
USB drive is only available in touch mode.

When connecting the display only via EXP3
the SD card in the display will not be available in Marlin with any settings.

Hope this helps

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-04-14 by sinole1987