bigtreetech/BIGTREETECH-TouchScreenFirmware #1839: PLR does not Start from where is was stopped.

sinole1987, 2021-04-14:
### Description
after powerlosss detected on the board Z raises and when power comes back. print doesnt start where it was stopped. It will Start from last layer which causes a layer inconsistency  where printed stops. doesnt completely destroy the print, but is cosmetically terrible and you have to shave it off.

### Steps to reproduce

1. power loss activated
2. go to the print directory Resume print

**Expected behavior**
nozzle go back to where it stopped.

**Actual behavior**
print starts printing last layer too close to the printed part.

### Hardware Variant
TFT 43 v3
SKR 1.4 turbo
UPS V1 24v

### Additional Information
seems like there are not PLR file saved on flash drive or SD card to know where print stopped.

oldman4U, 2021-04-14:
This so by design. The last layer which has started is written to the storage you print from.

sinole1987, 2021-04-14:
> This so by design. The last layer which has started is written to the storage you print from.

is it not possible to have it start mid layer like marlin instead of beginning of that layer. because it is printing some of the layer 2 times and since parameter starts on each layer it will show on the print.
BTW apparently PLR on marlin 2.0.7 doesn't work, looks like it was working on older versions but they broke it . I tried to use bugfix marlin, but that one is all kinds of messed up. my axis didn't even home in TFT mode. so I'm stuck with only TFT mode for PLR.

oldman4U, 2021-04-14:
Without knowing that the power will be lost very soon, the information where to start after a power loss would have to be written after every line in the gCode file. I would expect that this could cause print issues on some printers. Not sure how it works with a UPS device, but it seems that this is a common issue for you, so maybe it makes sense to buy such a device.

sinole1987, 2021-04-14:
yes, I have ups and im buying one of those prusa power panic modules, those detect power failure a bit more accurate and faster. in the mean time BTT UPS does sends the signal but i cant compile a firmware that can actually do the steps correct.

sinole1987, 2021-04-14:
I thought maybe I show you the setup here, I'm pretty good on doing hardware, not so much on software, but I am starting to understanding a little bit.
[image]

oldman4U, 2021-04-14:
I like orange👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you

bigtreetech, 2021-04-16:
> 
> **Expected behavior**
> nozzle go back to where it stopped.
> 
> **Actual behavior**
> print starts printing last layer too close to the printed part.
> 

This is a limitation of TFT needs to be optimized. It also needs the compatibility of Marlin communication protocol

sinole1987, 2021-04-16:
THANK YOU ,  I had another short power outage, I hope we can fix this soon. They are building a lot of new apartment in my area and power grid is not handling it very well.

sinole1987, 2021-04-16:
I Don't think its on every variant. I have a TFT 43
[image]

oldman4U, 2021-05-01:
@BTT. How would you like to change this in a new version? I checked with Marlin today, and it works the same. PLR starts with the last layer it has started and not with the last position.

truglodite, 2021-06-30:
Been pondering this and it seems unlikely to work well for most if positions are recorded more often, to the point of recording every line sent. Perhaps with the proper auxiliary circuit (and I'm not sure the available UPS boards can do this) the last line number executed before the power good pin goes down can be recorded to sd. This would require keeping the board alive long enough to get a line number (maybe from planner if available, for more accuracy) and store it to sd.

oldman4U, 2021-06-30:
This is about having this feature without additional hardware which would make it easier.

sinole1987, 2021-06-30:
it is possible with or without additional hardware, what marlin does is if you don't have ups to tell you power is out it constantly writes onto SD card and eventually kills it lol. but if you have ups there is plenty of time to write exact location of the nozzle. btt ups  is very beefy. Ups suppose to work in marlin mode we just need to have the same program in tft mode.

oldman4U, 2021-06-30:
Last time I tested this with Marlin it has been working the same way the TFT works. But I can only say how it was without UPS.

sinole1987, 2021-06-30:
I have been very busy with life and been experimenting with resin. I didn't have time to upgrade marlin. on marlin 2.0.7, I setup marlin to go to where print stopped exactly. Unfortunately there was a bug in marlin and didn't keep on printing it just stopped at the break off location. hopefully they addressed that on marlin 2.0.8. TFT has another issue as well. since we don't have M75 on resume, you can only recover once and it recovers from last layer line which you printed most likely half way.

sinole1987, 2021-07-09:
I was thinking about M600 and how it pauses the print and starts exactly where it stopped, for people who have a ups script is as easy as saving the location where print stopped and start it from there. with ups heads up printer have plenty of time to do that. I don't know why we can't have that.

closed 2024-03-15 by github-actions[bot]