bigtreetech/BIGTREETECH-TouchScreenFirmware #1840: After upgrade to TFT43_V3.0.27 firmware, printer does not respond to touch command

twinsenfan, 2021-04-14:
**Your question or details you need. Please describe.**
Where can I find the old TFT43_V3.0.26 firmware ? 

**Specify the device variant related to your question.**
TFT43_V3.0 updated to TFT43_V3.0.27 version
I had TFT43_V3.0.26 before the update, and it was running just fine. I am trying to downgrade before doing any further troubleshooting to isolate if the issue is caused by the new firmware.

**Additional context**
Add any other context or screenshots about the question here.

kisslorand, 2021-04-14:
Put an empty `reset.txt` on you SDCard and reboot TFT. This will recalibrate the touch on your TFT.

twinsenfan, 2021-04-14:
Thank you for the prompt response. Yes, I have done that, touched the red dot etc. The TFT will respond to the touch actions and I can switch between different menus. It also shows the current temperature of the hotend, bed and printer is ready status. Just non of the command(movement, heat up hotend seem to do anything)

oldman4U, 2021-04-14:
Click. Commits to get access to all "versions" which have been released so far.
[image]

twinsenfan, 2021-04-14:
Didn't have luck finding the bin file for TFT43, is it normal that the bin for TFT43 screen is missing ?
[image]

oldman4U, 2021-04-14:
For a version which has been released before the hardware was available - yes.

I just tested v27 on my TFT43 and it works. Given the fact that the update alone did not work properly, I would expect a problem with the SD card or the downloaded file.

Clicking on the "868 commits" I get this page:
[image]

twinsenfan, 2021-04-14:
Sorted it out, it was a weird cable problem. Not firmware related. Thank you all for the input.

closed 2021-04-14 by twinsenfan