bigtreetech/BIGTREETECH-TouchScreenFirmware #1845: [BUG]  #1785 breaks mesh fine tuning

lightmaster, 2021-04-17:
### Description

#1785 added some code to change the way mesh fine tuning works. Previous to this PR, it was possible to use a piece of paper to fine tune your UBL mesh, lower the nozzle until the paper was a little snug under the nozzle. After fine tuning all mesh points, you could make sure that the UBL mesh was active and then tell the nozzle to go to Z0 and verify that the paper was still a little snug across the entire bed.

Now, once you fine tune a point, the mesh editing UI sets your z-offset for that point to 0.200mm too low. For example. the correct Z-offset for a point is 0.060mm (Z0.060 is the height at which the paper has just become snug). Instead of setting the Z-offset for that point to 0.060mm, it instead sets it to -0.140mm. This in turn makes it so that Z0.14 is where the paper is snug, and sending the nozzle to Z0 causes the nozzle to crash into the bed hard.

Previous to PR #1785, verified by reverting my fork to the commit before the PR, mesh fine tuning with a piece of paper works perfectly. I propose undoing any changes from PR #1785 that modified the mesh edit fine tuning to restore that functionality. Unfortunately, I don't know the code language anywhere near well enough to be able to revert just this part myself.

### Steps to reproduce

1. Create a mesh with `G29 P1`
2. Fine tune the mesh with Mesh Editor and a piece of paper
3. Observe the Z-offset for the spot being set far lower than what you selected
4. Send command to send the nozzle to Z0
5. Observe the nozzle crashing into the bed.

**Expected behavior**
Should be able to fine tune a mesh with a piece of paper to set the exact right nozzle offset across your entire bed.

**Actual behavior**
Every point's z-offset is set to much lower than the value you select.

### Hardware Variant
TFT43

### TFT Firmware Version & Main Board Firmware details
TFT firmware: Last commit as of 2021-04-16
Marlin firmware: Last commit of `bugfix-2.0.x` as of 2021-04-16

### Additional Information
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/6328801/Configuration.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.

digant73, 2021-04-17:
It is the opposite. The old mesh editor didn't apply any shim to use a paper. Nozzle was moved to the probe point with no extra shim.
The new mesh editor added that shim requested by other people, The same is applied also to Probe Offset. You can set the shim changing the config param "level_z_pos".
Furthermore mesh editor now always disable BL just to avoid to apply meshes during the mesh tuning.

oldman4U, 2021-04-20:
Why not using a feeler gauge instead of a paper or nothing?

Using paper for one method but not for another is not very common and the firmware has finally to work for the majority of the users. I think you agree?

digant73, 2021-04-20:
@lightmaster please, no reason for any disappointment. I need further details based on my following explanation/clarification.
The previous Mesh Editor (and also the Probe Offset) were not working in case BL was active. Furthermore the use of a shim (a paper in your case) was not also possible because the nozzle was moved exactly to the probing point with no extra shim to use a paper. I also don't understand what you mean when you say "Z-offset for a point is 0.060mm (Z0.060 is the height at which the paper has just become snug)". It seems that you used a paper during UBL process while I expect that 0.060 is the probing point (so with no shim at all taken into account). For that reason it was added the shim. 0.060 + 0.200 will allow a paper (having a thickness of 0.200) to become snug. So you will then need to remove the shim to obtain the correct z height (the nozzle is close to the bed). That means the final result of 0.060. If you say that with a paper you need to lower the nozzle to 0.060 so you will obtain a final z-0.140 (0.060 - 0.200) it means you used a paper also during the entire UBL process. If so, is there any reason for doing that?

sarvenn, 2021-04-24:
This might be relevant:
https://github.com/MarlinFirmware/Marlin/pull/21692

digant73, 2021-04-26:
@sarvenn it simply someone asking to restore the previous MBL logic. But there is no problem even the current one. Simply a different starting Z height when moving to the next point

lightmaster, 2021-04-26:
Seems my issue stemmed from both setting the probe offset to a paper's thickness instead of the bed itself, and that apparently my cheap printer paper is half as thick as the paper @digant73 has.

closed 2021-04-26 by lightmaster

digant73, 2021-04-26:
good to hear you solved the problem. What setting did you use at the end? that info could help other users facing similar problems. The paper thickness could be very different (0.1 in my case). Other people prefer to use a filler gauge (e.g. 0.2) etc...

lightmaster, 2021-04-26:
I had to use 0.1mm. And that was using plain old cheap printer paper.

I do think that the shim value should be editable directly on the printer (Settings => Features screen) since someone might have multiple types of paper on hand. No sense having to adjust and flash `config.ini` for that value imo.

digant73, 2021-04-26:
@lightmaster the configuration of diffferent paramters at runtime is a well known request from many users. The main problem is that it takes space on flash so a list of the most important params was made available at the moment