bigtreetech/BIGTREETECH-TouchScreenFirmware #1849: BTT Ups not working after upgrade to version 0.27

ahmetbozaci, 2021-04-17:
After upgrade to  this "BIGTREE_TFT35_V3.0.27.bin" version. Ups is not working anymore with marlin mode or touch screen mode. I don't know I miss something or not.

**Here is my configs:**

- marlin/src/feature/powerloss.h

`//#define DEBUG_POWER_LOSS_RECOVERY`
`#define SAVE_EACH_CMD_MODE`
`#define SAVE_INFO_INTERVAL_MS 0`

- marlin/configuration_adv

`#define POWER_LOSS_RECOVERY`
`#if ENABLED(POWER_LOSS_RECOVERY)`
`#define PLR_ENABLED_DEFAULT true `
`#define BACKUP_POWER_SUPPLY     `
`//#define POWER_LOSS_RECOVER_ZHOME  `
`#define POWER_LOSS_ZRAISE 10    `
`#define POWER_LOSS_PIN P1_25      `
`#define POWER_LOSS_STATE HIGH    `
`#define POWER_LOSS_PULL   `
`#define POWER_LOSS_PURGE_LEN 20  `
`#define POWER_LOSS_RETRACT_LEN 10`
`#define POWER_LOSS_MIN_Z_CHANGE 0.05 `
`#endif`

- marlin/pins

`#ifndef POWER_LOSS_PIN`
`#define POWER_LOSS_PIN P1_00 // PWRDET` // **I connected ups cable to this pin. I used 'P1_25' for btt relay**
`#endif`

- tft/config

`pl_recovery_en:1`
`pl_recovery_home:0`
`pl_z_raise:10`
`btt_mini_ups:1`
`start_gcode_enabled:1`
`end_gcode_enabled:1`
`cancel_gcode_enabled:1`
`start_gcode:M75\nG28 XY R20\n`
`end_gcode:M77\nM104 S0\nM140 S0\nM107\nM18\n`
`cancel_gcode:M77\nM104 S0\nM140 S0\nG28 XY R10\nM107\nM18\n`

oldman4U, 2021-04-18:
What exactly does not work?

The UPS is connected to the SKR and the config files you sent are from Marlin. In case power is lost the UPS let the SKR know and not the TFT, right? How is this related to the TFT firmware?

ahmetbozaci, 2021-04-18:
> What exactly does not work?
> 
> The UPS is connected to the SKR and the config files you sent are from Marlin. In case power is lost the UPS let the SKR know and not the TFT, right? How is this related to the TFT firmware?

Z axis is not raising after plug out. Yes ups is connected to board. I already added tft config. last of the issue

oldman4U, 2021-04-18:
Have you tried to disable M600 emulation in the Settings?

ahmetbozaci, 2021-04-18:
In tft config? It is filament change gcode. It is relate with ups?

oldman4U, 2021-04-19:
It is the only setting which is somehow related which changed.

You can change it directly using the TFT.

ahmetbozaci, 2021-04-19:
I changed. it is still same. Z is not raising. Maybe problem is marlin version. I read ups is not working.2.0.7.2 version. But ı remember it was worked before upgrade to TFT

oldman4U, 2021-04-19:
Ok. 

If it is really TFT firmware related you could try to find a version which works and then the PR which broke it.

oldman4U, 2021-04-26:
Hi.

What is the status of the reported issue? Have you found a TFT firmware which works?

Please let us know, thank you

ahmetbozaci, 2021-04-26:
No ı didnt try other tft firmware. I just try marlinbugfix. But it is still not working. 
https://github.com/MarlinFirmware/Marlin/issues/19602

oldman4U, 2021-04-26:
You wrote it does not work after updating the TFT firmware so I assume it has been working before updating it. In this case you should try to find the last TFT firmware which works and let us know.

ahmetbozaci, 2021-04-26:
OK ı will try v26 with marlinbugfix and write in here.

ahmetbozaci, 2021-04-26:
Nope it is still not working. I dont know what is wrong. I install v26 and marlinbugfix
Here is my settings:
https://github.com/ahmetbozaci/mini-UPS-test

oldman4U, 2021-04-27:
Makes sense because the UPS controls Marlin on the Mainboard and not the TFT. Most likely there is a problem with Marlin or a setting of it.

oldman4U, 2021-04-28:
Check the master of today to see if it works for you.

oldman4U, 2021-04-28:
Sorry. It is not released.

oldman4U, 2021-04-28:
But you can test the branch if you want: https://github.com/Msq001/BIGTREETECH-TouchScreenFirmware.git

I do not expect it to fix your problem, but you never know

oldman4U, 2021-04-28:
Now it is in master branch: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware.git

Please test and let us know

ahmetbozaci, 2021-04-28:
> But you can test the branch if you want: https://github.com/Msq001/BIGTREETECH-TouchScreenFirmware.git
> 
> I do not expect it to fix your problem, but you never know

ı tested it not working. I will double check settings and cables again try master branch.

sinole1987, 2021-05-03:
do you have m75, M77 in your start and end Gcode ? because mine is working.

ahmetbozaci, 2021-05-03:
> do you have m75, M77 in your start and end Gcode ? because mine is working.
I added gcode and using marlinbugfix.
Here is my settings
https://github.com/ahmetbozaci/mini-UPS-test

closed 2021-07-09 by stale[bot]