bigtreetech/BIGTREETECH-TouchScreenFirmware #1855: E3-TFT35 seems unaware that print has started from DWC on E3-RRF board

Wombat37, 2021-04-19:
### Description

When uploading g code file to print via WiFi using DWC or Cura Plugin, the local TFT35 display appears unaware that this is happening. 

### Steps to reproduce

1. Send file to print via DWC to E3-RRF board

**Expected behavior**
When a print is initiated from the TFT35, a print status screen is shown and there's access to buttons relevant to the active print (like pause etc.). When a file to print is sent remotely via WiFi, I would like the TFT to be aware of this and go into print mode - to provide status and functions related to active printing and not allow the user to press buttons that cause it to crash.

**Actual behavior**
The only indication that something is going on is that the temperatures appear to change - that's it. If other buttons (like movement, homing etc.) are now pressed, the system locks up

### Hardware Variant
E3-TFT35
### TFT Firmware Version & Main Board Firmware details
3.0.27 (RRF 3.2.2) 

### Additional Information

bigtreetech, 2021-04-20:
Is it the same problem as in #1746? Did you made the required sys files changes mentioned in #1743 ?

Wombat37, 2021-04-25:
Just made the sys file changes listed in #1743 

This fixed the issue - thanks

Not sure how I would have found that fix unless told as above.

Will these changes find their way into the supplied sys files in the BTT-E3-RRF GitHub repository?

closed 2021-04-27 by Wombat37

bigtreetech, 2021-04-27:
The sys file of BTT-E3-RRF has be modify in https://github.com/bigtreetech/BTT-E3-RRF/commit/e1ec3b4ddd719c813a09f34c2a4d2be5376973c9

Thank you for your advice.