bigtreetech/BIGTREETECH-TouchScreenFirmware #1857: RRF Macros [FR] (feature request title)

TechnicalWorld2016, 2021-04-20:
Hi 
I want to use make some macros for Idex mode for my 3d printer which has a Bigtree tech TFT28.
Can someone help me how to create or add custom macros for BTT TFT

Any solutions will be help-full for me.
Thanks!

oldman4U, 2021-04-20:
Please use #1791 for this kind of FR.

@bttguy can you help?

bttguy, 2021-04-20:
Hi! Can you specify in what do you ask for help? It's not clear for me.TFT 28 is supported, so does the macros. You can set up new macros in DWC, or you can place them on the mainboard sd card to /macros as .gcode files. The specific macros are machine specifics, you have to make them yourself to suit your machine.

Eshwar1414, 2021-04-21:
> Hi! Can you specify in what do you ask for help? It's not clear for me.TFT 28 is supported, so does the macros. You can set up new macros in DWC, or you can place them on the mainboard sd card to [...]

Hi Thank you for the reply.

I was asking for the process of setting up the macros of our own choice, and what are the steps exactly.
If any solutions please post here.

TechnicalWorld2016, 2021-04-21:
> Hi! Can you specify in what do you ask for help? It's not clear for me.TFT 28 is supported, so does the macros. You can set up new macros in DWC, or you can place them on the mainboard sd card to [...]

Hi Thank you for the reply.

I was asking for the process of setting up the macros of our own choice, and what are the steps exactly need to follow.
If any solutions please post here

bigtreetech, 2021-04-21:
Hello, The current firmware can read the Macro of RRF directly, and then choose to execute it. 
You just need to put the Macro in the macros folder in the sys folder of RRF according to the normal RRF usage method.
And rename the pre provided `config_rrf.ini` file to `config.ini`, and then put it into SD card to update the configuration file. Then the `Custom` button of the screen will automatically modify to `Macro` button, This button will read the macro of RRF

TechnicalWorld2016, 2021-04-21:
> Hello, The current firmware can read the Macro of RRF directly, and then choose to execute it.
> You just need to put the Macro in the macros folder in the sys folder of RRF according to the normal [...]

Thank you for the reply!
Actually I am using Marlin 2.0 
So I am confused to where to add or define the macros in the marlin 2.0 file.

If possible please suggest any process.

radek8, 2021-04-21:
When you use Marlin, why do you solve macros?

guruathwal, 2021-04-22:
> > Hello, The current firmware can read the Macro of RRF directly, and then choose to execute it.
> > You just need to put the Macro in the macros folder in the sys folder of RRF according to the normal RRF usage method.
> > And rename the pre provided `config_rrf.ini` file to `config.ini`, and then put it into SD card to update the configuration file. Then the `Custom` button of the screen will automatically modify to `Macro` button, This button will read the macro of RRF
> 
> Thank you for the reply!
> Actually I am using Marlin 2.0
> So I am confused to where to add or define the macros in the marlin 2.0 files.
> 
> If possible please suggest any process.

There are no macros in the Marlin firmware. you need to use the custom gcodes option in the config.ini file to change any of the custom gcodes.

TechnicalWorld2016, 2021-04-22:
> There are no macros in the Marlin firmware. you need to use the custom gcodes option in the config.ini file to change any of the custom gcodes.

Thank you @guruathwal 
Even I came to know it yesterday itself.. And I had also tried the custom gcodes option in config.ini file .
It is working simply superb ...
But I am getting the labels of gcodes in the list view. Is there any solution so that I can see the contents of Custom menu in grid view.

Thank you

oldman4U, 2021-04-22:
@bttguy

Maybe we can make an extra rrf section in the Readme. Also the need to rename the config file is missing and therefore wrong and it seems that you should also include some basic information like supported firmware version snd so on. Could you please check the current ReadMe and make suggestions. Thank you

bttguy, 2021-04-22:
@oldman4U Yeah, I already started working on it in the discussions. Will expand it and try to add to readme.
Can you suggest were should I add RRF related infos within readme? Also I am not sure how to include it to the table of content as a "bookmark" for quick jump to it.

oldman4U, 2021-04-22:
I can add it to the ReadMe or I can show you. What would you prefer?

oldman4U, 2021-04-22:
Btw, TechnicalWorld2016, you can close this ticket whenever you want.

Thank you

TechnicalWorld2016, 2021-04-22:
> Btw, TechnicalWorld2016, you can close this ticket whenever you want.
> 
> Thank you

closed 2021-04-22 by TechnicalWorld2016